LibUsbDotNet/LibUsbDotNet #5: Repos and refactoring

andrasfuchs, 2017-02-01:
I think there is a lot of confusion in forums, on websites etc. about where should someone look if they wanted to use USB with C#.

Here are the repos which are somehow related to the issue:
- this one, obviously
- https://github.com/libusb/libusb: original C++ libusb
- https://github.com/Stevie-O/LibUsbDotNet: we are originated from this. I think we should ask Stevie-O to transfer the ownership to you. It makes no sense to maintain that repo without merging the updates happening here
- https://github.com/madwizard-thomas/winusbnet/: libusbdotnet uses this , it is a wrapper to WinUSB. He might be open to a cooperation and bring all USB-related .NET libraries into one organization (on GitHub).

Architecture-wise we have now the opportunity to modernize the structure of the projects and their source code. As I worked on the libusbdotnet source code, I got the impression that it would be a good idea to totally separate some of its parts. These are the following:
- WinUSBNet: hosted in the repo mentioned above, wrapper of winusb.dll. I think an old version of this is in the LibUsbDotNet.WinUsb namespace.
- LibUSBNet: that would be a lightweight wrapper of libusb-1.0.dll/.o. Its now under the MonoLibUsb namespace in the LibUsbDotNet project.
- LibUSBwin32Net: that would be a lightweight wrapper of libusb0.dll. Its now under the LibUsbDotNet.LibUsb namespace in the LibUsbDotNet project.
- LibUsbDotNet: this would use the three libraries above. This way anyone could decide if they want a light-weight wrapper for one particular implementation or a more general library, the LibUsbDotNet.
- tools: InfWizard, Benchmarks should be also in a separate repo IMHO.
- example code: the same as above, they are very useful, but only loosely related to the code project.

I don't know it yet, how exposed the methods are in the three wrapper libraries, but if they are not, then this separation and refactoring would be possible with backward compatibility of the LibUsbDotNet. Btw, do you think it is important to have backward compatibility?

We should also get rid of the 'stage' folder, I think it come from the older SVN source.

Travis created libusbK.dll too, it might be also good to have that wrapper in a repo in our organization, although that's not needed by LibUsbDotNet. 

We could finally have one, unified place to explain and host all these, so planning and working with .NET USB would be not such a confusing task.

What do you think?

mcuee, 2017-02-01:
I do not think libusbdotnet uses any codes from WinUSBNet since it is much earlier than WinUSBNet.

As for the tool, I think InfWizard is obsolete and likely do not work under later version of Windows. Zadig or libusbK's GUI installer is the preferred tool.

mcuee, 2017-02-01:
Personally I do not think backward compatibility is that important if you guys are trying to modernize the project. If that is still desired, then an idea is to call the original libusbdotnet libusbdotnet-legacy and then call the new project libusbdotnet where you forget about the compatibility with the legacy. Or call the new project libusbdotnet-NG or things like that.

mcuee, 2017-02-01:
https://github.com/libusb/libusb: original C++ libusb --> actually should be "C libusb", there is no C++ there.

qmfrederik, 2017-02-01:
Inspiring stuff!

Yes, absolutely, I've noticed the lack of information every time I've worked with libusb and the related projects.

I think it makes sense to position this as a ".NET API for USB devices" project which allows you to communicate with USB devices using .NET code.

It's a bit unfortunate there are multiple USB libraries which you can use (winusb, libusb0, libusb-1.0, libusbK). But I think each of them has some unique advantages (for example, as far as I know, only libusb0 allows you to change the configuration of a device) so we'll have to maintain that for now.

As for the naming, I'm open to anything; perhaps it may make sense to use a common prefix though? Like 'LibUsbDotNet' for the project and the common interfaces, and then something along the lines of 'LibUsbDotNet.WinUSB' , 'LibUsbDotNet.LibUsb'  and 'LibUsbDotNet.LibUsb32'.

I would not object to removing the InfWizard; Zadig / libusbK' s installer being better alternatives.
Similarly, the 'stage'  folder indeed can go.

I've created a LibUsbDotNet organization and I can suggest we move this repo there. I also think we can ask GitHub to remove the "forked from" link (since there really is no activity in the upstream repository).

An added advantage of that is that we can then also host a GitHub site, which could contain additional documentation/information, similar to http://libusbdotnet.sourceforge.net/.

Let me know if creating an additional organization makes sense to you guys.

mcuee, 2017-02-02:
Good idea to have a dedicated repo: https://github.com/LibUsbDotNet

A ".NET API for USB devices" project, that is quite nice. Hopefully more people will be interested and contribute.

andrasfuchs, 2017-02-02:
Great!

@mcuee Maybe the WinUSB wrapper in libusbdotnet is not related to the WinUSBNet wrapper then? Anyhow, they serve the same purpose, so we might spare some time there, by use some code from that repo (if the license permits it)

I would be happily leave the backward-compatibility, it would make our job much easier.

libusb is C indeed, sorry, my bad!

@qmfrederik I like the naming, good idea!

About the different libraries, a question came to my mind. 
What should be the actual scope of the project?
I haven't worked enough with libusbdotnet yet, but it is very strange for me that it needs wrappers for winusb and libusb0. I thought it should use libusb-1.0 only, which is an abstraction of USB access on different platforms. So, I might be wrong, but I think under the hood libusb-1.0 chooses the most suitable driver depending on the target platform already, so why should libusbdotnet bother with those?
We discussed earlier that libusb-1.0's Windows support lag a little behind the Mac and Unix, but I don't think we should make it up in our C# library, that would be the C library's task, isn't it? 

Could you share your thoughts on this issue?

qmfrederik, 2017-02-02:
@andrasfuchs I may be wrong but I think I had to use libusb-win32 at one point in time to set the configuration of a device to a configuration which is not the first one. But I may be wrong.

So I guess we can go ahead and target libusb-1.0, but just make sure we also keep abstractions in the form of interfaces (would also make unit testing easier).

andrasfuchs, 2017-02-03:
Alright then, sounds good to me.

@qmfrederik Would you contact GitHub because of the fork? Or it would be even better to get the current maintainer (Stevie-O) on board, or not? Or I can even imagine that he is gladly transfers the repo to our organization if we explain the situation. It would be important only because of the users: we should have only one repo, duplicates on GitHub could cause confusion.

So, regarding my part of the work, I have already started to increase the libusb-1.0 coverage by adding a few more methods to its wrapper. I've also created some NUnit unit tests (it will need a small modification in AppVeyor's config, to run them). They are not in a pull request yet, but it would probably make sense to pull them before we start the separation of the different wrappers into the 'LibUsbDotNet.WinUSB' , 'LibUsbDotNet.LibUsb' and 'LibUsbDotNet.LibUsb32' libraries.

What do you guys think, should we have separate projects for the libraries, or even separate repos within our organization? I think both are fine approaches, and I'm a little bit for the separate repos, because then we can handle the issues/wikis/commits/releases of them separately as well.

mcuee, 2017-02-03:
libusb-win32 kernel driver (libusb0.sys) supports USB device with multiple configurations whereas WinUSB and libusbK.sys (libusbK kernel driver) supports only single configuration. Other than this limitation, libusbK.sys can be considered as a superset of libusb0.sys.

libusb0.sys also has the filter driver mode (the device driver mode is in general stable but may not work with some USB device). 

The above two are the main benefits of the libusb0.sys kernel driver.

On the other hand, there is a distinction between the underline driver and the library (API, or the dll ). 

libusb-win32 API is only for Windows but it is a superset of the cross-platform libusb-0.1 API, it can support device using libusb0.sys and libusbK.sys.

libusb-1.0 API is the most popular cross-platform API, under Windows it can support device using WinUSB (stable), libusbK.sys (stable) and libusb0.sys (unstable). It also supports generic HID device and usbdk ( a new open source generic USB driver).

libusbK API is only for Windows. It can support device using WinUSB, libusbK.sys and libusb0.sys.

WinUSB API (winusb.dll) is only for Windows, it can supports device using WinUSB.

mcuee, 2017-02-03:
For low level API, I think it is good to separate into different repos. For high level API abstraction, it may be better to use a single repo to support multiple drivers if possible. An example is Python based pyusb, which is kind of high level API which supports libusb-0.1 API, libusb-1.0 API and OpenUSB API.

As for why libusbdotnet has something only for libusb0.sys and WinUSB and not using libusb-1.0 API, I think it is because libusb-1.0 support was added later.

andrasfuchs, 2017-02-03:
@mcuee Wow, I need your brain! :) This is even more complicated than I thought! No wonder people got confused. Have you thought of writing a nice, easy to understand tutorial about this whole situation? You have the knowledge for it, that's for sure.
Do you know by any chance which version of WinUSB does libusb-1.0 supports? I'm sure how much it changed, but you mentioned earlier that libusb-1.0 does support isochronous transfer, although WinUSB since Win 8.1 does: https://msdn.microsoft.com/en-us/library/windows/hardware/ff540196(v=vs.85).aspx

mcuee, 2017-02-03:
BTW, it is possible to directly talk to device using libusb0.sys without using libusb-win32 API (eg: libusbK.dll is doing that). It is actually possible to directly talk to libusbK.sys without libusbK API as well but I have not seen people doing it. 

Ref: driver API using IOCTL codes
https://github.com/mcuee/usb-travis/blob/master/libusbK/src/drv_api.h
https://github.com/mcuee/libusb-win32/blob/master/libusb/src/driver/driver_api.h

From the above, you can see that libusbK.sys is compatible with libusb0.sys and it is actually a superset.

For WinUSB, you will have to go through winusb.dll.

mcuee, 2017-02-03:
@andrasfuchs I will try to write some FAQs here. I understand that it is confusing.
https://github.com/libusb/libusb/wiki/FAQ

As for libusb-1.0 Windows, it does not support isochronous transfer so far with the WinUSB/libusbK backend even though libusbK supports isoc transfer and WinUSB post Windows 8.1 supports isoc transfer as well.
    https://github.com/libusb/libusb/issues/46

However, the new Windows usbdk backend supports isoc transfer. usbdk is interesting.

mcuee, 2017-02-03:
usbdk
https://github.com/daynix/UsbDk

Talking about usbdk, it is another open-source generic USB driver. So far I know libusb-1.0 API supports it. It has its own simple API here.
https://github.com/daynix/UsbDk/blob/master/UsbDkHelper/UsbDkHelperHider.h
https://github.com/daynix/UsbDk/blob/master/UsbDkHelper/UsbDkHelperHider.h

andrasfuchs, 2017-02-10:
@qmfrederik How are things going? Could you let us know, when the new organization and the repos are ready? What do you think about Stevie-O's repo?

Should we put together an action plan? I have a few extra hours now, but to be honest I don't know what to do on libusbdotnet next, so I'll move to another project today.

mcuee, 2017-02-13:
https://github.com/libusb/libusb/wiki/FAQ
I have tried to update the libusb FAQ to try to explain the differences among libusb, libusb-win32 and libusbK.

andrasfuchs, 2017-02-15:
@mcuee Cool stuff, thank you!

qmfrederik, 2017-02-16:
@andrasfuchs Well, I think the repo is all ready by now :) I would let Stevie-O's repo be for now, I've submitted a ticket with GitHub to have them remove the "forked from" label - let's see if that works.

Regarding the next steps, my thinking would be that there are three major tasks now:

1. Remove the stage folder & remove dead code (e.g. Benchmark, InfWizard, LibWinUsb, Utility,...)
2. Come up with a proposal for the API rework
3. Update the license headers to mark all code as LGPL (as agreed with Travis)

My time to work on this will be a bit limited in the coming weeks, but happy to look at issues/PRs as I have time.

qmfrederik, 2017-02-16:
@andrasfuchs Looks like GitHub removed the "forked from" label from this repo, so we're good on that one ;-)

andrasfuchs, 2017-02-20:
@qmfrederik I know we are all busy, but could you estimate how much time could you spend on this project weekly?

I started to use WinUSBNet for my project, so I'm not so eager to work on LibUsbDotNet anymore, but if you also put in the hours, I'm in ;) I think I will be able to spare 4 hours per week if you guys are partners in it. 

Relating to my repo question: my impression was that it would be better to separate the library projects ('LibUsbDotNet.WinUSB' , 'LibUsbDotNet.LibUsb' and 'LibUsbDotNet.LibUsb32') and have their own repos.  Are you willing to create them if I make the code separation?

level54, 2017-04-15:
Hi all; glad to see you guys are pulling all the projects together. Have you perhaps given any thought to including support for Windows 10 IoT Core? I am playing around with IoT Core on a Raspberry PI and Microsoft's support for USB devices (and serial devices for that matter) on IoT Core is almost non-existant.
I have spent the last two days searching the internet for any form of usable code that I can use to access a Vendor Specific HID Device, and the closest that I got was http://allican.be/blog/2017/01/28/reverse-engineering-cypress-serial-usb.html
It will be great if you would consider support for IoT Core specifically running on ARM.
I am willing to help with the UWP wrapper but I need the ARM/x86/x64 (sys) drivers to do that.

Thanks in advance.

nzsmith1, 2017-05-08:
Hey all, I am very interested in this project and would like to contribute, I am traveling starting Thursday for 2 weeks but can commit a few hours a week before that.

@qmfrederik I like your proposal for refactoring:
>As for the naming, I'm open to anything; perhaps it may make sense to use a common prefix though? Like >'LibUsbDotNet' for the project and the common interfaces, and then something along the lines of '>LibUsbDotNet.WinUSB' , 'LibUsbDotNet.LibUsb' and 'LibUsbDotNet.LibUsb32'.
>-@qmqmfrederik 

If no one else is working on this at the moment I would like to take this on to get my feet wet on the code-base.

qmfrederik, 2017-05-08:
@nzsmith1 Sure, all help is welcome!

I guess a first step would be to try & define the common interface(s), re-organizing the code on a per-backend basis could come after that.

Perhaps for work that you plan to take on:
1. Create a GitHub issue outlining what you will do, so others can provide feedback
2. I'll then assign the issue to you, letting others know it's WIP
3. I'd suggest to try and land the work in multiple, smaller PRs - this will make it easier to review.

Thanks!

andreymrt, 2017-10-04:
@andrasfuchs Couple comments from a past libusbdotnet user (me)
1. Since Microsoft has sunset XP, the only practical versions of windows to target are 7 and 10, both of which have WinUSB.
   In Win10, WinUSB is complete and can serve as the master driver.
   In Win7, WinUSB lacks isochronous mode, which shouldn't be a showstopper for the overwhelming majority of use cases.
   In regards to distributing windows applications, it's much easier to ship libusbdotnet dll and NOT have to install a separate driver (libusb), so splitting the development into a single WinUSB backend, and a *nix backend should make things easier.
2. The InfWizard is awesome (I just started exploring Win10, so I can't comment on its usefulness there), it would be even better to generate a few udev and setup scripts for Linux, as that has been a pain point for app distribution.
3. The SF site is outdated, but you already know that. Any plans to set up and move the official documentation to github pages?
4. The releases are also outdated. While the NuGet packaging is nice, it would also be helpful to have GitHub release versions of zipped code and zipped compiled libraries.

@level54 I'm not sure I understand your question, are you asking for libusbdotnet slave configuration support? if so, that would be a whole different beast.
If it's about connecting to a custom device, it's not super difficult and most of the documentation for it is already in the sourceforge doc site.

qmfrederik, 2018-09-10:
So, wow, time flies 😄 .

Anyway, I have taken a closer look at LibUsbDotNet in the past couple of days, and done some work to split off the .NET wrapper for libusb-1.0 (the currently the  `MonoLibUsb  namespace) from the .NET wrappers for the Windows-specific drivers (`winusb` and friends).

The question is where to go from here. Looking at the code, I've come to the conclusion that:
1. wrapping libusb-1.0 is really what we should be focusing on, as this gives us a fairly feature-complete, cross-platform way of interacting with USB devices.
2. as libusb-1.0 also supports winusb, libusb-win32 and libusbK backends, there's really little value in providing a wrapper for those in managed code.
3. there's a lot of work that can be done to improve the libusb-1.0 wrapper. I'm doing some work to set up unit tests on Windows, Linux and macOS and they already highlight some issues. There are also reports of compatibility issues on ARM hardware.

Therefore, I'm thinking that it may make sense to remove the .NET wrappers for the Windows-specific wrappers altogether and stick with libusb-1.0 only.

It seems a bit of a drastic move, though, so I wanted to consult with the people on this thread first to get a quick show of hands.

If you're currently using LibUsbDotNet, which USB drivers are you using? Would it be fine if we only kept support for libusb-1.0?

chrfin, 2018-09-10:
No Problem from our side. We use our "own package" at the moment (as 2.2.8 had the a revision missing, which had a fix we needed) and will continue to do so, if the new "official" package won't work.
But we will try to use the new official package and delete our own as soon/long it works for us...

AshleyRoll, 2018-09-10:
Does removing the support for WinUSB imply I will need to ship a libusb-1.0 dll under windows, and if so, does that dll need to be associated with the device using an INF file? This would be an issue for me. 

Our devices are naively WinUSB - meaning that we take advantage of the full enumeration process and automatically have windows associate the WinUSB drivers with the device. We do this with the _Extended Compatibility ID OS Feature Descriptor_ and the _Extended Property Descriptor_ providing the `DeviceInterfaceGUID` property. Essentially this tells windows to directly associate WinUSB.dll as the driver and gives us access to search for the device using the GUID. 

Our device doesn't ship with any drivers or INF files at all. The hardware is plugged in, Windows automatically configures drivers and the C# application runs as and works as the user space driver (assuming Vista or later from memory, but we state Win10 as required). 

We originally were using native P/Invoke to the WinUSB system calls, but swapped to LibUsbDotNet when we were experimenting with an embedded Linux device using our hardware and re-using our C# code under mono. We have since moved that particular hardware platform and switched to C/C++ and direct libusb-1.0 calls, but left LibUsbDotNet in our C# main application as it was a nicer to use.

chrfin, 2018-09-10:
@AshleyRoll That should still work as WinUSB is still compatible as a driver when using libusb-1.0 as the "top-level interface".
BTW: We use libusb0 and libusbk as our driver as we had Bluescreens with WinUSB (at least with Windows 7).

AshleyRoll, 2018-09-10:
@chrfin Cool. I'm happy as long as I don't need an INF file :)

WinUSB was pretty stable for me. But I was on Win10 at the time. From memory Win 7 support for WinUSB was not great - manual install etc. 

Worst case for me is to remain on the current version. I'm happy to trial new versions as needed.

Proclivis, 2018-09-10:
> Also cc'ing @Canomation, @BendRocks, @metanoic, @AshleyRoll, @claunia, @TAGC, @serhanyilmaz, @seuzht, @Proclivis, @chrfin . You've opened or commented on issues in this repo, and it would be worthwile to get your take on this.
>
> So simple question: are you currently using the .NET wrapper around libusb-1.0, or other parts of LibUsbDotNet? Would you be impacted if we reduced the scope of LibUsbDotNet to support libusb-1.0 only? How?

We are using libusb on linux. For Windows we use a Cypress driver after failing with LibUsbDotNet. For me the big issue is the complexity of the code base makes it impossible to debug and add to the code base due to time constraints. If limiting to libusb simplifies go for it. If I ever make it work on Windows, note that I must support Win 7 and 10 and have a signed driver.

claunia, 2018-09-10:
for our project we need it to be completely multiplatform (linux + mono, mac + xamarinmac, windows + wpf). as long as it works like that, I don't care.

qmfrederik, 2018-09-10:
@AshleyRoll You'll need libusb-1.0.dll as a userspace library. It will ship as part of the NuGet package. You won't need to create an INF file or register your own driver for the device.

AshleyRoll, 2018-09-10:
@qmfrederik great! thanks for the clarifications. Then I think that would be fine for my application.

MelbourneDeveloper, 2019-01-03:
@andrasfuchs, have you looked at Device.Net? https://github.com/MelbourneDeveloper/Device.Net/wiki

The design of this library uses a single C# interface across Android, UWP, and Windows that is Task Based Async: https://github.com/MelbourneDeveloper/Device.Net/wiki/IDevice. It can also wrap LibUsbDotNet: https://github.com/MelbourneDeveloper/Trezor.Net/blob/LibUsbUnitTest/src/Trezor.Net.LibUsbUnitTest/LibUsbDevice.cs. Also, it switches between USB, and Hid seamlessly like this:

        private static async Task InitializeTrezor()
        {
            //Register the factory for creating Usb devices. This only needs to be done once.
            UWPUsbDeviceFactory.Register();

            //Register the factory for creating Usb devices. This only needs to be done once.
            UWPHidDeviceFactory.Register();

            //Note: other custom device types could be added here

            //Define the types of devices to search for. This particular device can be connected to via USB, or Hid
            var deviceDefinitions = new List<DeviceDefinition>
            {
                new DeviceDefinition{ DeviceType= DeviceType.Hid, VendorId= 0x534C, ProductId=0x0001, Label="Trezor One Firmware 1.6.x" },
                new DeviceDefinition{ DeviceType= DeviceType.Usb, VendorId= 0x1209, ProductId=0x53C1, ReadBufferSize=64, WriteBufferSize=64, Label="Trezor One Firmware 1.7.x" },
                new DeviceDefinition{ DeviceType= DeviceType.Usb, VendorId= 0x1209, ProductId=0x53C0, ReadBufferSize=64, WriteBufferSize=64, Label="Model T" }
            };

            //Get the first available device and connect to it
            var devices = await DeviceManager.Current.GetDevices(deviceDefinitions);
            var trezorDevice = devices.FirstOrDefault();
            await trezorDevice.InitializeAsync();

            //Create a buffer with 3 bytes (initialize)
            var buffer = new byte[64];
            buffer[0] = 0x3f;
            buffer[1] = 0x23;
            buffer[2] = 0x23;

            //Write the data to the device
            await trezorDevice.WriteAsync(buffer);

            //Read the response
            var readBuffer = await trezorDevice.ReadAsync();
        }

I'd be keen to offer some input on refactoring ideas if you are interested...

mcuee, 2021-06-10:
Just want to say thanks to Frederik Carlier for taking over libusbdotnet project from Travis Robinson and update to its current stage which seems to work very well based on my testing.

closed 2021-06-11 by mcuee