LibUsbDotNet/LibUsbDotNet #6: LibUsbDotNet NuGet package

andrasfuchs, 2017-02-01:
Since this repo is the successor of LibUsbDotNet appointed by Travis, I think you should discuss with GeorgeHahn, the owner of the current LibUsbDotNet NuGet package (https://www.nuget.org/packages/LibUsbDotNet), to let you manage it from now on.

qmfrederik, 2017-02-01:
Hi @GeorgeHahn !

We're bringing some new life in the LibUsbDotNet project (amongst other, .NET Core compatibility & reviewing the architecture).

Since you are the owner of the NuGet package for LibUsbDotNet and we're managing the code from here, would it be possible for you to add me as a maintainer to the NuGet package? You can use the quamotion handle on NuGet.

Thanks!

GeorgeHahn, 2017-02-01:
Good to see this library getting some care! All the best!

qmfrederik, 2017-02-01:
@GeorgeHahn Thanks!

owenkimbrell, 2017-03-18:
Still cant get the package to work. Thoughts?

qmfrederik, 2017-04-07:
@owenkimbrell Can you open a separate issue with any problem you have getting the package to work?

closed 2017-04-07 by qmfrederik