LibUsbDotNet/LibUsbDotNet #8: Update for the "old" 2.2.8 release package on NuGet.org

chrfin, 2017-03-27:
Hi,

We are using LibUsbDotNet in our project but we needed the two fixes made after the 2.2.8 release. As we have a local ProGet for internal packages anyhow I created a new LibUsbDotNet build from the latest sources some time ago and just uploaded that to our local feed as version 2.2.8.500 so it updates nicely to an "official version 2.2.9+". But now we would like to upload our projects packages to NuGet.org for others to use, but that fails because there is no LibUsbDotNet 2.2.8.500...
Would you maybe consider adding my new package as an update to the currently released package? 
[LibUsbDotNet.2.2.8.500.nupkg.zip](https://github.com/LibUsbDotNet/LibUsbDotNet/files/872184/LibUsbDotNet.2.2.8.500.nupkg.zip)
Otherwise we would be forced to upload it as a new package which would lead to two/three LibUsbDotNet on NuGet.org, which I would like to avoid...

regrads
Christoph

qmfrederik, 2017-04-07:
Hi @chrfin 

Sorry it took so long. Can you try https://www.nuget.org/packages/LibUsbDotNet/2.2.9-r9 and let me know if that package works for you? I can promote it to a released version, but wanted to get your confirmation first everything is OK.

Thanks!

chrfin, 2017-04-08:
Thanks for the package - right in time, as we plan to release mid of next week.  
I will try it out on monday and report back right away...

chrfin, 2017-04-08:
Ok, checked it out right away. Assembly looks great, but the license is set to GPL. Could you change that to LGPL (as we have non-GPL compatible dependencies too and both are valid for LibUsbDotNet) and then make it a released version (maybe 2.2.9.114 as it is the state of commit 114?)?
THANKS

chrfin, 2017-04-11:
Any chance you could create the release-package today? We are planning to release tomorrow...
Thanks again!

qmfrederik, 2018-09-25:
This has (finally) been fixed, [version 2.2.28](https://www.nuget.org/packages/LibUsbDotNet/2.2.28) should now correctly mention the license is LGPL.

closed 2018-09-25 by qmfrederik