LibUsbDotNet/LibUsbDotNet #10: VendorID&ProductID out of range

mo10, 2017-05-03:
Hi,

I use "Info.Descriptor.VendorID" got a wrong value.
Info.Descriptor.VendorID and Info.Descriptor.VendorID uses an **short** type,Its maximum range is 32767,but my usb device ProductID is 0xEFE8(61416).
61416 over the short type range.
How should I get the right ProductID value?

Thanks!

cristianst85, 2017-06-01:
Hi @mo10,

I had a similar issue when needed to find a device that had a PID value (0xca01) that exceeded the **short** type range and I've used this workaround instead:

`profile.DeviceDescriptor.ProductID == unchecked((short)0xca01))`

mo10, 2017-06-01:
Thank you, @cristianst85 

I found another way to solve the problem,  `UsbRegistry.Pid` and `UsbRegistry.Vid` can get the right value:
`public virtual int Pid { get; }`
and
`public virtual int Vid { get; }`

See:http://libusbdotnet.sourceforge.net/V2/html/a432caef-6b30-6fd2-3475-146c30266188.htm

mcuee, 2021-06-11:
I will re-open this as this seems to be a common question.

mcuee, 2023-04-23:
This should be fixed by #215.

PR #215 (2023-04-23): vid/pid should be ushort and not short
This should fix #10 and #91.
closed 2023-04-23 by mcuee