LibUsbDotNet/LibUsbDotNet #17: Is there a way to subscribe to system-wide USB device notifications under .NET Core 2.0?

TAGC, 2017-10-04:
I've been using LibUsbDotNet for cross-platform USB communication under .NET Core for a while now, and it's worked well.

However, I'm thinking about splitting up my applications into self-healing microservices, and for a microservice responsible for USB communication I imagine it would be helpful to be able to subscribe to system-wide USB events (like device reconnections) to be able to judge when to re-attempt communication with a device if communications drop.

Unfortunately, I'm not able to see `DeviceNotifier` under the `LibUsbDotNet.DeviceNotify` namespace and I guess this is because it's conditionally compiled out within the NuGet packages I reference. Is there a workaround?

TAGC, 2017-10-30:
`LinuxDeviceNotifier` is available and can be used for monitoring USB system events when the .NET application is running on Linux. For Windows I created a custom "DummyDeviceNotifier" that implements `IDeviceNotifier` but essentially does nothing (i.e. Null Object Pattern).

mcuee, 2021-06-13:
Please refer to #32 as well.

mcuee, 2021-06-25:
I will close this one and please use #32 for further discussions.

closed 2021-06-25 by mcuee