LibUsbDotNet/LibUsbDotNet #18: DeviceNotifyEventArgs cannot be properly subclassed

TAGC, 2017-10-05:
Not sure why it's been implemented this way, but [`IDeviceNotifier`](https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/stage/LibUsbDotNet/DeviceNotify/IDeviceNotifier.cs) is declared as a public interface, yet the [`DeviceNotifyEventArgs`](https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/stage/LibUsbDotNet/DeviceNotify/DeviceNotifyEventArgs.cs) class it references is abstract and there's no way to properly subclass it as all fields are only accessible internally.

I'm trying to write unit tests for a class that depends on an injected `IDeviceNotifier` dependency and should perform certain behaviour when the notifier raises events showing a USB device has been added or removed, yet because of this weird implementation of `DeviceNotifyEventArgs` I can neither directly instantiate `DeviceNotifyEventArgs` nor subclass it and set the event arg properties to what I want.

TAGC, 2017-10-05:
Working around it like this:

```cs
private class MockDeviceNotifyEventArgs : DeviceNotifyEventArgs
{
    /// <inheritdoc />
    public MockDeviceNotifyEventArgs(EventType eventType, int vendorId, int productId)
    {
        var device = new MockUsbDeviceNotifyInfo { IdVendor = vendorId, IdProduct = productId };
        var baseType = typeof(DeviceNotifyEventArgs);
        const BindingFlags bindingFlags = BindingFlags.NonPublic | BindingFlags.Instance;

        baseType.GetField("mDevice", bindingFlags).SetValue(this, device);
        baseType.GetField("mEventType", bindingFlags).SetValue(this, eventType);
    }
}
```

qmfrederik, 2017-11-14:
@TAGC Care to send over a PR? 😄

closed 2021-06-11 by mcuee