LibUsbDotNet/LibUsbDotNet #21: Consider improving version naming scheme

TAGC, 2017-10-06:
This is just a heads up.

The latest pre-release version for LibUsbDotNet right now is v2.2.0-r9. If you release v2.2.0-r10, it's going to cause problems for people as [SemVer 2.0.0](http://semver.org/) uses lexical comparison of identifiers that contain non-numeric characters.

This means that v2.2.0-r10 actually has *lower* precedence than v2.2.0-r9:

[image]

The result will be if you push a package with that version, people won't see it show up as an update in Visual Studio and if they *do* update to it, VS will actually show the older r9 package as an "update".

The standard convention for .NET package versioning is to use 4 digit zero-padding, so like v2.2.0-r0009, v2.2.0-r0010, etc.

[image]

qmfrederik, 2017-10-09:
@TAGC Thanks for noticing this! I'll fix it. The number after the `r` is basically the AppVeyor build number; for other projects I increased it to a large number (say `1000`) to workaround this but I forgot to enable it for LibUsbDotNet.

qmfrederik, 2017-11-14:
Changed the version number to `2.2.10-r105`.

qmfrederik, 2018-09-06:
With #36, we're not using NerdBank.GitVersioning, which should create fully SemVer-compliant numbers 😄 .

closed 2018-09-06 by qmfrederik