LibUsbDotNet/LibUsbDotNet #22: LibUsbDotNet Bulk Transfer Usb Communication Win32Error

serhanyilmaz, 2017-10-10:
We have a vendor-specific device for USB connections.

We used LibUsbDotNet Bulk Transfer communication. In the first case, I did 64 byte data transfer. In the second case I made 1024 byte data transfer. In case of 64 byte transfer, there is usually no problem (Problems are rare). In case of 1024 byte transfer, there is a problem (almost always in USB 2.0 port, also rare in USB 3.0 port) and Finally, when doing USB Write, I get Win32Error error in LibUsbDotNet ErrorCode class.

What causes to this problem

seuzht, 2018-05-12:
Maybe your USBdriver does'not support 1024 bytes length.

qmfrederik, 2018-09-25:
I think this is a duplicate of #23. For this kind of issues, it's better to ask on StackOverflow.

closed 2018-09-25 by qmfrederik