LibUsbDotNet/LibUsbDotNet #24: LinuxDeviceNotifier "MonitorDevDirectory" mode does not seem to work

TAGC, 2017-10-30:
LibUsbDotNet contains two strategies for publishing notifications about USB system events when running under Linux - a "MonitorDevDirectory" strategy and a "PollDeviceList" fallback option.

I'm trying to build an application that responds to USB system events when running on Linux and I've noticed it always seems to fall back to the "PollDeviceList" strategy even though the host machines have a standard `/dev` directory (this is how I noticed #19).

I didn't give much thought to this as the fallback strategy seemed to work well enough. Unfortunately, I've run into problems with that strategy as well which seem like they could be avoided if the `MonitorDevDirectory` strategy were used.

It's hard to track exactly what's happening as this library doesn't have any logging and I can't easily debug it when it's running within a Docker container on my Raspberry Pi, but From [here](https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/stage/LibUsbDotNet/DeviceNotify/Linux/LinuxDeviceNotifier.cs#L48) it seems that `LinuxDeviceNotifier` will fall back from the "Monitor" strategy to the "Poll" strategy if `mLinuxDevItemList` is empty after `BuildDevList` returns.

Here's what looks odd to me: `BuildDevList` seems to be looking for USB device handles with this: `string[] deviceInterfaceFiles = Directory.GetFiles(mDevDir, "usbdev*", SearchOption.TopDirectoryOnly);`

Assuming that `mDevDir` is its default value `dev/`, this will match things like `/dev/usbdev*`. However, on typical Linux machines USB devices are enumerated under `/dev/bus/usb/...`, so I don't get where that comes from.

TAGC, 2017-10-30:
Oh I see, apparently a really long time ago they used to `/dev/usbdev*.*` format for these things.

mcuee, 2021-06-11:
Please also refer to #32.

mcuee, 2021-06-25:
I believe the current Linux device notifier was a quick one by Travis long time ago. The proper way is to use libudev or use libusb hotplug API (which uses libudev under Linux). Please refer to #32 for further discussions.

closed 2021-06-25 by mcuee