LibUsbDotNet/LibUsbDotNet #29: DllNotFoundException on dotnet core 2.0 on Debian stretch x64

MaitreDede, 2017-11-21:
Hi,
I am running Debian Stretch x64, with dotnet core sdk 2.0.3. I have installed the package `libusb-1.0.0`. I am using `LibUsbDotNet` version `2.2.10-r110`.

When I launch my app (to connect to my device), I get a `System.DllNotFoundException` :
```
Unhandled Exception: System.DllNotFoundException: libusb-1.0 library not found.  This is often an indication that libusb-1.0 was installed to '/usr/local/lib' and mono.net is not looking for it there. To resolve this, add the path '/usr/local/lib' to '/etc/ld.so.conf' and run 'ldconfig' as root. (http://www.mono-project.com/DllNotFoundException) ---> System.DllNotFoundException: Unable to load DLL 'libusb-1.0.dll': The specified module or one of its dependencies could not be found.
 (Exception from HRESULT: 0x8007007E)
   at MonoLibUsb.MonoUsbApi.Init(IntPtr& pContext)
   at MonoLibUsb.MonoUsbSessionHandle..ctor()
   --- End of inner exception stack trace ---
   at MonoLibUsb.MonoUsbSessionHandle..ctor()
   at MonoLibUsb.MonoUsbEventHandler.Init(UnixNativeTimeval unixNativeTimeval)
   at MonoLibUsb.MonoUsbApi.InitAndStart()
   at LibUsbDotNet.LudnMonoLibUsb.MonoUsbDevice.get_MonoUsbDeviceList()
   at LibUsbDotNet.Main.LegacyUsbRegistry.get_DeviceList()
   at LibUsbDotNet.UsbDevice.get_AllLibUsbDevices()
   at LibUsbDotNet.UsbDevice.get_AllDevices()
   at LibUsbDotNet.UsbDevice.OpenUsbDevice(Predicate`1 findDevicePredicate)
   at PrintConnectors.UsbPrintConnector.Open(Int32 vendorId, Int32 productId, String profile)
```

Code is :
```
UsbDeviceFinder finder = new UsbDeviceFinder(vendorId, productId);
UsbDevice device = UsbDevice.OpenUsbDevice(finder);
```

I have found as :
```
/usr/lib/x86_64-linux-gnu/libusb.so
/lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0
/lib/x86_64-linux-gnu/libusb-1.0.so.0
```
Same problem if forcing library path :
`LD_LIBRARY_PATH=/lib/x86_64-linux-gnu/ dotnet run`

qmfrederik, 2017-11-21:
The problem is [that `MonoLibUsbApi` hard-codes the library name as `libusb-1.0.dll`](https://github.com/LibUsbDotNet/LibUsbDotNet/blob/d9e63572aee1c76d858b737cc74807c915eff58f/stage/LibUsbDotNet/MonoLibUsb/MonoLibUsbApi.cs#L36). So it'll end up looking for a `.dll` file instead of `.so` file, even on Linux.

If you're compiling from source, you should be able to change that line to:

```csharp
internal const string LIBUSB_DLL = "libusb-1.0";
```

or

```csharp
internal const string LIBUSB_DLL = "usb-1.0";
```

to get it working.

Can you give that a try and let me know if it works for you?

MaitreDede, 2017-11-21:
I made a simple poc :
- `LIB = "libusb-1.0"` = have to make symbolic link to `/lib/x86_64-linux-gnu/libusb-1.0.so.0` to work.
- `LIB = "usb-1.0"` = DllNotFoundException
- `LIB = "libusb-1.0.so.0"` = works (but not portable)

qmfrederik, 2017-11-21:
Yeah, `libusb-1.0` seems like the lesser evil at this point. If you open a PR, I'd be happy to merge it.

MaitreDede, 2017-11-22:
Also, when merged, if you could publish a new nuget please ? Thanks :smile:

qmfrederik, 2017-11-22:
Thanks, just merged the PR. The AppVeyor build should automatically push a new version to NuGet when it completes, let me know if this is not the case.

qmfrederik, 2018-09-06:
@MaitreDede It's been a while :). I think this issue was fixed with #31, hope it works. Let me know if otherwise.

closed 2018-09-06 by qmfrederik