LibUsbDotNet/LibUsbDotNet #30: Unable to retrieve Device Descriptor on ARM cortex-a9 platform

AshleyRoll, 2017-11-21:
We have a minimal install of mono running on a Freescale i.MX6 platform. This was build with `--enable-minimal=aot,profiler,debug,logging --enable-small-config=yes`. The mono platform is able to run several test applications, but I am unable to get the LibUSBDotNet based code working.

We have a test application using LibUSBDotNet working under Windows (WinUSB) and on Ubuntu, however when we attempt to run this on the i.MX6 platform, it fails to enumerate any devices.

LibUsb-1.0 is installed and working on the platform, the `lsusb` application is able to identify our device and successfully probe its device descriptors, interfaces and endpoints. It is also able to retrieve various strings from the device.

In trying to identify the problem, I used the MonoLibUsb ShowInfo sample code to do low level calls:

```C#
private static MonoUsbSessionHandle __sessionHandle;
public static MonoUsbSessionHandle Session
{
	get
	{
		if (ReferenceEquals(__sessionHandle, null))
			__sessionHandle = new MonoUsbSessionHandle();
		return __sessionHandle;
	}
}
public static void Main(string[] args)
{
	int ret;
	MonoUsbProfileList profileList = null;
	
	// Initialize the context.
	if (Session.IsInvalid)
		throw new Exception("Failed to initialize context.");

	MonoUsbApi.SetDebug(Session, 3);
	// Create a MonoUsbProfileList instance.
	profileList = new MonoUsbProfileList();

	// The list is initially empty.
	// Each time refresh is called the list contents are updated. 
	ret = profileList.Refresh(Session);
	if (ret < 0) throw new Exception("Failed to retrieve device list.");
	Console.WriteLine("{0} device(s) found.", ret);

	// Iterate through the profile list; write the device descriptor to
	// console output.
	foreach (MonoUsbProfile profile in profileList)
	{
		Console.WriteLine("Bus: {0}, Device: {1}", profile.BusNumber, profile.DeviceAddress);
		Console.WriteLine(profile.DeviceDescriptor);
	}

	// Since profile list, profiles, and sessions use safe handles the
	// code below is not required but it is considered good programming
	// to explicitly free and close these handle when they are no longer
	// in-use.
	profileList.Close();
	Session.Close();
}
```
I added an extra line of output to verify the bus and device addresses are correct, and set the debug level to 3, however I get no debugging output.

The output (for each device) is:
```
root@OpenWrt:~/usb# mono LibUSBTest.exe
7 device(s) found.
Bus: 1, Device: 5
Length:18
DescriptorType:Device
BcdUsb:0x0000
Class:PerInterface
SubClass:0x00
Protocol:0x00
MaxPacketSize0:0
VendorID:0x0000
ProductID:0x0000
BcdDevice:0x0000
ManufacturerStringIndex:0
ProductStringIndex:0
SerialStringIndex:0
ConfigurationCount:0

... etc ...
```

In the `MonoUsbProfile` constructor, the calls the to 

```C#
mBusNumber = MonoUsbApi.GetBusNumber(mMonoUSBProfileHandle);
mDeviceAddress = MonoUsbApi.GetDeviceAddress(mMonoUSBProfileHandle);
```

are correctly enumerating the bus and device numbers. I have compared this to the list from `lsusb`.

However the call to 
```C#
GetDeviceDescriptor(out mMonoUsbDeviceDescriptor);
``` 
does not seem to be filling the MonoUsbDeviceDescriptor with data, it all remains zero.

From this I believe that the call to `MonoUsbApi.GetDeviceDescriptor()` is not correctly marshalling the out data here:

```C#
[DllImport(LIBUSB_DLL, CallingConvention = CC, SetLastError = false, EntryPoint = "libusb_get_device_descriptor")]
        public static extern int GetDeviceDescriptor([In] MonoUsbProfileHandle deviceProfileHandle, [Out] MonoUsbDeviceDescriptor deviceDescriptor);
```

Has anyone else run up against this issue? Any clues? Are there mono build options that I need to reconfigure?

qmfrederik, 2017-11-21:
Out of curiosity, did you try to run this using .NET Core instead of Mono?

AshleyRoll, 2017-11-21:
@qmfrederik Unfortunately I don't have .NET Core running on the platform, and it isn't something I will get time to look at. 

It may be our particular setup and build environment causing issues with P/Invoke. It all works fine under Ubuntu etc.. we were running an older build of mono (5.4.0.56) but I updated it to 5.4.1.6 today and still get the same issue. 

Feel free to close this out if there isn't really much you can do. 

Thanks for your time.

claunia, 2018-09-06:
I’m having the exact same problem in mono.

qmfrederik, 2018-09-06:
@claunia It's a small world 😄 .

I'll probably resume some work on LibUsbDotNet in the coming weeks, hope to have time to look at this too.

claunia, 2018-09-06:
I have sent a PR from my organization (as I’ve done it in work hours :p) that seems to fix it in Linux/mono. And yes it’s a small world xD

mandawah, 2018-09-21:
I have the exact same issue, here. Unfortunately @claunia PR doesn't help.

I identified that it worked fine at commit 03d3f6a26db68f6bef29c51fdc413399145c5c58 , but commit 0dd6a3811aca90a07dcf8b4ef54c2a7d5b091dc1 (July 11, 2017) broke mono, and it remained broken since then.

I also tested with Ubuntu x86_64 mono : "Example/Show.Info" correctly reports devices before 0dd6a3811aca90a07dcf8b4ef54c2a7d5b091dc1  but fail to report any device starting from this commit.

As far as I can tell the problem is refactoring MonoUsbDeviceDescriptor from a single class that did not use inheritance to a class that inherits from UsbDeviceDescriptorBase.

I tried all possible combination of playing with the "Pack" value or other tricks, but the only things that worked is to simply revert to the monolithic MonoUsbDeviceDescriptor . Reverting MonoUsbDeviceDescriptor.cs file from 03d3f6a26 into 0dd6a381 lets Example/Show.Info display devices again (both x86_64 and ARM).

It seems that mono is not capable of properly marshaling classes that uses inheritance.  I suppose the inheritance that is used for all Mono*Descriptor breaks LibUsbDotNet for mono everywhere one of those descriptor is marshaled for P/Invoke. 

By the way, in my opinion, and as far as I understand the goal of it, the use of "#if WIN" and "#if LINUX" in NativeMethods.cs is not a good idea and probably adds to the problem described here.

I think that the same LibUsbDotNet dll shall work for mono whatever the run-time OS is. Therefore CallingConvention, library name, or any other parameters, should be determined at run-time, not at build-time.

qmfrederik, 2018-09-21:
@mandawah Thanks for looking into this.

0dd6a38 was an attempt to extract common interfaces for the different back-ends that LibUsbDotNet uses (especially on Windows, in interfaces with various backends).

libusb-1.0 already provides an abstraction layer for different backends and we've come to realize there's little point in duplicating that effort in LibUsbDotNet.

I'm working to simplify LibUsbDotNet and reduce it to a "light" interface on top of libusb-1.0. It turned to be out a bit of a rabbit hole which landed me in libusb-1.0, libusbK and libusb-win32 territory.

Long story short, if you look at master, there's work being done to:
1. Automatically generate the P/Invoke code based on the C source code. This means that the struct/class inheritance is gone, and the problem reported here should be fixed.
2. Add (very basic) CI system to at least excercise some P/Invoke methods, so that we can catch this kind of issues

Related to CallingConvention, library name,... , these are compile-time constants in .NET so there are certain limitations there.

mandawah, 2018-09-21:
@qmfrederik Thank you for your message. I didn't state it in my previous post in order not to be to verbose, but I obviously thank you for all your efforts toward improving and maintaining LibUsbDotNet.

I understand the reason why you did what you did starting from commit 0dd6a38 . I have little objections regarding the purpose of it. It's just that it seems that mono is not happy with it. I don't know why, this is purely an empirical observation.

Those ongoing efforts seems to be going in the very good direction as far as I can judge, thank you very much.

Regarding CallingConvention etc. : yes I know. My point/idea was that it would be better to have all 3 variants (Windows, Linux, OSX) in, say, 3 different static classes, all built into the dll, without any #if.  

Then at run-time, invoke a mechanism to select which implementation to use. This can be done in several ways, including setting a (static or not) property with the correct implementation of an interface. Those implementations could be singletons that simple call the correspoding static P/Invoke functions (either the Windows, or the Linux ones, etc.)

I though about submitting a PR going in this direction, but the fact I was not able to have something working correctly on mono without undoing all these inheritance changes started with 0dd6a38 stopped me to jump in...

qmfrederik, 2018-09-21:
Having 3 copies of the P/Invoke methods may be doable now that the P/Invoke code is generated dynamically. Another option is to compile for the 3 platforms and use the 'bait and switch' approach used by .NET Core.

Regarding the PR: Yes, the codebase is in a state of flux right now so it's an uphill battle to generate a PR now. It will probably take a week or so to get something stable. I'll update this thread when I get there - that would be a good time to test things out and see whether the changes indeed went in the right direction :).

mandawah, 2018-09-21:
OK, perfect. I'll wait for your update. Thanks.

qmfrederik, 2018-09-25:
Although I think the libusb-related refactoring is the correct way to implement this, it will
a) take some time and 
b) change the shape of the LibUsbDotNet API in a significant way.

Therefore, I think it's best to do that work in a "v3" release of LibUsbDotNet.

For this issue (and other minor issues which may pop up in meanwhile), I've created a new v2 branch which excludes 0dd6a38 and related commits. 

I've released [2.2.27](https://www.nuget.org/packages/LibUsbDotNet/2.2.27) to NuGet which should fix this issue.

Can you give that version a try and let me know if that works?

mandawah, 2018-09-25:
Hello Frederik,

Thank you very much!

Yes with pleasure, I'll test it. I'll come back to you asap.

mandawah, 2018-09-25:
By the way: would you mind signing the assembly you put in the NuGet package?

qmfrederik, 2018-09-25:
@mandawah Version 2.2.29 on NuGet should be strongly-named (see 5dd91a2fda393cf11db2072f2b45c3aee2750388).

mandawah, 2018-09-25:
@qmfrederik : OK, great! I cannot use the non-signed version in my full application,

mandawah, 2018-09-25:
@qmfrederik First class delivery, thank you!

closed 2021-06-11 by mcuee