LibUsbDotNet/LibUsbDotNet #32: DeviceNotifier with .NetCore and using .NetStandard libs

Canomation, 2018-04-04:
On NetFramework 4.6.1 the DeviceNotifier is avialable, on .NetCore 2.0 using the NetStandard Libraries the DeviceNotifier is not avialable.  

Is the there a workaround possible to implemed these feature or what is the reason for these behavior?

Best reagards

Martin Rottmann

BendRocks, 2018-06-14:
Ditto for me too...Any updates here? Thanks very much

metanoic, 2018-06-28:
Looking at the code, those parts are wrapped in conditionals that check the target framework.  Part of the issue is that the device notification in windows happens via the os sending a message to a window handle, which is squarely in WinForms territory - NetStandard is lower level than that so you wouldn't be able to reference that assembly even if the code wasn't conditionally excluded.  

I don't think there's much the maintainers of this project can do with that.  If this project is going to support .NetStandard then it can't concern itself with System.Windows.Forms, the application developer would need to implement the notification in their own project when targeting Windows.

qmfrederik, 2018-09-25:
@metanoic is correct, the device notifiers depend on Windows Forms and that's something that doesn't fly well in a cross-platform project.

The "correct" way to fix this would be to rely on hotplug support in libusb. Unfortunately, hotplug on Windows isn't a thing yet - see https://github.com/libusb/libusb/issues/86 .

It's unlikely this will get fixed in LibUsbDotNet until https://github.com/libusb/libusb/issues/86 is implemented.

mcuee, 2021-06-13:
There are many issues related to this one. 

libusb-1.0 Windows backend is lagging behind and hotplug will take time to be implemented under Windows. Maybe one way is to use libusb-1.0 hotplug for Linux and macOS to improve DeviceNotifier under Linux and macOS.

mcuee, 2023-04-18:
> @metanoic is correct, the device notifiers depend on Windows Forms and that's something that doesn't fly well in a cross-platform project.
> 
> The "correct" way to fix this would be to rely on hotplug support in libusb. Unfortunately, hotplug on Windows isn't a thing yet - see [libusb/libusb#86](https://github.com/libusb/libusb/issues/86) .
> 
> It's unlikely this will get fixed in LibUsbDotNet until [libusb/libusb#86](https://github.com/libusb/libusb/issues/86) is implemented.

It is unlikely https://github.com/libusb/libusb/issues/86 get sorted out any time soon as more than 4.5 years have passed after the above comment. So probably one potential solution is to keep using Windows Forms under Windows, and then libusb-1.0 hotplug under Linux and macOS.

mcuee, 2023-04-26:
For those who are interested, please try #216.
* #216

mcuee, 2023-10-07:
#216 has been merged into git HEAD.

closed 2023-10-07 by mcuee