LibUsbDotNet/LibUsbDotNet #42: LibUsbDotNet doesn't work

gaminn, 2018-09-10:
Hi,
my system is Win10, Visual Studio 2015, LibUsbDotNet nuget package used. I have USB device with 1 configuration and 3 interfaces (interface 0 is used for libusb with 0x81 interrupt IN endpoint, 0x02 interrupt OUT endpoint and 0x83 bulk IN endpoint). I use libusb-1.0.21 in my Qt/C++ app and everything works fine.

However with LibUsbDotNet I'm not able to communicate with my device. This is my code:

`                UsbDeviceFinder MyUsbFinder = new UsbDeviceFinder(0xABCD, 0xABCD);

                UsbDevice MyUsbDevice = UsbDevice.OpenUsbDevice(MyUsbFinder);

                IUsbDevice wholeUsbDevice = MyUsbDevice as IUsbDevice;
                wholeUsbDevice?.SetConfiguration(1);
                wholeUsbDevice?.ClaimInterface(0);

                UsbEndpointReader reader = MyUsbDevice.OpenEndpointReader(ReadEndpointID.Ep0);
                reader.DataReceived += (s, e) => {
                    Console.Write(e.Count);
                };
                reader.DataReceivedEnabled = true;`

With libusb-win32 (v1.2.6.0) or libusbK (v3.0.7.0) driver installed for interface 0 of my USB device I can't open the device (MyUsbDevice  == null), with WinUSB driver (v6.1.7600.16385) the program runs but I never receive any data using the reader. No matter what driver I use wholeUsbDevice is null.

Am I missing something?

qmfrederik, 2018-09-17:
In general, StackOverflow may be a better place for questions, requests for help,... .

gaminn, 2018-09-19:
I asked on stackoverflow https://stackoverflow.com/questions/52405085/libusbdotnet-doesnt-open-device . Now I'm even not able to open my device even though LibUsbDotNET sees it in UsbDevice.AllDevices . Any advice highly appreciated.

SimantoR, 2019-01-04:
In my knowledge, LibUsbDotNet for Windows would need the libusb in correct space and the `LibUsbDotNet_Setup.exe` should provide that. And I do agree that no one answers here. Maybe create a fork and make this work there?

SimantoR, 2019-01-04:
@gaminn so the program finds the dll if I'm not wrong? Just doesn't return the device you're looking for?

closed 2021-06-11 by mcuee