LibUsbDotNet/LibUsbDotNet #68: LibUsbDotNet 2.2.29 Device Notifier not working

KishoreGunalan, 2018-11-25:
Hi,
Currently I am developing a module to track the USB device status (device arrival, device removal) on Windows 7 environment using LibUsbDotNet 2.2.29. The solution was working fine and after a week I found that the custom Device Notifier is not tracking any USB device connect and disconnect.

Any suggestions / ideas would help.

Ledrunning, 2018-12-04:
Hi! I Have the same, so I downgraded nougat to 2.2.28 - and Notifier works great;
`public static IDeviceNotifier UsbDeviceNotifier = DeviceNotifier.OpenDeviceNotifier();`
` // Hook the device notifier event
            UsbDeviceNotifier.OnDeviceNotify += OnDeviceNotifyEvent;`

MelbourneDeveloper, 2019-01-30:
@kishoregunalan is this issue related?

https://github.com/LibUsbDotNet/LibUsbDotNet/issues/82

I'm just taking a guess but is the device notifier supposed to let you know when a device is connected?

mcuee, 2021-06-11:
Please also refer to here.
https://github.com/LibUsbDotNet/LibUsbDotNet/issues/32

mcuee, 2021-06-29:
I tried the latest v2 branch and it seems to work fine based on the example.

You can try out the snapshot here.
https://github.com/mcuee/LibUsbDotNet/releases

mcuee, 2023-04-23:
Close this for now. Please reopen if you still encounter issue from the v2 git head.

closed 2023-04-23 by mcuee