LibUsbDotNet/LibUsbDotNet #69: MonoApiError:MonoApiError:SubmitTransfer Ep 0x01  -1:ErrorIO:Input/output error

gqbgqb, 2018-11-30:
Hi all,

I'm developing a service using .Net core, LibUSBDotNet (2.2.29), and Linux (Ubuntu 18.04). Has anyone encountered this error when calling the: 

Line of code (using the sample code):
writer.Write(Encoding.Default.GetBytes(cmdLine), 2000, out bytesWritten); 

Error:
"MonoApiError:MonoApiError:SubmitTransfer Ep 0x01  -1:ErrorIO:Input/output error"

I've confirmed that I'm using the correct End point (using this command: sudo lsusb -v -d vendorid:productid). I've also given the correct permission for the device (using udev rules). 
However, someone from SO has suggested that its because of the multiple "End Point Description" returned by the device. He managed to solve it by removing 1 on the firmware (which I don't have access). 

Sample:
 Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

This library has been around for a while. I was hoping someone has more experience in implementing the library or can confirm that this is the correct behavior of the API.

Thanks ,
Glenn

YfavreUnige, 2019-07-01:
Hi Glenn,
I've got exactly the same issue on a MonoUsbEndpointWriter.
Did you solve the issue on your side and if yes could you share it ?
Thanks in advance ,
Yannick

codeoflin, 2019-10-30:
I've got exactly the same issue too

mcuee, 2021-06-10:
You have to know the communication protocol of the device in order to use libusb and libusbdotnet. This usually means that the device does not understand the commands sent to the device.

closed 2021-06-11 by mcuee

AsylbekA, 2024-11-05:
has anyone solved this problem? or is there an alternative option?