LibUsbDotNet/LibUsbDotNet #72: UsbContext Not Exposed

MelbourneDeveloper, 2018-12-09:
I downloaded the NuGet. UsbContext is not exposed:

[image]

MelbourneDeveloper, 2018-12-27:
It seems as though this is not necessary and the old sample is just a red herring.

jaimestuardo, 2019-01-08:
Hello... how have you solved it? I am facing the same problem.

ahmetgul93, 2019-06-18:
Hello, Could u please how u solve it ? I'm having the same error ?

blue00, 2020-02-28:
You can use UsbDevice.OpenUsbDevice instead:
```
// Todo: Use correct USB vendor and product ID.
UsbDeviceFinder usbDeviceFinder = new UsbDeviceFinder(0x0AA2, 0xB01);

// Todo: Dispose usbDevice.
usbDevice = UsbDevice.OpenUsbDevice(usbDeviceFinder);
```

mcuee, 2021-06-11:
Keep this open as a FAQ item.

closed 2021-06-11 by mcuee

mcuee, 2021-06-11:
Close this but keep as FAQ.

mcuee, 2021-06-11:
Please refer to #89.

MelbourneDeveloper, 2021-06-11:
@mcuee thanks for looking into all these items. I'm not able to do any investigation at the moment, but when I get time I will do a round of work to try to get Device .Net working with the latest LibUsbDotNet version.

It seems you have been working on it, but the only things I will need a working NuGet package, and working samples with the latest code. If that's all here, I won't need to bother you again in future.

mcuee, 2021-06-12:
You can try the latest nuget package which is 3.097 alpha now to see if it works for you or not.
https://www.nuget.org/packages/LibUsbDotNet/3.0.97-alpha

Frederik has quite some improvements recently so that I can run benchmarkcon (V2 branch only, not yet the V3 main branch). So I am happy to join the project to help on the testing and supporting side. I was also on the testing side last time when Travis Robinson started the project. Take note I am not the developer myself and I knew very little about dotnet, even though I know quite a bit about libusb as a non-developer admin of libusb project and libusb-win32/libusbk/pyusb project.

mcuee, 2021-06-12:
If you find any issues, please help to raise the issues in github. Thanks.