LibUsbDotNet/LibUsbDotNet #73: Can't Connect To Device

MelbourneDeveloper, 2018-12-09:
[image]

[image]

MelbourneDeveloper, 2018-12-09:
Same thing if I connect using the UsbContext

[image]

[image]

SpoinkyNL, 2018-12-10:
Getting the same issue unfortunately. Tried with the latest build 3.0.50

[image]

MelbourneDeveloper, 2018-12-10:
@SpoinkyNL , have you ever had the library working? Is there a version you could go back to that does work?

MelbourneDeveloper, 2018-12-10:
I just tried every single NuGet version, and none of them worked. At the same time, in the Read.Write sample there is a comment in that seems to indicate that I am doing something wrong:

> If this is a "whole" usb device (libusb-win32, linux libusb) it will have an IUsbDevice interface. If not (WinUSB) the variable will be null indicating this is an interface of a device.

If I read this correctly, the device I see in Device Manager is an "interface" of a device, and not the "whole" USB Device. What does that mean?

SpoinkyNL, 2018-12-10:
@MelbourneDeveloper never used this library before. What I did was download the artifact of the latest build (https://dev.azure.com/libusbdotnet/libusbdotnet/_build/results?buildId=147) and simply extract the library from the Nuget package (they're just ZIPs). 

Then I noticed if I enumerated context.List() I got a bunch of devices including the one I'm looking for. After that the library worked just fine following the Read/Write example

You can find the code I'm using here if it's any help: https://github.com/SpoinkyNL/SwitchArmyKnife/blob/master/src/SwitchArmyKnife.Core/Tinfoil/UsbInstaller.cs#L45

MelbourneDeveloper, 2018-12-10:
@qmfrederik I just downloaded the alpha and just get this:

[image]

qmfrederik, 2018-12-10:
Are you on .NET Core or the .NET Framework?

MelbourneDeveloper, 2018-12-10:
.NET Framework but can switch to Core.

MelbourneDeveloper, 2018-12-10:
When I drop the DLLs in the folder, I get this:

[image]

qmfrederik, 2018-12-10:
If you're on the .NET Framework, can you try this:
- Download the latest drop from https://dev.azure.com/libusbdotnet/libusbdotnet/_build/results?buildId=147&view=results
- Depending on the bitness of your project (32-bit or 64-bit), copy the dlls from `drop\runtimes\win-x86` or `drop\runtimes\win-x64` to your output folder

(The error you get, 0x8007000B, is when you have 32-bit dlls for a 64-bit .NET app and vice versa)

qmfrederik, 2018-12-10:
To fix this, we'll need a .target file in the NuGet package which copies the correct files to the output folder when building on .NET Framework. NET Core should handle this transparently.

https://github.com/libimobiledevice-win32/imobiledevice-net/blob/master/iMobileDevice-net/imobiledevice-net.targets has an example of what such a targets file can look like.

MelbourneDeveloper, 2018-12-10:
@qmfrederik actually, I downloaded the binary from the libusb site and it works fine with the 32 bit DLL. The old version was working with the 64bit DLL, but now it doesn't.

MelbourneDeveloper, 2018-12-10:
I finally see my device!!!

[image]

MelbourneDeveloper, 2018-12-10:
This is what I get:

[image]

MelbourneDeveloper, 2018-12-10:
I needed to open the device. I have moved forward:

[image]

Except that the data returned is empty... I don't know if that is the device or an issue with Libusb...

qmfrederik, 2018-12-10:
You should see libusb debug output on the console if this is a console application. You can also try to increase the log level by calling `context.SetDebugLevel(...)`, that may give you a bit of information about what's going on under the hoods.

MelbourneDeveloper, 2018-12-10:
> libusb: error [composite_submit_bulk_transfer] unable to match endpoint to an open interface - cancelling transfer

qmfrederik, 2018-12-10:
Did you claim the interface, via `UsbDevice.ClaimInterface`?

MelbourneDeveloper, 2018-12-10:
Tried, but don't know what number to put in there:

[image]

MelbourneDeveloper, 2018-12-10:
If I put the port number in there, I get this:

[image]

qmfrederik, 2018-12-10:
Yeah, this is where you'll need to know the specs of your USB device, unfortunately. Is there any useful output in the console?

MelbourneDeveloper, 2018-12-10:
Got it!!! The interface number is here. It works!!!!!

[image]

MelbourneDeveloper, 2018-12-10:
This code works for my case. You guys should probably fix up the samples because that's a major impediment for getting your head around this library.

      ```
      using (var context = new UsbContext())
            {
                context.SetDebugLevel(LogLevel.Info);

                var usbDeviceCollection = context.List();
                var switchUsbDevice = usbDeviceCollection.FirstOrDefault(d => d.ProductId == 0x0001 && d.VendorId == 0x534C);

                switchUsbDevice.Open();

                switchUsbDevice.ClaimInterface(0);

                var writeEndpoint = switchUsbDevice.OpenEndpointWriter(WriteEndpointID.Ep01);
                var readEnpoint = switchUsbDevice.OpenEndpointReader(ReadEndpointID.Ep01);

                var buffer = new byte[64];
                buffer[0] = 0x3f;
                buffer[1] = 0x23;
                buffer[2] = 0x23;

                writeEndpoint.Write(buffer, 3000, out var bytesWritten);

                var readBuffer = new byte[64];

                readEnpoint.Read(readBuffer, 3000, out var readBytes);
            }
```

FarzanehK, 2019-06-03:
@MelbourneDeveloper @SpoinkyNL @qmfrederik 
Thank you for sharing all your experience. 

I'm new in C# and I'm spending lots of time to read USB using LibUsbDotNet. Using "Read.Only" example (or Read.Write) , first I got the error "Device Not Found." After installing filters, the error is "No more bytes!". However, data is received from USB continuously as I checked it in another program. As here explained, I tried to replace :
"context = new UsbContext();
                var MyUsbDevice = context.List().FirstOrDefault(d => d.ProductId == 0x03EB && d.VendorId == 0x3110);
"
instead of 
"MyUsbDevice = UsbDevice.OpenUsbDevice(MyUsbFinder);"

But the "UsbContext" could not be found: (the type or namespace UsbContext could not be found...". 
I'm really confused. I've googled a lot and try to test all other people experiences, but I couldn't figure it out. I reallyyyyyy appreciate if you could help me and explain clearly how I should do. If it's possible, would you please share your project file completely? Thank you so much
(I even don't understand how I can save and use one of the examples in a different direction from installed folder. Although I installed LibUsbDotNet in NuGet, many commands are not defined!)

mcuee, 2021-06-12:
UsbContext() is only for main line (V3). For V2, you can refer to the examples in the V2 branch.

erfan132, 2022-01-21:
Hi i have a same issue 
i connect to device ok but when i try to read our write its going wrong 
my source
[image]
Operating System:Microsoft Windows NT 6.2.9200.0
LibUsbDotNet Version:2.2.8.104
LibUsb Kernel Type:NativeLibUsb
LibUsb Kernel Version:1.2.6.0

SerialNumber:15016037005E600569775180B81
Class Guid:a5dcbf10-6530-11d2-901f-00c04fb951ed
DeviceDesc:USB Mass Storage Device
HardwareId:
           USB\VID_13FE&PID_5500&REV_0100
           USB\VID_13FE&PID_5500
CompatibleIds:
              USB\Class_08&SubClass_06&Prot_50
              USB\Class_08&SubClass_06
              USB\Class_08
Class:USB
ClassGuid:{36fc9e60-c465-11cf-8056-444553540000}
Driver:{36fc9e60-c465-11cf-8056-444553540000}\0014
Mfg:Compatible USB storage device
FriendlyName:
LocationInformation:Port_#0001.Hub_#0005
PhysicalDeviceObjectName:\Device\USBPDO-6
BusTypeGuid:9d7debbc-c85d-11d1-9eb4-006008c3a19a
LegacyBusType:15
BusNumber:0
EnumeratorName:USB
Address:1
UiNumber:
InstallState:0
RemovalPolicy:3
SymbolicName:\??\USB#VID_13FE&PID_5500#15016037005E600569775180B81#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Have Any Idea for this?

mcuee, 2022-01-21:
Your issue is different. Please create a new issue. And most likely the reason is that your device does not uderstand the command you send. You have to know the USB Communication Protocol in order to communicate with the device.

hasann92, 2023-07-07:
I  have exactly the same issue @MelbourneDeveloper I tried to use your code but UsbContext cant be found.

mcuee, 2023-07-14:
> I have exactly the same issue @MelbourneDeveloper I tried to use your code but UsbContext cant be found.

Have you tried the latest git master branch?

hasann92, 2023-07-26:
I have the following code working in console app. However libusbdotne 3.0.102 does not work properly in a form application.  in the console app it returns all device information but in the form does not return device info(name, manufacturer etc. ) of the device. Besides, as far as I check it does not have eventhandler that we can catch plugged and unplugged event. Sometimes it  returns protected memory error once it lists all connected devices. Could you please tell me how to solve these problems?

using (var context = new UsbContext())
            {
                context.SetDebugLevel(LogLevel.Info);

                //Get a list of all connected devices
                var usbDeviceCollection = context.List();

                //Narrow down the device by vendor and pid
                var selectedDevice = usbDeviceCollection.FirstOrDefault(d => d.ProductId == ProductId && d.VendorId == VendorId);

                //Open the device
                selectedDevice.Open();

                //Get the first config number of the interface
                //selectedDevice.ClaimInterface(selectedDevice.Configs[0].Interfaces[0].Number);
                selectedDevice.ClaimInterface(0);

                //Open up the endpoints
                var writeEndpoint = selectedDevice.OpenEndpointWriter(WriteEndpointID.Ep01);
                var readEnpoint = selectedDevice.OpenEndpointReader(ReadEndpointID.Ep01);

                //Create a buffer with some data in it
                var buffer = new byte[64];
                buffer[0] = 0x05;
                buffer[1] = 0x00;
                buffer[2] = 0x00;
                buffer[3] = 0x00;

                //Write three bytes
                writeEndpoint.Write(buffer, 0, 4, 1000, out var bytesWritten);

                var readBuffer = new byte[64];

                //Read some data
                readEnpoint.Read(readBuffer, 0, 8, 1000, out var readBytes);
     

                Console.ReadLine();
            }

hasann92, 2023-10-13:
The issue is fixed after upgrading the project to .net7 framework

vigji, 2023-11-29:
I am trying to have this working in .NET 4.7.2, and I am also having a 
```
"UsbContext" could not be found: (the type or namespace UsbContext could not be found...".
```
error. Anyone's got any suggestion or workaround for this net framework?

hasann92, 2023-11-29:
I utilize the 3.0.102 alpha release for automatic detection, which does not require driver installation. Initially, it was necessary to have the 32-bit version of the libusb-1.0 DLL and place it in the appropriate debug directory (either x32 or x64). However, after upgrading the project to .NET 7, the use of an external DLL file is no longer required. here is the dll, i hope it is the correct one 
[libusb.zip](https://github.com/LibUsbDotNet/LibUsbDotNet/files/13502983/libusb.zip)