LibUsbDotNet/LibUsbDotNet #74: AllDevices Yields No Devices

MelbourneDeveloper, 2018-12-09:
[image]

seuzht, 2018-12-18:
I also met similar problem. But usually when I set a breakpoint to debug my code, the device (even if exists) would suddenly change to null.

MelbourneDeveloper, 2018-12-18:
@seuzht the NuGets are not in a working state. I had to grab an alpha version of a new release. Please follow the issue here: https://github.com/LibUsbDotNet/LibUsbDotNet/issues/73#issuecomment-445978402  . I have added a working sample in that thread.

raulhaag, 2018-12-19:
@MelbourneDeveloper  the example works well but i have problem to release all resources, it throws
```
 System.AccessViolationException: Intento de leer o escribir en la memoria protegida. A menudo, esto indica que hay otra memoria dañada.
   en LibUsbDotNet.NativeMethods.UnrefDevice(IntPtr dev)
   en LibUsbDotNet.Device.ReleaseHandle()
   en System.Runtime.InteropServices.SafeHandle.InternalFinalize()
   en System.Runtime.InteropServices.SafeHandle.Dispose(Boolean disposing)
   en System.Runtime.InteropServices.SafeHandle.Finalize()
```
at end or when retry. Any hint?  i tried 
```
usbDevice.ReleaseInterface(usbDevice.Configs[0].Interfaces[0].Number);
usbDevice.Close();
```

MelbourneDeveloper, 2018-12-20:
@raulhaag sorry I don't have any hints. The library has undergone some major changes. I think you will have to try to fix the bug and then submit a pull request.

MelbourneDeveloper, 2018-12-27:
@raulhaag , this seems like a separate bug. Do you want to document this up completely for the LibUsbDotNet team and create a new issue? I'd like to close this ticket seeing technically it is possible to connect.

raulhaag, 2018-12-28:
@MelbourneDeveloper no problem close it and i go to make a new issue more detailed and with a example proyect

closed 2018-12-28 by MelbourneDeveloper