LibUsbDotNet/LibUsbDotNet #78: Issue System.AccessViolationException on net 4+

raulhaag, 2018-12-28:
```
 System.AccessViolationException: Intento de leer o escribir en la memoria protegida. A menudo, esto indica que hay otra memoria dañada.
   en LibUsbDotNet.NativeMethods.UnrefDevice(IntPtr dev)
   en LibUsbDotNet.Device.ReleaseHandle()
   en System.Runtime.InteropServices.SafeHandle.InternalFinalize()
   en System.Runtime.InteropServices.SafeHandle.Dispose(Boolean disposing)
   en System.Runtime.InteropServices.SafeHandle.Finalize()
```
the exception come from 
```
LibUsbDotNet.dll!LibUsbDotNet.Device.ReleaseHandle() Línea 31
	en D:\a\1\s\src\LibUsbDotNet\Generated\Device.ReleaseHandle.cs(31)
````
and example project
[ErrorExample.zip](https://github.com/LibUsbDotNet/LibUsbDotNet/files/2715064/ErrorExample.zip)

it's like the lib try to free the same handler mora than one time, using the native library i have not problems.

MelbourneDeveloper, 2019-01-16:
@raulhaag , I might be having a similar issue. What I am finding is that cannot close the handle for a Usb device interface. I am calling CloseHandle, but it fails. I think the two problems are related. With LibUsbDotNet, my [device ](https://github.com/trezor/trezor-mcu/issues/444) always crashes on the second use because the handle has not been released. Please have a look at this: https://stackoverflow.com/questions/54189056/c-sharp-winusb-cant-call-closehandle-on-interface

fritz-net, 2020-04-15:
I also have the issue. My Application randomly crashes with System.AccessViolationException and no Stacktrace. Most of the time the debugger breaks at `reader.Read(...)`

mariopasquali, 2020-05-13:
I am trying to use the latest nuget (3.0.75-alpha) on Windows.  I get AccessViolationException when executing NativeMethods.GetDeviceList.  I cloned the github repo, same result.  I tried on both .net framework 4.6.1 and .net core 3.1.  The legacy V2 is fine, I am using it since many years. 

Is anyone using the V3 successfully on Windows?

scottpidzarko, 2020-05-24:
Using .NET core 3.1.0 with LibUsbDotNet 3.0.81-alpha.

Also having this issue. Crashes at random places, AccesViolationException occurs. When I did get an error with a stack trace I got something like:

[code]libusb: warning [libusb_exit] some libusb_devices were leaked
libusb: warning [libusb_exit] application left some devices open
Fatal error. System. AccessViolationEWxception: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
    at LibUsbDotNet.NativeMethods.UnrefDevice(IntPtr)
    at LibUsbDotNet.NativeMethods.UnrefDevice(IntPtr) 
    at LibUsbDotNet.Device.ReleaseHandle()
    at System.Runtime.InteropServices.SafeHandle.InternalRelease(Boolean)
    at System.Runtime.InteropServices.SafeHandle.Dispose(Boolean)
    at System.Runtime.InteropServices.SafeHandle.Finalize()
[/code]

And I'm pretty much going out back and shooting it when I'm done:

[code]
//Device is IUsbDevice
//writer is UsbEndpointwriter
//reader is UsbEndpointreader
if (Device != null)
{
    if (Device.IsOpen)
    {              
        bool result = Device.ReleaseInterface(Device.Configs[0].Interfaces[0].Number);
                    
        writer.Device.Close();
        reader.Device.Close();
        Device.Close();
        writer.Device.Dispose();
        reader.Device.Dispose();
        Device.Dispose();
        Device = null;
    }
}
[/code]

This much more reproducible if I repeatedly create, open, and close IUsbDevice, creating a new instance of the class each time. Something isn't tidying itself properly. I need this behavior where I'm creating a new instance of the class multiple times because I eventually want to use this as a packaged dll in a powershell script to read and write to a usb device, where the script might be run multiple times.

sarbos, 2021-02-11:
I am also experiencing this issue on 3.0.78-alpha. Is there a fix or workaround that anybody has found?

mcuee, 2021-06-10:
I was trying to reproduce with the first example "ErrorExample.zip" but I could not reproduce it.

Not so sure if this is similar to libusb issue here.
https://github.com/libusb/libusb/issues/703

ahmtmlh, 2021-06-30:
I had this issue recently, I don't know if that is supposed to happen or that is really a bug. However, I've found a workaround for my case. I was disposing the current UsbContext and creating a new one when specified device is not found by the application. 
I actually use a singleton UsbContext and call List() function everytime to refresh devices. This solved the problem for me. I am using a single UsbContext and multiple device handlers from that context.
Hope this helps

mcuee, 2021-07-01:
Another workaround is to use the V2 branch as per the report here.
https://github.com/LibUsbDotNet/LibUsbDotNet/tree/v2

neilenns, 2022-01-25:
I am also hitting this with the Alpha release on Windows. I'm using a stripped down version of the `Show.Info` example that only does `context.List()` to get VID, PID, bus number, and address for connected device. As soon as my method exits I get a memory access violation.

neilenns, 2022-01-25:
Ok I think I figured this out. When you call `UsbContext.List()` to get a list of devices you have to make sure you call `Dispose()` on that list before you return from your function.  Adding an extra `using` block makes things work without a crash:

```C#
            using (var context = new UsbContext())
            {
                using (var allDevices = context.List())
                {
                    foreach (UsbDevice deviceInfo in allDevices)
                    {
                           // Do something with the device info
                    }
                }
            }
```

SoftubeBask, 2022-01-25:
> Ok I think I figured this out. When you call `UsbContext.List()` to get a list of devices you have to make sure you call `Dispose()` on that list before you return from your function. Adding an extra `using` block makes things work without a crash:
> 
> ```cs
> using (var context = new UsbContext())
> {
>     using (var allDevices = context.List())
>     {
>         foreach (UsbDevice deviceInfo in allDevices)
>         {
>             // Do something with the device info
>         }
>     }
> }
> ```

This worked for me as well to get rid of the error. Had to use .Clone() to get the usbdevice from the list though.

HuoYingCunCunZhang, 2022-09-19:
`public class iUsbClient : ICommClient, IDisposable
    {

        
        private IUsbDevice _Client;
        private UsbEndpointWriter _UsbEndpointWriter;
        private UsbEndpointReader _UsbEndpointReader;
        /// <summary>
        /// 设备通讯上下文 很关键的字段，第三方库非托管资源释放不彻底的根源，需要将集合全部释放。否则会出现 
        /// Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
        /// 异常
        /// </summary>
        private IUsbContext _Context;
        /// <summary>
        /// 设备集合 很关键的字段，第三方库非托管资源释放不彻底的根源，需要将集合全部释放。否则会出现 
        /// Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
        /// 异常
        /// </summary>
        private UsbDeviceCollection _UsbClients;
        private bool _Disposed;
        ...
       private object _lock = new object();
        /// <summary>
        /// 显式调用资源回收
        /// </summary>
        /// <param name="disposing"></param>
        protected virtual void Dispose(bool disposing)
        {
            lock (_lock)
            {
                if (!_Disposed)
                {
                    _Disposed = true;
                    if (disposing)
                    {
                        _UsbEndpointReader = null;
                        _UsbEndpointWriter = null;
                    }
                    //注意释放非托管资源的的顺序，不能改变否则会出现意想不到的异常！
                    _Client?.Close();
                    _UsbClients?.Dispose();
                    _Context?.Dispose();
                }
            }
        }

    }
`

chenxuuu, 2022-11-10:
> Ok I think I figured this out. When you call `UsbContext.List()` to get a list of devices you have to make sure you call `Dispose()` on that list before you return from your function. Adding an [...]

it works! thanks😄

mcuee, 2023-04-23:
I will close this for now.

closed 2023-04-23 by mcuee

narcis-ro, 2024-10-02:
Also make sure UsbCoxtext is NOT disposed. If it is the error will happen after 10-15 seconds!
Also what others have said about the List() dispose as well (use Clone() on the device)