LibUsbDotNet/LibUsbDotNet #79: DeviceNotify support on macOS (or at least warn there is no support!)

aroman, 2019-01-03:
It seems that DeviceNotify only works on Windows or Linux (not macOS). If running on macOS, the library tries to load Windows-only DLLs, and can cause strange behavior (infinite hang in my case).

The library should at least detect if the OS is macOS/darwin and raise an error/log a warning to the user instead of falling back to broken behavior.

mcuee, 2021-06-10:
I believe this has never been implemented under macOS. On the other hand now that hotplug is available under libusb for Linux/MacOS (not yet on Windows), probably this can be improved.

mcuee, 2021-06-11:
Please also refer to this one.
https://github.com/LibUsbDotNet/LibUsbDotNet/issues/32

mcuee, 2021-06-25:
The proper way is to use libusb hotplug API. Please refer to #32 for further discussions.

closed 2021-06-25 by mcuee