LibUsbDotNet/LibUsbDotNet #80: AllDeices does not list USB thermal printer

jaimestuardo, 2019-01-08:
I am using LibUsbDotNet version 2.2.29 and when I call UsbDevice.AllDevices, it only lists a USB fingerprint device but not the USB printer. 

Unfortunately, the code here in GIT Hub is not the same as the one it is installed in Nuget Package Manager (version 2.2.29) so I am not able to debug the code.

Any help, please?

shurkin, 2020-10-26:
Hi @jaimestuardo,

Did you resolve it ? I have the same issue. I cannot find my USB thermal printer connected...

thank you

mcuee, 2021-06-11:
For Windows, you have to use libusbdotnet supported driver like WinUSB (recommended) or libusbk. If your printer does not use these drivers, you can not use libusbdotnet.

closed 2021-06-11 by mcuee