LibUsbDotNet/LibUsbDotNet #86: Message "Closed safe handle" throwed when runing and remove USB device

lee8871, 2019-04-15:
Hello,
There is a .net project using the dotnetLibUsb.dll. 
An message is throw form mscorlib.dll. ones I run the execute, open USB, and remove USB device(by unplug the cable) .  Errorcode can be captured , like this:

 ```
ErrorCode ec = srb_writer.Write(send_to_usb_buf, 0, send_len, 2000, out send_done_len);
                switch (ec)
                {
                    case ErrorCode.None:
                        DateTime t = DateTime.Now;
                        access.sendTime = t;
                        access.sendDone();
                        send_access_counter++;//发送成功了,转向下一个
                        break;
                    default:
                        reopenPort();
                        //throw new Exception(ec.ToString());
                        break;
                }
```
and I try to opend it again:

```
        internal void reopenPort()
        {
            ClosePort();
            scanDevice();
            selected_device = devicesDIC[selected_device_name].Device;
            if (!(selected_device.Open()))
            {
                throw new Exception(string.Format("open USB (port)device {0} fail", selected_device_name));
            }
            IUsbDevice wholeUsbDevice = selected_device as IUsbDevice;
            if (!ReferenceEquals(wholeUsbDevice, null))
            {
                // This is a "whole" USB device. Before it can be used, 
                // the desired configuration and interface must be selected.
                // Select config #1
                wholeUsbDevice.SetConfiguration(1);
                // Claim interface #0.
                wholeUsbDevice.ClaimInterface(0);
            }
            else
            {
                throw new Exception(string.Format("selected_device can not as IUsbDevice"));
            }
            srb_reader = selected_device.OpenEndpointReader((ReadEndpointID)(1 | 0x80), 1000, EndpointType.Interrupt);
            srb_writer = selected_device.OpenEndpointWriter((WriteEndpointID)2);
            // srb_reader.DataReceived += mEp_DataReceived;
            srb_reader.Flush();
        }

```

 ```
       internal void ClosePort()
        {
            lock (lock_access)
            {
                if (Is_opened())
                {
                    srb_reader.DataReceivedEnabled = false;
                    // srb_reader.DataReceived -= mEp_DataReceived;
                    
                    srb_reader.Dispose();
                    srb_reader = null;
                    srb_writer.Abort();
                    srb_writer.Dispose();
                    srb_writer = null;

                    IUsbDevice wholeUsbDevice = selected_device as IUsbDevice;
                    if (!ReferenceEquals(wholeUsbDevice, null))
                    {
                        // Release interface #0.
                        wholeUsbDevice.ReleaseInterface(0);
                        wholeUsbDevice.ReleaseInterface((1 | 0x80));
                        wholeUsbDevice.ReleaseInterface(2);
                        wholeUsbDevice.Close();
                    }
                    selected_device.Close();

                }
            }
        }

```

Then ,I can read datas from my device. It may run about 10 seconds. and Pull this message:

System.ObjectDisposedException
  HResult=0x80131622
  Message=已关闭(closed) Safe handle
  Source=mscorlib
  StackTrace:
   在 System.Runtime.InteropServices.SafeHandle.DangerousAddRef(Boolean& success)
   在 System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)

I have Tried more than 10 times. Is it something wring in the library?

mcuee, 2023-10-18:
#32 has been merged.

mcuee, 2023-10-18:
On the other hand, even without hotplug support, maybe there are still work-around which can be done to sort out this issue. Hopefylly someone can chime in and provide some ideas. Thanks.

mcuee, 2025-04-12:
Not so sure if this helps or not.
https://github.com/Jinjinov/Usb.Events