LibUsbDotNet/LibUsbDotNet #90: Can't write To Device

davweiss, 2019-07-07:
Hello, I can't write to a USB-Device.  
I got the following Error:
> [000055f4] libusb: error [hid_submit_bulk_transfer] unable to match endpoint to an open interface - cancelling transfer

Version: 3.0.48-alpha

I used the Code from [ReadWrite.cs](https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/src/Examples/Read.Write/ReadWrite.cs).
Screenshots:
[image]
[image]

The Logs are the following:

> from context.SetDebugLevel(LogLevel.Debug); 
> 
> ->      var usbDeviceCollection = context.List();
> [timestamp] [threadID] facility level [function call] <message>
> --------------------------------------------------------------------------------
> [ 0.133601] [000055f4] libusb: debug [libusb_get_device_list]
> [ 0.144419] [000055f4] libusb: debug [winusb_get_device_list] allocating new device for session [79]
> [ 0.144748] [000055f4] libusb: debug [winusb_get_device_list] allocating new device for session [7A]
> [ 0.145001] [000055f4] libusb: debug [winusb_get_device_list] allocating new device for session [7B]
> [ 0.145264] [000055f4] libusb: debug [winusb_get_device_list] allocating new device for session [7C]
> [ 0.145499] [000055f4] libusb: debug [winusb_get_device_list] allocating new device for session [7D]
> [ 0.145758] [000055f4] libusb: debug [winusb_get_device_list] allocating new device for session [7E]
> [ 0.146210] [000055f4] libusb: debug [get_api_type] driver(s): BTHUSB
> [ 0.146347] [000055f4] libusb: debug [winusb_get_device_list] allocating new device for session [18]
> [ 0.147339] [000055f4] libusb: debug [get_api_type] driver(s): usbccgp
> [ 0.147538] [000055f4] libusb: debug [get_api_type] matched driver name against Composite API
> [ 0.147641] [000055f4] libusb: debug [winusb_get_device_list] allocating new device for session [64]
> [ 0.147873] [000055f4] libusb: debug [get_api_type] driver(s): HidUsb
> [ 0.147985] [000055f4] libusb: debug [get_api_type] matched driver name against HID API
> [ 0.148071] [000055f4] libusb: debug [winusb_get_device_list] allocating new device for session [65]
> [ 0.148295] [000055f4] libusb: debug [get_api_type] driver(s): usbccgp
> [ 0.148410] [000055f4] libusb: debug [get_api_type] matched driver name against Composite API
> [ 0.148493] [000055f4] libusb: debug [winusb_get_device_list] allocating new device for session [66]
> [ 0.148707] [000055f4] libusb: debug [get_api_type] driver(s): usbccgp
> [ 0.148813] [000055f4] libusb: debug [get_api_type] matched driver name against Composite API
> [ 0.148896] [000055f4] libusb: debug [winusb_get_device_list] allocating new device for session [67]
> [ 0.149098] [000055f4] libusb: debug [get_api_type] driver(s): usbccgp
> [ 0.149204] [000055f4] libusb: debug [get_api_type] matched driver name against Composite API
> [ 0.149287] [000055f4] libusb: debug [winusb_get_device_list] allocating new device for session [68]
> [ 0.149837] [000055f4] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_15B6&SUBSYS_11112222&REV_00\6&1B4CA3C7&0&001000E0' bus number 1
> [ 0.151897] [000055f4] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_10DE&DEV_1AD6&SUBSYS_37111462&REV_A1\4&2DB3ECDA&0&0208' bus number 2
> [ 0.152184] [000055f4] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A12F&SUBSYS_50071458&REV_31\3&11583659&0&A0' bus number 3
> [ 0.154244] [000055f4] libusb: debug [winusb_get_device_list] found existing device for session [18]
> [ 0.154447] [000055f4] libusb: debug [init_device] found 1 configurations (active conf: 1)
> [ 0.154655] [000055f4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 177 bytes)
> [ 0.154906] [000055f4] libusb: debug [init_device] (bus: 3, addr: 8, depth: 2, port: 2): 'USB\VID_0A12&PID_0001\6&1D74254F&0&2'
> [ 0.155222] [000055f4] libusb: debug [winusb_get_device_list] found existing device for session [68]
> [ 0.155442] [000055f4] libusb: debug [init_device] found 1 configurations (active conf: 1)
> [ 0.155606] [000055f4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 84 bytes)
> [ 0.155821] [000055f4] libusb: debug [init_device] (bus: 3, addr: 7, depth: 1, port: 7): 'USB\VID_1038&PID_170E\5&146917D0&0&7'
> [ 0.156465] [000055f4] libusb: debug [winusb_get_device_list] found existing device for session [64]
> [ 0.156609] [000055f4] libusb: debug [init_device] found 1 configurations (active conf: 1)
> [ 0.156807] [000055f4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 289 bytes)
> [ 0.157048] [000055f4] libusb: debug [init_device] (bus: 3, addr: 2, depth: 1, port: 5): 'USB\VID_17A0&PID_0310\5&146917D0&0&5'
> [ 0.157330] [000055f4] libusb: debug [winusb_get_device_list] found existing device for session [67]
> [ 0.157479] [000055f4] libusb: debug [init_device] found 1 configurations (active conf: 1)
> [ 0.157578] [000055f4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 84 bytes)
> [ 0.157842] [000055f4] libusb: debug [init_device] (bus: 3, addr: 10, depth: 2, port: 1): 'USB\VID_1038&PID_1206\6&787D0C4&0&1'
> [ 0.158159] [000055f4] libusb: debug [winusb_get_device_list] found existing device for session [7B]
> [ 0.158308] [000055f4] libusb: debug [init_device] found 1 configurations (active conf: 1)
> [ 0.158497] [000055f4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 41 bytes)
> [ 0.158620] [000055f4] libusb: debug [init_device] (bus: 3, addr: 3, depth: 1, port: 6): 'USB\VID_05E3&PID_0610\5&146917D0&0&6'
> [ 0.159393] [000055f4] libusb: debug [winusb_get_device_list] found existing device for session [79]
> [ 0.159566] [000055f4] libusb: debug [init_device] found 1 configurations (active conf: 1)
> [ 0.159672] [000055f4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
> [ 0.159886] [000055f4] libusb: debug [init_device] (bus: 3, addr: 4, depth: 1, port: 8): 'USB\VID_05E3&PID_0608\5&146917D0&0&8'
> [ 0.160295] [000055f4] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\5&2942B015&0&0'
[... 150 lines cut ...]
> [ 0.446545] [000055f4] libusb: debug [libusb_alloc_transfer] transfer 000001A85EB8EB88
> [ 0.448771] [000055f4] libusb: debug [libusb_submit_transfer] transfer 000001A85EB8EB88
> [ 0.449605] [000055f4] libusb: debug [usbi_add_pollfd] add fd 2 events 1
> [ 0.451173] [000055f4] libusb: debug [hid_submit_bulk_transfer] matched endpoint 81 with interface 0
> [ 0.451702] [000055f4] libusb: debug [hid_submit_bulk_transfer] reading 65 bytes (report ID: 0x00)
> [ 0.452746] [000055f4] libusb: debug [libusb_get_next_timeout] next timeout in 2.996800s
> [ 0.454391] [000055f4] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
> [ 0.454928] [000055f4] libusb: debug [handle_events] poll fds modified, reallocating
> [ 0.455419] [000055f4] libusb: debug [handle_events] poll() 2 fds with timeout in 2997ms
> [ 3.453494] [000055f4] libusb: debug [handle_events] poll() returned 0
> [ 3.453783] [000055f4] libusb: debug [libusb_cancel_transfer] transfer 000001A85EB8EB88
> [ 3.455990] [000055f4] libusb: debug [hid_abort_transfers] will use interface 0
> [ 3.456594] [000055f4] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
> [ 3.457654] [000055f4] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
> [ 3.458220] [000055f4] libusb: debug [handle_events] poll() 2 fds with timeout in 60000ms
> [ 3.459097] [000055f4] libusb: debug [handle_events] poll() returned 1
> [ 3.459678] [000055f4] libusb: debug [windows_handle_events] checking fd 2 with revents = 0001
> [ 3.460625] [000055f4] libusb: debug [usbi_remove_pollfd] remove fd 2
> [ 3.461211] [000055f4] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 995, size 0
> [ 3.462855] [000055f4] libusb: debug [windows_transfer_callback] detected operation aborted
> [ 3.463574] [000055f4] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
> [ 3.465350] [000055f4] libusb: debug [usbi_handle_transfer_completion] transfer 000001A85EB8EB88 has callback 00007FFC3DF88EC0
> [ 3.469844] [000055f4] libusb: debug [sync_transfer_cb] actual_length=0
> [ 3.470354] [000055f4] libusb: debug [libusb_free_transfer] transfer 000001A85EB8EB88
> Buffer Read: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 Press Key to Exit!

mcuee, 2021-06-10:
If this is an HID device, try not use libusb and libusbdotnet. It may not work. In fact, HID device will only support control transfer and interrupt transfer. I am not so sure why there is this "hid_submit_bulk_transfer".

Try use HIDAPI and HIDAPI based dotnet binding if there are any.

closed 2021-06-11 by mcuee

mcuee, 2021-06-12:
Or you can use Device.Net.
https://github.com/MelbourneDeveloper/Device.Net