LibUsbDotNet/LibUsbDotNet #91: The type for VID & PID should not be "short"

andyhau, 2019-07-31:
Instead of using "**short**", they should be "**ushort**" in order to allow all possible VIDs and PIDs. Here's one example:

[image]

 
- **MonoLibUsbApi.cs**
```csharp
[DllImport(LIBUSB_DLL, CallingConvention = CC, SetLastError = false, EntryPoint = "libusb_open_device_with_vid_pid")]
private static extern IntPtr OpenDeviceWithVidPidInternal([In]MonoUsbSessionHandle sessionHandle, ushort vendorID, ushort productID);
```

- **MonoLibUsbApiHelpers.cs**
```csharp
public static MonoUsbDeviceHandle OpenDeviceWithVidPid([In]MonoUsbSessionHandle sessionHandle, ushort vendorID, ushort productID)
 {
        IntPtr pHandle = OpenDeviceWithVidPidInternal(sessionHandle, vendorID, productID);
        if (pHandle == IntPtr.Zero) return null;
        return new MonoUsbDeviceHandle(pHandle);
 }
```

bryancostanich, 2019-08-29:
I just ran into this issue as well, can confirm it should be ushort.

mo10, 2019-09-07:
See: https://github.com/LibUsbDotNet/LibUsbDotNet/issues/10 
May be helpful.

mcuee, 2023-04-23:
This should be fixed by #215.

closed 2023-04-23 by mcuee