LibUsbDotNet/LibUsbDotNet #93: Operation not supported or unimplemented on this platform

realivanjx, 2019-12-03:
Hi,

When trying to call `IUsbDevice.Open()` throws `Operation not supported or unimplemented on this platform`.

Also makes the app crashes with `AccessViolationException`.
Device im trying to connect is a samsung phone with adb enabled. What im trying to do is to implement adb in my software (works with winusb).

Thanks.

alex-belikov, 2019-12-16:
Hi,

I have the same issue under Windows 7 x86. I'm using NuGet version 3.0.48-alpha and DLL which is included in the package.

UsbContext.List() has my device in the list. Exception is thrown when I try to Open it. Let me know what additional information can help or how to dig further.

alex-belikov, 2019-12-16:
Ok, it's solved for me after installing this package:
**libusb-win32-devel-filter-1.2.6.0.exe**
I installed a filter for my device. But now I have a different issue....

mcuee, 2021-06-06:
Try to use WinUSB which may be more stable and not libusb-win32 filter driver.

mcuee, 2021-06-11:
For Windows, it is recommended to use Zadig to install drivers.

closed 2021-06-11 by mcuee