LibUsbDotNet/LibUsbDotNet #109: Many pieces of example code are out of date for 3.0.81-alpha

scottpidzarko, 2020-05-23:
There seems to be some changes with how certain methods work and what they return. For example it looks like UsbContext.Find(UsbDeviceFinder) now returns IUsbDevice instead of UsbDevice, making Visual Studio 2019 throw an implicit conversion error:

[image]

The above implicit conversion error is present in:

https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/src/Examples/Read.Only/ReadOnly.cs
https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/src/Examples/Read.Write.Async/ReadWriteAsync.cs

ReadWriteAsync.cs also has references to classes and namespaces that have gotten renamed or deleted:

[image]

https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/src/Examples/Read.Isochronous/ReadIsochronous.cs also has similar issues with implicitly trying to convert to IUsbDevice and references to classes and types that no longer exist.

Can we get a refresh to these pieces of example code? I'm trying to use this library with a dot net core project and I'm getting memory leak issues with LibUSB and I'm really not sure if I'm Opening/Closing IUsbDevice correctly - I can create a separate issue about this if needed.

alexgubanow, 2020-05-28:
im having even more issues, i don't have UsbContext class in LibUsbDotNet.LibUsb
Do you have any ideas why?
have same behavior for just cloned repo of this

scottpidzarko, 2020-06-29:
Nope no idea still, I ended up having to use an older version to get it working (2.2.29).

I'm running into portability issues on 2.2.29 now though ... can't find the libusb-1.0 dll.

doug62, 2020-11-06:
+1 (Glad i found this issue, was pulling my hair out)

alexgubanow, 2020-11-06:
@doug62 @scottpidzarko i have wrote my library hidapi wrapper for .net, u can see it here https://github.com/alexgubanow/hidapiw
it have been used to two projects, seems to be working, but im not sure about nuget pakage, since i was using it as git submodule

mcuee, 2021-06-06:
It will be great to have the examples working.

alexgubanow, 2021-06-07:
@mcuee what examples you are talking about?

mcuee, 2021-06-07:
Benchmark and Benchmarkcon.

mcuee, 2021-06-10:
Now I can say the examples works fine with the latest v2 branch. Asysnc examples like Benchmarkcon will not work under the mainline V3 branch as explained by Frederik.
And #126, #137 and #138 are kind all resolved from my side.

closed 2021-06-11 by mcuee