LibUsbDotNet/LibUsbDotNet #111: which version support usbcontext?

lxa520, 2020-06-24:
1. which version support usbcontext
2. How to know the version of libusbdotnet corresponding to the submitted version

qzzf1w, 2020-06-28:
I am running samples with latest code and usbcontext isnt there?

shawty, 2021-01-12:
Snap, same here....

shawty, 2021-01-12:
Update:

I have gone back through every version offered on NUGet:

[image]

And "UsbContext()" is not found in any of them.

shawty, 2021-01-12:
Update 2:

After examining the object browser in VS, I find this (For the most recent version)

[image]

When looking in the same examples folder as the previously mentioned C# example, there is also a Visual Basic sample (https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/src/Examples/Read.Write/ReadWrite.vb)

Upon examining the code in the visual basic sample, it appears to NOT USE the UsbContext as the C# sample does, but instead uses the "UsbDeviceFinder" as seen in the object explorer screen shot.

Translating the bulk of the VB code to C# gives us the following:

[image]

And that code compiles and works with the current NuGet version.

It appears the solution to this issue is to bring the examples for C# up to the same version the VB samples clearly target.

qzzf1w, 2021-01-12:
Ch-Ching!

shawty, 2021-06-11:
So has this been closed without updating the C# example in the repo then?

By doing that, you'll only have future developers who are not aware of this, opening new issues saying that "UsbContext" is not found.

mcuee, 2021-06-11:
Hmm, examples have been updated.
https://github.com/LibUsbDotNet/LibUsbDotNet/blob/v2/stage/Examples/Read.Write/ReadWrite.cs

shawty, 2021-06-11:
You missed one:

https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/src/Examples/Read.Write/ReadWrite.cs

This was the actual C# example I cited in "Post Number 3" above ^^^^^

😊

mcuee, 2021-06-11:
Okay, I will reopen this one as Example.

shawty, 2021-06-11:
No worries, just wanted to make sure you where not causing yourself any extra work with other devs reading this, seeing it closed and asking for help.

mcuee, 2021-06-11:
@shawty If you use mainline git or the nuget 3.x-alpha version, UsbContext() is there. If you use V2 branch, the examples has no UsbContext().
https://github.com/LibUsbDotNet/LibUsbDotNet/issues/89

shawty, 2021-06-11:
Yup, go for it.

closed 2021-06-11 by mcuee