LibUsbDotNet/LibUsbDotNet #114: System.DllNotFoundException

sbogutzky, 2020-08-02:
I use macOS 10.15.5 and Visual Studio for Mac Community 8.6.8 (build 2). I installed libusb-1.0.23.

"libusb-1.0 library not found. This is often an indication that libusb-1.0 was installed to '/usr/local/lib' and mono.net is not looking for it there. To resolve this, add the path '/usr/local/lib' to '/etc/ld.so.conf' and run 'ldconfig' as root. (http://www.mono-project.com/DllNotFoundException)"

I installed libusb via homebrew.

/usr/local/Cellar/libusb/1.0.23: 26 files, 474.2KB

The mono resource above only describe how to fix it on a linux system. Ldconfig is not available on macOS. The installation of libusb is not the problem. But how can we configure mono on macOS to search in the path /usr/local/bin for libusb?

mcuee, 2021-06-10:
Worst case just copy libusb-1.0.dylib to your bin directory.

mcuee, 2021-06-10:
The other way is to set LD_LIBRARY_PATH.
Ref: https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/azure-pipelines.yml

closed 2021-06-11 by mcuee