LibUsbDotNet/LibUsbDotNet #116: can not claim interface in linux

zmarsel1, 2020-09-08:
can not claim interface in linux
```
  var usbDeviceCollection = context.List();
  var selectedDevice = usbDeviceCollection.FirstOrDefault(d => d.ProductId == ProductId && d.VendorId == VendorId);
  selectedDevice.Open();

  selectedDevice.SetConfiguration(0);

  selectedDevice.ClaimInterface(selectedDevice.Configs[0].Interfaces[0].Number);// << returns Resourse busy error
  
  var writeEndpoint = selectedDevice.OpenEndpointWriter(WriteEndpointID.Ep01);
  
  var buffer = System.Text.UTF8Encoding.UTF8.GetBytes(test);
  writeEndpoint.Write(buffer, 3000, out var bytesWritten);
  selectedDevice.ReleaseInterface(selectedDevice.Configs[0].Interfaces[0].Number);

  selectedDevice.Close();
```
returns Resource busy error
using 3.0.81-alpha version

cannot access to NativeMethods.DetachKernelDriver(), because it is in internal class

how can i fix this error?

mcuee, 2021-06-06:
You can detach the kernel driver outside of the application, if you can not do it within your application.

mcuee, 2021-06-10:
A better solution is to use udev rules.
https://github.com/LibUsbDotNet/LibUsbDotNet/issues/128

closed 2021-06-11 by mcuee