LibUsbDotNet/LibUsbDotNet #117: I wonder if LibUsbDotNet would work with UWP App

longst, 2020-10-05:
I am trying to use LibUsbDotNet in an UWP App. I got 
Exception thrown: 'System.PlatformNotSupportedException' in LibUsbDotNet.LibUsbDotNet.dll

from 

UsbDevice usbDevice = UsbDevice.OpenUsbDevice(finder);

I wonder if someone happen to know this issue please?

Thanks

mcuee, 2021-06-06:
I do not think it will work as it depends on libusb.

mcuee, 2021-06-11:
@qmfrederik Correct me if I am wrong. Thanks.

mcuee, 2021-06-11:
This may be of interests as well.
https://github.com/libusb/libusb/issues/926

qmfrederik, 2021-06-11:
I know very little about UWP. I seem to remember there may be some restrictions w.r.t. how you invoke native methods (e.g. calls into `libusb.dll`), so you'll have to try.

mcuee, 2021-06-11:
It seems to me the restrictions are very severe.
https://docs.microsoft.com/en-us/uwp/api/windows.devices.usb?view=winrt-20348
>This namespace defines Windows Runtime classes that a UWP app can use to communicate with WinUSB devices. These devices are handled by the inbox winusb.sys driver and are identified by a specific Microsoft OS Descriptor.

mcuee, 2021-06-11:
Example device with WinUSB Microsoft OS Descriptor is the Benchmark FW device.

```
.\examples\xusb.exe 04d8:fa2e
Using libusb v1.0.24.11627

Opening device 04D8:FA2E...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 04D8:FA2E
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 00.00.00
       endpoint[0].address: 01
           max packet size: 0020
          polling interval: 00
       endpoint[1].address: 81
           max packet size: 0020
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Travis Robinson"
   String (0x02): "Benchmark Device"
   String (0x03): "LUSBW1"

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 57 49 4e 55 53 42 00 00 00 00 00 00 00 00  ..WINUSB........
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  92 00 00 00 00 01 05 00 01 00 84 00 00 00 07 00  ................
  00000010  00 00 2a 00 44 00 65 00 76 00 69 00 63 00 65 00  ..*.D.e.v.i.c.e.
  00000020  49 00 6e 00 74 00 65 00 72 00 66 00 61 00 63 00  I.n.t.e.r.f.a.c.
  00000030  65 00 47 00 55 00 49 00 44 00 73 00 00 00 50 00  e.G.U.I.D.s...P.
  00000040  00 00 7b 00 46 00 37 00 30 00 32 00 34 00 32 00  ..{.F.7.0.2.4.2.
  00000050  43 00 37 00 2d 00 46 00 42 00 32 00 35 00 2d 00  C.7.-.F.B.2.5.-.
  00000060  34 00 34 00 33 00 42 00 2d 00 39 00 45 00 37 00  4.4.3.B.-.9.E.7.
  00000070  45 00 2d 00 41 00 34 00 32 00 36 00 30 00 46 00  E.-.A.4.2.6.0.F.
  00000080  33 00 37 00 33 00 39 00 38 00 32 00 7d 00 00 00  3.7.3.9.8.2.}...
  00000090  00 00                                            ..

Releasing interface 0...
Closing device...

```

mcuee, 2021-06-11:
Keep this open as a future enhancement ideas. Not so sure if it is feasible or not.

mcuee, 2021-06-12:
@longst You may want to look at Device.Net which seems to support UWP (HID and USB).
https://github.com/MelbourneDeveloper/Device.Net

mcuee, 2023-04-22:
Device.Net development seems to have stopped. Still I will keep this issue open for future enhancement idea if it is feasible.

closed 2025-04-12 by mcuee