LibUsbDotNet/LibUsbDotNet #122: Unable to get the serial number and name of devices

toffuser, 2020-12-03:
Running this command :             var allDevices = UsbDevice.AllDevices;

[image]

I am unable to see what is the serial number or manufacturer number... but if I run this command in linux : usb-devices 
I can see what are the devices serial number. 

I run LibUsbDotNet with a .net5 application on a raspberryPi

mcuee, 2021-06-06:
Sorry just wondering how you build with .NET5 under Raspberry Pi. There seem to be issue building libusbdotnet under .Net 5.0.

mcuee, 2021-06-10:
> Sorry just wondering how you build with .NET5 under Raspberry Pi. There seem to be issue building libusbdotnet under .Net 5.0.

I am fine with the latest git now.

mcuee, 2023-04-15:
PR #203 should have fix this issue. I will close this for now.

@toffuser 

Please try latest v2 branch to see if you still have issues or not. If you still have the issues, please reopen.

closed 2023-04-15 by mcuee