LibUsbDotNet/LibUsbDotNet #123: MacOS - receiving exception when calling LibUsbDevice.Open(): "System.DllNotFoundException: Unable to load shared library 'kernel32.dll' ...

greglund, 2020-12-17:
Calling, for example: 
`LibUsbDevice.Open("usbdev1.1", out LibUsbDevice libUsbDevice)`

I get an exception:

> System.DllNotFoundException: Unable to load shared library 'kernel32.dll' or one of its dependencies.

For some reason it LibUsbDotNet has possibly mis-identified the system as a windows environment.  Is there any additional information i need to provide the LibUsbDotNet to identify the system correctly?

Project info:

>       Microsoft.NETCore.App (3.1.0)
>       LibUsbDotNet (2.2.29)

Visual Studio info:

> === Visual Studio Community 2019 for Mac ===
> 
> Version 8.8.4 (build 30)
> Installation UUID: aaf8a3be-6fe5-4b53-8e07-f51730f8a694
> 	GTK+ 2.24.23 (Raleigh theme)
> 	Xamarin.Mac 6.18.0.23 (d16-6 / 088c73638)
> 
> 	Package version: 612000113
> 
> === Mono Framework MDK ===
> 
> Runtime:
> 	Mono 6.12.0.113 (2020-02/4fdfb5b1fd5) (64-bit)
> 	Package version: 612000113
> 
> === Roslyn (Language Service) ===
> 
> 3.8.0-5.20519.18+4c195c3ac1974edcefa76774d7a59a2350ec55fa
> 
> === NuGet ===
> 
> Version: 5.8.0.6860
> 
> === .NET Core SDK ===
> 
> SDK: /usr/local/share/dotnet/sdk/5.0.101/Sdks
> SDK Versions:
> 	5.0.101
> 	5.0.100
> 	3.1.404
> MSBuild SDKs: /Applications/Visual Studio.app/Contents/Resources/lib/monodevelop/bin/MSBuild/Current/bin/Sdks
> 
> === .NET Core Runtime ===
> 
> Runtime: /usr/local/share/dotnet/dotnet
> Runtime Versions:
> 	5.0.1
> 	5.0.0
> 	3.1.10
> 
> === .NET Core 3.1 SDK ===
> 
> SDK: 3.1.404
> 
> === Xamarin.Profiler ===
> 
> Version: 1.6.15.68
> Location: /Applications/Xamarin Profiler.app/Contents/MacOS/Xamarin Profiler
> 
> === Updater ===
> 
> Version: 11
> 
> === Xamarin.Android ===
> 
> Not Installed
> 
> === Microsoft OpenJDK for Mobile ===
> 
> Java SDK: Not Found
> 
> Android Designer EPL code available here:
> https://github.com/xamarin/AndroidDesigner.EPL
> 
> === Android SDK Manager ===
> 
> Version: 16.8.0.32
> Hash: 01a7774
> Branch: remotes/origin/d16-8
> Build date: 2020-12-10 00:32:24 UTC
> 
> === Android Device Manager ===
> 
> Version: 16.8.0.46
> Hash: 0a81419
> Branch: remotes/origin/d16-8
> Build date: 2020-12-10 00:32:44 UTC
> 
> === Apple Developer Tools ===
> 
> Xcode 12.3 (17715)
> Build 12C33
> 
> === Xamarin.Mac ===
> 
> Version: 7.0.0.15 (Visual Studio Community)
> Hash: 87a1b18d8
> Branch: d16-8
> Build date: 2020-11-16 21:39:40-0500
> 
> === Xamarin.iOS ===
> 
> Xamarin.iOS not installed.
> Can't find mtouch or the Version file at /Library/Frameworks/Xamarin.iOS.framework/Versions/Current.
> 
> === Xamarin Designer ===
> 
> Version: 16.8.0.510
> Hash: 44e3f3ce9
> Branch: remotes/origin/d16-8
> Build date: 2020-12-10 00:06:14 UTC
> 
> === Build Information ===
> 
> Release ID: 808040030
> Git revision: eaa21a8983aa5c4bcf6a665892478b4f3d413418
> Build date: 2020-12-10 11:06:12-05
> Build branch: release-8.8
> Xamarin extensions: eaa21a8983aa5c4bcf6a665892478b4f3d413418
> 
> === Operating System ===
> 
> Mac OS X 10.15.7
> Darwin 19.6.0 Darwin Kernel Version 19.6.0
>     Tue Nov 10 00:10:30 PST 2020
>     root:xnu-6153.141.10~1/RELEASE_X86_64 x86_64
>

txbrown, 2020-12-19:
@greglund  try doing

```c#
            var device = LibUsbDotNet.UsbDevice.OpenUsbDevice(d => d.Pid == 123);
```

where Pid is the product id of your usb device. You can also use the name and port as strings as your predicte.

Hope that works.

mcuee, 2021-06-11:
Please try the latest version (main line or v2 branch) and it should work.

closed 2021-06-11 by mcuee