LibUsbDotNet/LibUsbDotNet #125: LibUsbDotNet.Generated Assembly/NuGet

MelbourneDeveloper, 2020-12-28:
This project is great because it allows us to use libusb from .NET. However, the wrapper classes (in the `LibUsbDotNet.LibUsb` namespace) are still in alpha mode. I would really like to be able to use the low level APIs from [Device.Net](https://github.com/MelbourneDeveloper/Device.Net). But, the problem is that [`NativeMethods`](https://github.com/LibUsbDotNet/LibUsbDotNet/blob/bd5e640c5c84536620f082ff66860aa6ed9aa324/src/LibUsbDotNet/Generated/NativeMethods.cs#L38) is not public. Is there any chance we could expose the NativeMethods API as public which would allow other libraries like Usb.Net to directly access the libusb dll imports?

I'm happy to do the bulk of work for this as long as I can get some guidance. I can submit a PR. Here is my suggestion:

- Break the library in to two assemblies LibUsbDotNet, and LibUsbDotNet.Generated. 
- Add a new NuGet package for LibUsbDotNet.Generated
- Change the access modifier of `NativeMethods` to public so that it can be accessed by referencing projects

mcuee, 2021-06-06:
Sorry I do not know much about .NET but I was involved in the development of libusbdotnet by Travis Robinson back in 2009/2010 time by helping on the testing. It seems to me your idea may work.

qmfrederik, 2021-06-07:
While I'm not opposed to exposing the `NativeMethods` library per se, I do think that you'll find that you'll end up rewriting most of the `LibUsbDotNet.LibUsb` namespace if you start using the native methods directly. The `LibUsbDotNet.LibUsb` namespace was intended to be a very tiny wrapper around the native methods.

Is your main concern that the 3.0 series is currently marked as alpha? In that case, the best way to move the 3.0 series to stable is to help with testing and send PRs for any issues you find 😄 .

mcuee, 2024-05-05:
Close this old issue as not planned.

closed 2024-05-05 by mcuee