LibUsbDotNet/LibUsbDotNet #126: Cannot Build

MelbourneDeveloper, 2020-12-28:
I genuinely want to get this compiling so I can contribute to the project.

I'm using the latest version of VS: 16.8.3. I get

> Severity	Code	Description	Project	File	Line	Suppression State
Error		Project "C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\16.0\Bin\Microsoft.CSharp.targets" was not imported by "C:\Program Files\dotnet\sdk\5.0.101\Sdks\Microsoft.NET.Sdk\Sdk\Sdk.targets" at (37,3), due to the file not existing.	LibUsbDotNet	C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\16.0\Bin\Microsoft.CSharp.targets	37	
Error	MSB4019	The imported project "C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\16.0\Bin\Microsoft.CSharp.targets" was not found. Confirm that the expression in the Import declaration "C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\16.0\Bin\Microsoft.CSharp.targets" is correct, and that the file exists on disk.	LibUsbDotNet	C:\Program Files\dotnet\sdk\5.0.101\Sdks\Microsoft.NET.Sdk\Sdk\Sdk.targets	37	

[image]

Rider:
[image]

Is there anything special I need to do or install?

quocdai92, 2021-02-03:
I got the same issue

sarbos, 2021-02-12:
I had this issue as well when opening in VS2019, but most of the issues went away when opening in VS2017.

alex-g-axion, 2021-04-27:
I get these errors in VS2019:

`Error		Project "C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\16.0\Bin\Microsoft.CSharp.targets" was not imported by "C:\Program Files\dotnet\sdk\5.0.201\Sdks\Microsoft.NET.Sdk\Sdk\Sdk.targets" at (37,3), due to the file not existing.	LibUsbDotNet	C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\16.0\Bin\Microsoft.CSharp.targets	37	
`
and 

`Error	MSB4044	The "GenerateSupportedTargetFrameworkAlias" task was not given a value for the required parameter "TargetFrameworkMoniker".	LibUsbDotNet	C:\Program Files\dotnet\sdk\5.0.201\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.Sdk.targets	1023`

qmfrederik, 2021-06-07:
I've pushed a couple of commits to master which should make it easier to build LibUsbDotNet from source. Can you try the latest version in master and let me know whether you still have issues compiling LibUsbDotNet itself? (Note: some of the examples still need to be updated).

mcuee, 2021-06-08:
Sorry I almost never use VS, this is the output, which may not have anything to do with 

```
Build started...
1>------ Build started: Project: LibUsbDotNet, Configuration: Debug Any CPU ------
Restored C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Examples\Read.Only\Read.Only.csproj (in 6 ms).
Restored C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\BenchmarkCon\BenchmarkCon.csproj (in 7 ms).
Restored C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Examples\Read.Write.Async\Read.Write.Async.csproj (in 5 ms).
Restored C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Test_Info\Test_Info.csproj (in 6 ms).
Restored C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Examples\Read.Write\Read.Write.csproj (in 7 ms).
Restored C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Examples\Show.Info\Show.Info.csproj (in 6 ms).
Restored C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Examples\Read.Isochronous\Read.Isochronous.csproj (in 6 ms).
Failed to restore C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet.Tests\LibUsbDotNet.Tests.csproj (in 14 ms).
Failed to restore C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj (in 41 ms).
Failed to restore C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet.Generator\LibUsbDotNet.Generator.csproj (in 49 ms).
NuGet package restore failed. Please see Error List window for detailed warnings and errors.
1>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj : warning NU1603: System.Memory 4.5.1 depends on System.Buffers (>= 4.4.0) but System.Buffers 4.4.0 was not found. An approximate best match of System.Buffers 4.5.0 was resolved.
1>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj : error NU1101: Unable to find package Nerdbank.GitVersioning. No packages exist with this id in source(s): Microsoft Visual Studio Offline Packages
1>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj : error NU1101: Unable to find package Microsoft.SourceLink.GitHub. No packages exist with this id in source(s): Microsoft Visual Studio Offline Packages
1>Done building project "LibUsbDotNet.csproj" -- FAILED.
1>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj : warning NU1603: System.Memory 4.5.1 depends on System.Buffers (>= 4.4.0) but System.Buffers 4.4.0 was not found. An approximate best match of System.Buffers 4.5.0 was resolved.
1>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj : error NU1101: Unable to find package Nerdbank.GitVersioning. No packages exist with this id in source(s): Microsoft Visual Studio Offline Packages
1>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj : error NU1101: Unable to find package Microsoft.SourceLink.GitHub. No packages exist with this id in source(s): Microsoft Visual Studio Offline Packages
1>Done building project "LibUsbDotNet.csproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

```

mcuee, 2021-06-08:
Same under Command Line (Windows Terminal).

```

C:\work\libusb\LibUsbDotNet\LibUsbDotNet [master ≡]> dotnet pack src\LibUsbDotNet\LibUsbDotNet.csproj -c Release
Microsoft (R) Build Engine version 16.10.0+4242f381a for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj : warning NU1603: System.Memory 4.5.1 depends on System.Buffers (>= 4.4.0) but System.Buffers 4.4.0 was not found. An approximate best match of System.Buffers 4.5.0 was resolved.
C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj : error NU1101: Unable to find package Nerdbank.GitVersioning. No packages exist with this id in source(s): Microsoft Visual Studio Offline Packages
C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj : error NU1101: Unable to find package Microsoft.SourceLink.GitHub. No packages exist with this id in source(s): Microsoft Visual Studio Offline Packages
  Failed to restore C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj (in 180 ms).

```

mcuee, 2021-06-09:
After proper configuration of Nuget, now it is okay.
Build started...

```
1>------ Build started: Project: LibUsbDotNet.Generator, Configuration: Debug Any CPU ------
2>------ Build started: Project: Test_Info, Configuration: Debug Any CPU ------
3>------ Build started: Project: Read.Write, Configuration: Debug Any CPU ------
4>------ Build started: Project: Read.Only, Configuration: Debug Any CPU ------
5>------ Build started: Project: Show.Info, Configuration: Debug Any CPU ------
6>------ Build started: Project: Read.Write.Async, Configuration: Debug Any CPU ------
7>------ Build started: Project: BenchmarkCon, Configuration: Debug Any CPU ------
8>------ Build started: Project: Read.Isochronous, Configuration: Debug Any CPU ------
9>------ Build started: Project: LibUsbDotNet.Tests, Configuration: Debug Any CPU ------
1>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet.Generator\LibUsbDotNet.Generator.csproj : warning NU1603: LibUsbDotNet.Generator depends on Core.Clang (>= 5.1.5-beta) but Core.Clang 5.1.5-beta was not found. An approximate best match of Core.Clang 6.0.0-alpha1 was resolved.
1>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet.Generator\LibUsbDotNet.Generator.csproj : warning NU1701: Package 'Nustache 1.16.0.8' was restored using '.NETFramework,Version=v4.6.1, .NETFramework,Version=v4.6.2, .NETFramework,Version=v4.7, .NETFramework,Version=v4.7.1, .NETFramework,Version=v4.7.2, .NETFramework,Version=v4.8' instead of the project target framework '.NETCoreApp,Version=v2.1'. This package may not be fully compatible with your project.
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\Microsoft.Common.CurrentVersion.targets(2202,5): warning MSB3270: There was a mismatch between the processor architecture of the project being built "MSIL" and the processor architecture of the reference "C:\Users\xfchen\.nuget\packages\core.clang\6.0.0-alpha1\lib\netstandard2.0\Core.Clang.dll", "AMD64". This mismatch may cause runtime failures. Please consider changing the targeted processor architecture of your project through the Configuration Manager so as to align the processor architectures between your project and references, or take a dependency on references with a processor architecture that matches the targeted processor architecture of your project.
1>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet.Generator\FunctionVisitor.cs(195,42,195,49): error CS0246: The type or namespace name 'Comment' could not be found (are you missing a using directive or an assembly reference?)
1>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet.Generator\StructVisitor.cs(275,49,275,56): error CS0246: The type or namespace name 'Comment' could not be found (are you missing a using directive or an assembly reference?)
1>Done building project "LibUsbDotNet.Generator.csproj" -- FAILED.
6>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Examples\Read.Write.Async\ReadWriteAsync.cs(7,20,7,28): error CS0234: The type or namespace name 'Internal' does not exist in the namespace 'LibUsbDotNet' (are you missing an assembly reference?)
6>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Examples\Read.Write.Async\ReadWriteAsync.cs(9,20,9,34): error CS0234: The type or namespace name 'LudnMonoLibUsb' does not exist in the namespace 'LibUsbDotNet' (are you missing an assembly reference?)
6>Done building project "Read.Write.Async.csproj" -- FAILED.
7>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\BenchmarkCon\BenchmarkConsole.cs(126,20,126,31): error CS0246: The type or namespace name 'UsbTransfer' could not be found (are you missing a using directive or an assembly reference?)
7>Done building project "BenchmarkCon.csproj" -- FAILED.
8>C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Examples\Read.Isochronous\ReadIsochronous.cs(184,42,184,58): error CS0246: The type or namespace name 'UsbTransferQueue' could not be found (are you missing a using directive or an assembly reference?)
8>Done building project "Read.Isochronous.csproj" -- FAILED.
3>Read.Write -> C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Examples\Read.Write\bin\Debug\net5.0\Read.Write.dll
5>Show.Info -> C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Examples\Show.Info\bin\Debug\net5.0\Show.Info.dll
4>Read.Only -> C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Examples\Read.Only\bin\Debug\net5.0\Read.Only.dll
2>Test_Info -> C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\Test_Info\bin\Debug\net5.0-windows\Test_Info.dll
9>LibUsbDotNet.Tests -> C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet.Tests\bin\Debug\netcoreapp3.1\LibUsbDotNet.Tests.dll
========== Build: 5 succeeded, 4 failed, 1 up-to-date, 0 skipped ==========
```

mcuee, 2021-06-09:
And also command line.

```
C:\work\libusb\LibUsbDotNet\LibUsbDotNet [master ≡]> dotnet pack src\LibUsbDotNet\LibUsbDotNet.csproj -c Release
Microsoft (R) Build Engine version 16.10.0+4242f381a for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
  All projects are up-to-date for restore.
  LibUsbDotNet -> C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\bin\Release\net45\LibUsbDotNet.dll
  LibUsbDotNet -> C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\bin\Release\netcoreapp3.1\LibUsbDotNet.dll
C:\Program Files\dotnet\sdk\5.0.300\Sdks\NuGet.Build.Tasks.Pack\buildCrossTargeting\NuGet.Build.Tasks.Pack.targets(221,5): warning NU5048: The 'PackageIconUrl'/'iconUrl' element is deprecated. Consider using the 'PackageIcon'/'icon' element instead. Learn more at https://aka.ms/deprecateIconUrl [C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj]
  Successfully created package 'C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\bin\Release\LibUsbDotNet.3.0.97-alpha.nupkg'.
  Successfully created package 'C:\work\libusb\LibUsbDotNet\LibUsbDotNet\src\LibUsbDotNet\bin\Release\LibUsbDotNet.3.0.97-alpha.snupkg'.
```

mcuee, 2021-06-10:
For those who need async examples, you can use V2 branch. It works very well under VS2019.

Example:
```

C:\work\libusb\LibUsbDotNet\libusbdotnet_test\stage\bin\Release [v2 ≡ +0 ~5 -0 !]> .\BenchmarkCon.exe vid=0x04d8 pid=0xfa2e
LibUsbDotNet USB Benchmark v2.2.9.110
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
Benchmark device 04D8:FA2E opened..
Loop Test Information
        Vid / Pid       : 04D8h / FA2Eh
        Interface #     : 00h
        Alt Interface # : 00h
        Priority        : Normal
        Buffer Size     : 4096
        Buffer Count    : 1
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
Bulk Write (Ep01h) max packet size: 32

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..
Avg. Bytes/s: 308094.76 Transfers: 72 Bytes/s: 308094.76
Avg. Bytes/s: 307636.66 Transfers: 148 Bytes/s: 307203.92
Avg. Bytes/s: 307826.11 Transfers: 224 Bytes/s: 308195.71
Avg. Bytes/s: 308287.01 Transfers: 301 Bytes/s: 309635.72
Avg. Bytes/s: 308144.11 Transfers: 377 Bytes/s: 307579.46
Avg. Bytes/s: 308087.22 Transfers: 453 Bytes/s: 307805.30
Avg. Bytes/s: 308286.51 Transfers: 529 Bytes/s: 309479.73
Avg. Bytes/s: 308337.12 Transfers: 605 Bytes/s: 308689.86
qAvg. Bytes/s: 308365.39 Transfers: 681 Bytes/s: 308590.65
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
        Vid / Pid       : 04D8h / FA2Eh
        Interface #     : 00h
        Alt Interface # : 00h
        Priority        : Normal
        Buffer Size     : 4096
        Buffer Count    : 1
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
        Total Bytes     : 2793472
        Total Transfers : 682
        Avg. Bytes/sec  : 308390.33
        Elapsed Time    : 9.06 seconds

Bulk Write (Ep01h) max packet size: 32
        Total Bytes     : 2793472
        Total Transfers : 682
        Avg. Bytes/sec  : 308390.33
        Elapsed Time    : 9.06 seconds

Press any key to exit..
```

mcuee, 2021-06-29:
In case you want to have the binary for the v2 branch for Windows, I have create a snapshot here (using VS2019).
https://github.com/mcuee/LibUsbDotNet/releases

closed 2023-01-06 by mcuee