LibUsbDotNet/LibUsbDotNet #128: Cannot SetConfiguration or ClaimInterface with .NET Core 3.1 on Linux

Sinipelto, 2021-01-10:
I have tested with the pre-installed libusb-1.0 (libusb-1.0.so.0.1.0) and also compiled the latest version libusb-1.0.24 from sources (libusb-1.0.so.0.3.0).

I'm using .NET Core 3.1 console application. I'm trying to send a status command to a VKP80II-sx POS printer using USB interface and also to read the status result back.

I'm able to find the correct device using the interface and open it, but when trying to set the configuration to 1 and claiming the interface 0 (also mentioned in the printer manual) the both commands fail, and trying to read or write to the USB device gives Input/Output error (MonoApiError, error num -16353).

` class Program
    {
        private const int Vid = 0x0dd4;
        private const int Pid = 0x015d;
        private const int BufferSize = 2024;

        static UsbDeviceFinder Finder =
            new UsbDeviceFinder(Vid, Pid);

        public static void Main()
        {
            SingleDevice();
            // DeviceEnumeration();
        }

        public static void SingleDevice()
        {
            var dev = UsbDevice.OpenUsbDevice(Finder);

            if (dev == null)
            {
                Console.WriteLine("ERROR!");
                return;
            }

            // Console.WriteLine($"Dev.Open: {dev.Open()}");

            Console.WriteLine($"DEVICE OPEN: {dev.IsOpen}");
            Console.WriteLine($"DEVICE ACTIVE ENDPOINTS: {dev.ActiveEndpoints.Count}");
            Console.WriteLine($"DEVICE CONFIGS: {dev.Configs.Count}");
            Console.WriteLine($"DEVICE DRIVER MODE: {dev.DriverMode}");
            Console.WriteLine($"DEVICE PATH: {dev.DevicePath}");

            Console.WriteLine($"Read Error Num: {UsbDevice.LastErrorNumber}");
            Console.WriteLine($"Write Error String: {UsbDevice.LastErrorString}");

            var idev = dev as IUsbDevice;

            // UsbRegistryInfo: null
            // Console.WriteLine($"USB REV: {idev.UsbRegistryInfo.Rev}");

            // Non-interface has no SetConfiguration or ClaimInterface methods
            if (!idev.SetConfiguration(1)) Console.WriteLine("CONFIGURE FAIL");

            if (!idev.ClaimInterface(0)) Console.WriteLine("CLAIM FAIL");
        }`

mcuee, 2021-06-05:
You have to set up the write permission (udev rules) in order to use libusb/libusbdotnet under Linux. You can also use sudo as a temp workaround.

sinipelto, 2021-06-05:
I tried to run the code as root user, it did not fix the issue.

mcuee, 2021-06-06:
You may need to detach the kernel driver first.

smdn, 2021-06-06:
I've faced same problem.
Try this udev rule. (Please change the `ATTRS{idVendor}` and `ATTRS{idProduct}` to your device's one)

https://github.com/smdn/Smdn.Devices.MCP2221/blob/47e047e42c918634dc465c83c445a3b3d3ea1daf/misc/udev/90-MCP2221-LibUsbDotNet.rules

mcuee, 2021-06-06:
Yes it is a good way to use udev rules to unbind kernel driver.