LibUsbDotNet/LibUsbDotNet #131: context.Find(MyUsbFinder) doesn't find device, though context.List() does

ryan4volts, 2021-03-17:
version from NuGet: v3.0.87-alpha

Using the example code in the repo, and placing my VID/PID into the `USBDeviceFinder` doesn't find my device at all, though id I do the match manually with context.List() and enumerate through it, it finds the target device just fine.

The below code outputs
```
Enumerate -> Device found

Device Not Found.
```

```
using LibUsbDotNet;
using LibUsbDotNet.LibUsb;
using LibUsbDotNet.Main;
using System;
using System.Text;

namespace ConsoleApp2 {
	class Program {
        public static IUsbDevice MyUsbDevice;

        #region SET YOUR USB Vendor and Product ID!

        //VID_0403+PID_6015
        public static int vendor_id = 0x0403;
        public static int product_id = 0x6015;
        public static UsbDeviceFinder MyUsbFinder = new UsbDeviceFinder(vendor_id, product_id);

        #endregion

        public static void Main(string[] args) {
            Error ec = Error.Success;
            
            using (UsbContext context = new UsbContext()) {
                try {
                    // Find and open the usb device.
                    var ls = context.List();

                    foreach (var item in ls) {
                        if (item.VendorId == vendor_id && item.ProductId == product_id) {
                            Console.WriteLine("Enumerate -> Device found");
						}
					}
                    MyUsbDevice = context.Find(MyUsbFinder);

                    // If the device is open and ready
                    if (MyUsbDevice == null)
                        throw new Exception("Device Not Found.");

                    // If this is a "whole" usb device (libusb-win32, linux libusb-1.0)
                    // it exposes an IUsbDevice interface. If not (WinUSB) the 
                    // 'wholeUsbDevice' variable will be null indicating this is 
                    // an interface of a device; it does not require or support 
                    // configuration and interface selection.
                    IUsbDevice wholeUsbDevice = MyUsbDevice as IUsbDevice;
                    if (!ReferenceEquals(wholeUsbDevice, null)) {
                        // This is a "whole" USB device. Before it can be used, 
                        // the desired configuration and interface must be selected.

                        // Select config #1
                        wholeUsbDevice.SetConfiguration(1);

                        // Claim interface #0.
                        wholeUsbDevice.ClaimInterface(0);
                    }

                    // open read endpoint 1.
                    var reader = MyUsbDevice.OpenEndpointReader(ReadEndpointID.Ep01);

                    byte[] readBuffer = new byte[1024];
                    while (ec == Error.Success) {
                        int bytesRead;

                        // If the device hasn't sent data in the last 5 seconds,
                        // a timeout error (ec = IoTimedOut) will occur. 
                        ec = reader.Read(readBuffer, 5000, out bytesRead);

                        if (bytesRead == 0)
                            throw new Exception(string.Format("{0}:No more bytes!", ec));
                        Console.WriteLine("{0} bytes read", bytesRead);

                        // Write that output to the console.
                        Console.Write(Encoding.Default.GetString(readBuffer, 0, bytesRead));
                    }

                    Console.WriteLine("\r\nDone!\r\n");
                }
                catch (Exception ex) {
                    Console.WriteLine();
                    Console.WriteLine((ec != Error.Success ? ec + ":" : String.Empty) + ex.Message);
                }
                finally {
                    if (MyUsbDevice != null) {
                        if (MyUsbDevice.IsOpen) {
                            // If this is a "whole" usb device (libusb-win32, linux libusb-1.0)
                            // it exposes an IUsbDevice interface. If not (WinUSB) the 
                            // 'wholeUsbDevice' variable will be null indicating this is 
                            // an interface of a device; it does not require or support 
                            // configuration and interface selection.
                            IUsbDevice wholeUsbDevice = MyUsbDevice as IUsbDevice;
                            if (!ReferenceEquals(wholeUsbDevice, null)) {
                                // Release interface #0.
                                wholeUsbDevice.ReleaseInterface(0);
                            }

                            MyUsbDevice.Close();
                        }
                        MyUsbDevice = null;
                    }

                    // Wait for user input..
                    Console.ReadKey();
                }
            }
        }
    }
}
```

ryan4volts, 2021-03-17:
Looking at the code, I believe the culprit is UsbDeviceFinder.cs:251. Here the revision is tested
```
                if (this.Revision != null && usbDevice.Info.Usb != this.Revision.Value)
                {
                    return false;
                }
```

Though since I never populated the revision in my constructor, it has defaulted to a max signed int value of 2147483647, not NULL. By comparison when inspecting the device found by doing a manual match, the usb.info is just 512.

[image]

[image]

mcuee, 2022-05-15:
Please try out pull request #87.
* https://github.com/LibUsbDotNet/LibUsbDotNet/pull/87

mcuee, 2023-04-22:
#87 has been merged. I will close this issue. Please reopen if the issue pops up again.

closed 2023-04-22 by mcuee