LibUsbDotNet/LibUsbDotNet #135: libusbdotnet for .net5.0

tercelshao, 2021-05-01:
hi
I am trying to convert libusbdotnet from netframework 4.5 to .net 5.0, but ir can not work. 

I just got a reminder from the .csproj,"
    
    <ReferenceAssemblyProjectReference Include="..\refs\LibUsbDotNet.LibUsb.refs.csproj" />
  </ItemGroup>

Who can give me suggestions to compile the project to .net5.0 successfully? thanks a lot.

tercelshao, 2021-05-01:
One error is 
 	CS0103	:“LibUsbNativeLibrary”	LibUsbDotNet	C:\tmp\22\LibUsbDotNet-master\src\LibUsbDotNet\Generated\NativeMethods.cs	59

mcuee, 2021-06-06:
Yes it will be good to add support of .NET 5.0 and .NET core 5.0.

I can see that .NET Core 2.1 LTS (end of support by Microsoft) and .NET Core 3.1 LTS are currently supported.

qmfrederik, 2021-06-07:
@tercelshao I pushed a couple of commits to master and LibUsbDotNet should now build for .NET 5.0. Can you give that a try and let me know if that works for you?

mcuee, 2021-06-10:
I believe it now works fine, tested under Windows 10 x86_64, Mac Mini M1 (macOS ARM64) and Raspberry Pi 400 (Ubuntu ARM64 Linux).

mcuee, 2023-01-08:
v2 branch now supports .Net 6.0.
It will be good to support .Net 6.0 in the master branch as well.

mcuee, 2023-02-06:
Results for .Net Core 7.0.

Master branch:
```
C:\work\libusb\libusbdotnet\LibUsbDotNet [master ≡]> dotnet pack src\LibUsbDotNet\LibUsbDotNet.csproj -c Release -o .\nuget
MSBuild version 17.4.1+9a89d02ff for .NET
  Determining projects to restore...
  Restored C:\work\libusb\libusbdotnet\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj (in 336 ms).
  LibUsbDotNet -> C:\work\libusb\libusbdotnet\LibUsbDotNet\src\LibUsbDotNet\bin\Release\net45\LibUsbDotNet.dll
  LibUsbDotNet -> C:\work\libusb\libusbdotnet\LibUsbDotNet\src\LibUsbDotNet\bin\Release\netcoreapp3.1\LibUsbDotNet.dll
C:\Program Files\dotnet\sdk\7.0.102\Sdks\NuGet.Build.Tasks.Pack\buildCrossTargeting\NuGet.Build.Tasks.Pack.targets(221,
5): warning NU5048: The 'PackageIconUrl'/'iconUrl' element is deprecated. Consider using the 'PackageIcon'/'icon' eleme
nt instead. Learn more at https://aka.ms/deprecateIconUrl [C:\work\libusb\libusbdotnet\LibUsbDotNet\src\LibUsbDotNet\Li
bUsbDotNet.csproj]
  Successfully created package 'C:\work\libusb\libusbdotnet\LibUsbDotNet\nuget\LibUsbDotNet.3.0.102-alpha.nupkg'.
  Successfully created package 'C:\work\libusb\libusbdotnet\LibUsbDotNet\nuget\LibUsbDotNet.3.0.102-alpha.snupkg'.

```

closed 2023-02-06 by mcuee

mcuee, 2023-02-06:
dotnet Core SDK build of libusbdotnet v2 branch.

```
PS C:\work\libusb\libusbdotnet\libusbdotnet_v2> dotnet build .\stage\LibUsbDotNet.sln -c Release
MSBuild version 17.4.1+9a89d02ff for .NET
  Determining projects to restore...
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\Examples\Device.Notification\Device.Notification.csproj (i
  n 403 ms).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\Test_DeviceNotify\Test_DeviceNotify.csproj (in 24 ms).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\Examples\Read.Write.Async\Read.Write.Async.csproj (in 2.49
   sec).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\Test_Bulk\Test_Bulk.csproj (in 2.02 sec).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\Test_Info\Test_Info.csproj (in 2.49 sec).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\Examples\Read.Only\Read.Only.csproj (in 2.49 sec).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\Benchmark\Benchmark.csproj (in 2.49 sec).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\Examples\Read.Isochronous\Read.Isochronous.csproj (in 2.49
   sec).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\Examples\Read.Write\Read.Write.csproj (in 2.49 sec).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\Examples\Read.Write.Event\Read.Write.Event.csproj (in 2.48
   sec).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\InfWizard\InfWizard.csproj (in 5 ms).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\BenchmarkCon\BenchmarkCon.csproj (in 3 ms).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\Examples\Show.Info\Show.Info.csproj (in 6 ms).
  Restored C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\LibUsbDotNet.csproj (in 89 ms).
  LibUsbDotNet -> C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\bin\Release\net46\LibUsbDotNet.LibUsbD
  otNet.dll
  LibUsbDotNet -> C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\bin\Release\netstandard2.0\LibUsbDotNe
  t.LibUsbDotNet.dll
C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\Internal\Kernel32.cs(89,62): warning CS0618: 'UnmanagedT
ype.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Specify the type you wish t
o marshal as.' [C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\LibUsbDotNet.csproj::TargetFramework=net
6.0]
C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\Internal\Kernel32.cs(99,62): warning CS0618: 'UnmanagedT
ype.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Specify the type you wish t
o marshal as.' [C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\LibUsbDotNet.csproj::TargetFramework=net
6.0]
C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\Internal\Kernel32.cs(119,70): warning CS0618: 'Unmanaged
Type.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Specify the type you wish
to marshal as.' [C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\LibUsbDotNet.csproj::TargetFramework=ne
t6.0]
C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\Internal\SetupApi.cs(239,77): warning CS0618: 'Unmanaged
Type.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Specify the type you wish
to marshal as.' [C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\LibUsbDotNet.csproj::TargetFramework=ne
t6.0]
C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\Internal\SetupApi.cs(296,81): warning CS0618: 'Unmanaged
Type.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Specify the type you wish
to marshal as.' [C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\LibUsbDotNet.csproj::TargetFramework=ne
t6.0]
C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\WinUsb\Internal\LibusbKAPI.cs(102,78): warning CS0618: '
UnmanagedType.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Specify the type
you wish to marshal as.' [C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\LibUsbDotNet.csproj::TargetFra
mework=net6.0]
C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\WinUsb\Internal\WinUsbAPI.cs(104,78): warning CS0618: 'U
nmanagedType.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Specify the type y
ou wish to marshal as.' [C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\LibUsbDotNet.csproj::TargetFram
ework=net6.0]
C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\MonoLibUsb\MonoUsbEventHandler.cs(188,25): warning SYSLI
B0006: 'Thread.Abort()' is obsolete: 'Thread.Abort is not supported and throws PlatformNotSupportedException.' [C:\work
\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\LibUsbDotNet.csproj::TargetFramework=net6.0]
C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\LibUsb\Internal\LibUsbRequest.cs(71,30): warning CS0618:
 'Marshal.ReadByte(object, int)' is obsolete: 'ReadByte(Object, Int32) may be unavailable in future releases.' [C:\work
\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\LibUsbDotNet.csproj::TargetFramework=net6.0]
C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\Main\UsbDeviceFinder.cs(261,20): warning SYSLIB0011: 'Bi
[... cut ...]
\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotNet\LibUsbDotNet.csproj]
C:\Program Files\dotnet\sdk\7.0.102\Sdks\NuGet.Build.Tasks.Pack\buildCrossTargeting\NuGet.Build.Tasks.Pack.targets(221,
5): warning NU5048: The 'PackageIconUrl'/'iconUrl' element is deprecated. Consider using the 'PackageIcon'/'icon' eleme
nt instead. Learn more at https://aka.ms/deprecateIconUrl [C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\LibUsbDotN
et\LibUsbDotNet.csproj]
    25 Warning(s)
    0 Error(s)

Time Elapsed 00:00:13.95

```

mcuee, 2023-02-06:
dotnet 7.0 sdk build of git HEAD.

```
PS C:\work\libusb\libusbdotnet\LibUsbDotNet> dotnet build .\src\LibUsbDotNet\LibUsbDotNet.csproj -c Release
MSBuild version 17.4.1+9a89d02ff for .NET
  Determining projects to restore...
  All projects are up-to-date for restore.
  LibUsbDotNet -> C:\work\libusb\libusbdotnet\LibUsbDotNet\src\LibUsbDotNet\bin\Release\net45\LibUsbDotNet.dll
  LibUsbDotNet -> C:\work\libusb\libusbdotnet\LibUsbDotNet\src\LibUsbDotNet\bin\Release\netcoreapp3.1\LibUsbDotNet.dll

Build succeeded.
    0 Warning(s)
    0 Error(s)

```