LibUsbDotNet/LibUsbDotNet #137: How to install under macOS Apple Silicon

mcuee, 2021-06-05:
Sorry but I can not seem to be able to install libusbdotnet under my Mac Mini M1. Just wondering if there is any reference. I have installed Visual Studion for Mac.

mcuee, 2021-06-05:
Error messages are like this.

```
Target CoreCompile:
        /Applications/Visual Studio.app/Contents/Resources/lib/monodevelop/bin/MSBuild/Current/bin/Roslyn/csc.exe /noconfig /unsafe+ /checked- /nowarn:1701,1702,1701,1702,2008 /fullpaths /nostdlib+ /errorreport:prompt /warn:4 /define:TRACE;WIN;LIBUSBDOTNET;DEBUG;NETSTANDARD;NETSTANDARD2_0 /errorendlocation /preferreduilang:en-US /highentropyva+ /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/Microsoft.Win32.Primitives.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/mscorlib.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/netstandard.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.AppContext.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Collections.Concurrent.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Collections.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Collections.NonGeneric.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Collections.Specialized.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.ComponentModel.Composition.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.ComponentModel.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.ComponentModel.EventBasedAsync.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.ComponentModel.Primitives.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.ComponentModel.TypeConverter.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Console.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Core.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Data.Common.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Data.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Diagnostics.Contracts.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Diagnostics.Debug.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Diagnostics.FileVersionInfo.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Diagnostics.Process.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Diagnostics.StackTrace.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Diagnostics.TextWriterTraceListener.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Diagnostics.Tools.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Diagnostics.TraceSource.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Diagnostics.Tracing.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Drawing.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Drawing.Primitives.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Dynamic.Runtime.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Globalization.Calendars.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Globalization.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.Globalization.Extensions.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.IO.Compression.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.IO.Compression.FileSystem.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.IO.Compression.ZipFile.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.IO.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.IO.FileSystem.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.IO.FileSystem.DriveInfo.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.IO.FileSystem.Primitives.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.IO.FileSystem.Watcher.dll /reference:/Users/mcuee/.nuget/packages/netstandard.library/2.0.3/build/netstandard2.0/ref/System.IO.IsolatedStorage.dll /reference:/Users/mcuee/.nuget/packages/netstandar [... cut ...]
        /Users/mcuee/build/libusb/LibUsbDotNet/src/LibUsbDotNet/Generated/HotplugCallbackFn.cs(37,45,37,68): error CS0117: 'NativeMethods' does not contain a definition for 'LibUsbCallingConvention'
        /Users/mcuee/build/libusb/LibUsbDotNet/src/LibUsbDotNet/Generated/TransferDelegate.cs(37,45,37,68): error CS0117: 'NativeMethods' does not contain a definition for 'LibUsbCallingConvention'
        /Users/mcuee/build/libusb/LibUsbDotNet/src/LibUsbDotNet/Generated/PollfdAddedDelegate.cs(37,45,37,68): error CS0117: 'NativeMethods' does not contain a definition for 'LibUsbCallingConvention'
...
```

mcuee, 2021-06-05:
Interestingly the old 2.2.8 binary still works under my Mac Mini M1 with Rosetta 2 and Homebrew x86_64 Mono.

```
build/libusb/LibUsbDotNet_Bin.2.2.8 ❯ mono BenchmarkCon.exe vid=0x04d8 pid=0xfa2e
WARNING: The runtime version supported by this application is unavailable.
Using default runtime: v4.0.30319
LibUsbDotNet USB Benchmark v2.2.8.104
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
Benchmark device 04D8:FA2E opened..
Loop Test Information
	Vid / Pid       : 04D8h / FA2Eh
	Interface #     : 00h
	Priority        : Normal
	Buffer Size     : 4096
	Buffer Count    : 1
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
Bulk Write (Ep01h) max packet size: 32

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..

Avg. Bytes/s: 307378.55 Transfers: 68 Bytes/s: 307378.55
Avg. Bytes/s: 303850.26 Transfers: 142 Bytes/s: 300678.73
Avg. Bytes/s: 300323.81 Transfers: 214 Bytes/s: 293603.41
Avg. Bytes/s: 303059.95 Transfers: 291 Bytes/s: 310932.91
Avg. Bytes/s: 301035.61 Transfers: 363 Bytes/s: 293122.20
Avg. Bytes/s: 301424.38 Transfers: 437 Bytes/s: 303346.07
Avg. Bytes/s: 302214.12 Transfers: 512 Bytes/s: 306899.24
qAvg. Bytes/s: 298860.72 Transfers: 580 Bytes/s: 275816.99
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
	Vid / Pid       : 04D8h / FA2Eh
	Interface #     : 00h
	Priority        : Normal
	Buffer Size     : 4096
	Buffer Count    : 1
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
	Total Bytes     : 2379776
	Total Transfers : 581
	Avg. Bytes/sec  : 298453.02
	Elapsed Time    : 7.97 seconds

Bulk Write (Ep01h) max packet size: 32
	Total Bytes     : 2379776
	Total Transfers : 581
	Avg. Bytes/sec  : 298844.74
	Elapsed Time    : 7.96 seconds

Press any key to exit..
```

mcuee, 2021-06-05:
Mono info.
```
build/libusb/LibUsbDotNet_Bin.2.2.8 took 11s ❯ which mono
/usr/local/bin/mono
build/libusb/LibUsbDotNet_Bin.2.2.8 ❯ mono --version
Mono JIT compiler version 6.12.0.122 (tarball Wed Feb 24 16:02:31 GMT 2021)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:
	SIGSEGV:       altstack
	Notification:  kqueue
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug
	Interpreter:   yes
	LLVM:          supported, not enabled.
	Suspend:       hybrid
	GC:            sgen (concurrent by default)
```

mcuee, 2021-06-05:
I can actually Mono xbuild to build 2.2.8 source.

```

BenchmarkCon/bin/Debug ❯ which xbuild
/Library/Frameworks/Mono.framework/Versions/Current/Commands/xbuild
BenchmarkCon/bin/Debug ❯ ls
BenchmarkCon.exe        BenchmarkCon.pdb        LibUsbDotNet.pdb
BenchmarkCon.exe.config LibUsbDotNet.dll
BenchmarkCon/bin/Debug ❯ cp /usr/local/lib/libusb-1.0.dylib .
BenchmarkCon/bin/Debug ❯ ls
BenchmarkCon.exe        BenchmarkCon.pdb        LibUsbDotNet.pdb
BenchmarkCon.exe.config LibUsbDotNet.dll        libusb-1.0.dylib
BenchmarkCon/bin/Debug ❯ mono BenchmarkCon.exe vid=0x04d8 pid=0xfa2e
WARNING: The runtime version supported by this application is unavailable.
Using default runtime: v4.0.30319
LibUsbDotNet USB Benchmark v2.2.8.104
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
Benchmark device 04D8:FA2E opened..
Loop Test Information
	Vid / Pid       : 04D8h / FA2Eh
	Interface #     : 00h
	Priority        : Normal
	Buffer Size     : 4096
	Buffer Count    : 1
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
Bulk Write (Ep01h) max packet size: 32

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..

Avg. Bytes/s: 306858.06 Transfers: 68 Bytes/s: 306858.06
Avg. Bytes/s: 297491.27 Transfers: 140 Bytes/s: 289155.21
Avg. Bytes/s: 299738.26 Transfers: 215 Bytes/s: 304024.77
Avg. Bytes/s: 300723.37 Transfers: 289 Bytes/s: 303622.59
Avg. Bytes/s: 300756.17 Transfers: 363 Bytes/s: 300884.38
Avg. Bytes/s: 300108.49 Transfers: 435 Bytes/s: 296885.10
Avg. Bytes/s: 300569.01 Transfers: 510 Bytes/s: 303268.13
qAvg. Bytes/s: 300907.29 Transfers: 584 Bytes/s: 303259.57
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
	Vid / Pid       : 04D8h / FA2Eh
	Interface #     : 00h
	Priority        : Normal
	Buffer Size     : 4096
	Buffer Count    : 1
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
	Total Bytes     : 2396160
	Total Transfers : 585
	Avg. Bytes/sec  : 300919.80
	Elapsed Time    : 7.96 seconds

Bulk Write (Ep01h) max packet size: 32
	Total Bytes     : 2396160
	Total Transfers : 585
	Avg. Bytes/sec  : 300899.25
	Elapsed Time    : 7.96 seconds

Press any key to exit..
```

mcuee, 2021-06-06:
Is the following considered a success for the build process under macOS? Sorry I have no clue on dotnet under macOS.

```
libUsbdotNet on master ❯ dotnet pack src/LibUsbDotNet/LibUsbDotNet.csproj -c Release
Microsoft (R) Build Engine version 16.10.0+4242f381a for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
  Restored /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/LibUsbDotNet.csproj (in 676 ms).
  1 of 2 projects are up-to-date for restore.
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netstandard2.0/win-x86/LibUsbDotNet.dll
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netstandard2.0/win-x64/LibUsbDotNet.dll
  LibUsbDotNet.LibUsb.refs -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/refs/bin/Release/netstandard2.0/LibUsbDotNet.dll
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netcoreapp2.1/linux-arm64/LibUsbDotNet.dll
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netcoreapp2.1/win-x86/LibUsbDotNet.dll
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netcoreapp2.1/linux-x64/LibUsbDotNet.dll
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netcoreapp2.1/osx-x64/LibUsbDotNet.dll
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netcoreapp2.1/win-x64/LibUsbDotNet.dll
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netstandard2.0/linux-x64/LibUsbDotNet.dll
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netstandard2.0/osx-x64/LibUsbDotNet.dll
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netstandard2.0/linux-arm/LibUsbDotNet.dll
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netcoreapp2.1/linux-arm/LibUsbDotNet.dll
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netstandard2.0/linux-arm64/LibUsbDotNet.dll
  LibUsbDotNet.LibUsb.refs -> /Users/mcuee/build/libusb/libusbdotnet_backup/LibUsbDotNet/src/refs/bin/Release/netstandard2.0/LibUsbDotNet.dll
```

mcuee, 2021-06-06:
My main target is actually to run the examples as I have the associated HW (Benchmark USB device) to carry out the tests.

mcuee, 2021-06-06:
Right now the examples seem to point to .NET Framework 4.5, and not .Net 5.0, or .NetCore 3.1/5.0.

qmfrederik, 2021-06-07:
@mcuee "Modern" C#/.NET code should work just fine when compiled for arm/rpi.

This project (still) uses `MSBuild.Sdk.Extras`, an add-on to the build process, in an attempt to work around some issues which existed when building cross-platform .NET code in the early days of .NET Core.

Updating this has been on my backburner for quite a while (admittedly, too long). Let me see if I can free up time the coming weeks to get this done. It will make things easier for everyone.

The goal still is:
- To make the .NET bindings just "tiny" wrappers around the native, C code
- Be compatible with the libusb packages which exist for the various platforms
- Support all platforms which .NET Core supports
- Ship with a compiled version of libusb for platforms which don't ship with libusb

mcuee, 2021-06-07:
@qmfrederik That will be great. I would like to use BenchmarkCon to test both libusbdotnet and libusb in Windows/Linux/Mac, I did that last time when I was able to build libusbdotnet using SharpDevelop under Windows and MonoDevelop under Linux/Mac.

mcuee, 2021-06-08:
There is a warning when building main libusbdotnet library.

```

LibUsbDotNet on  master [!] ❯ dotnet pack src/LibUsbDotNet/LibUsbDotNet.csproj -c Release
Microsoft (R) Build Engine version 16.10.0+4242f381a for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
  All projects are up-to-date for restore.
  LibUsbDotNet -> /Users/mcuee/build/libusb/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netcoreapp3.1/LibUsbDotNet.dll
/usr/local/share/dotnet/sdk/5.0.300/Sdks/NuGet.Build.Tasks.Pack/buildCrossTargeting/NuGet.Build.Tasks.Pack.targets(221,5): warning NU5048: The 'PackageIconUrl'/'iconUrl' element is deprecated. Consider using the 'PackageIcon'/'icon' element instead. Learn more at https://aka.ms/deprecateIconUrl [/Users/mcuee/build/libusb/LibUsbDotNet/src/LibUsbDotNet/LibUsbDotNet.csproj]
  Successfully created package '/Users/mcuee/build/libusb/LibUsbDotNet/src/LibUsbDotNet/bin/Release/LibUsbDotNet.3.0.97-alpha.nupkg'.
  Successfully created package '/Users/mcuee/build/libusb/LibUsbDotNet/src/LibUsbDotNet/bin/Release/LibUsbDotNet.3.0.97-alpha.snupkg'.

LibUsbDotNet on  master [!] took 4s ❯ ls -la src/LibUsbDotNet/bin/Release                        
total 96
drwxr-xr-x  5 mcuee  staff    160 Jun  8 20:09 .
drwxr-xr-x  5 mcuee  staff    160 Jun  8 20:09 ..
-rw-r--r--  1 mcuee  staff  26955 Jun  8 20:09 LibUsbDotNet.3.0.97-alpha.nupkg
-rw-r--r--  1 mcuee  staff  18796 Jun  8 20:09 LibUsbDotNet.3.0.97-alpha.snupkg
drwxr-xr-x  5 mcuee  staff    160 Jun  8 20:09 netcoreapp3.1

```

qmfrederik, 2021-06-08:
@mcuee Yes, you are correct. The benchmark library uses async transfers which haven't been ported yet to the 3.0 series. That's causing the errors you are seeing.

On a related note - I'd like to add some sort of integration tests to LibUsbDotNet. Do you have any tips for that? (E.g. a QEMU VM with emulated USB devices, a Linux kernel driver which loads phony devices,...)?

mcuee, 2021-06-09:
Yes from the very beginning of working with Travis Robinson on libusb-win32/libusbdotnet/libusbk back in 2009/2010 time, we were thinking about how to carry out tests, in the end he settled on using PIC MCU based USB examples as both of us were familiar with PIC USB stuff. Later Tarvis ported it to AVR32 as we need high speed USB examples as well and I got the support from Microchip/Atmel with some tools for Travis at that time. Pete Batard (original developer of libusb-1.0 Windows backend) actually ported the example to ARM as well.
https://github.com/mcuee/libusbk/tree/master/BmFW

I admit I knew very little about dotnet so I was not able to support it, but I did use the libusbdotnet benchmark example extensively over the years as a testing tool for my engagement in libusb-1.0 project as it is cross-platform. It is great that you took over libusbdotnet project from Travis.

I was also involved in pyusb and we had similar discussion as well. Right now Jonas has to disable the HW based tests as he does not have the necessary HW. Original pyusb developer adopted the same PIC MCU based approach but new developer does not have the HW.
https://github.com/pyusb/pyusb/issues/235

mcuee, 2021-06-09:
Using a QEMU VM to emulate the USB device may be a good idea. Or maybe a Raspberry Pi running Linux USB Gadget driver.

qmfrederik, 2021-06-09:
Thanks, @mcuee . Both the Linux loopback driver and the QEMU approach look interesting, then. I'll explore them a bit further.

The QEMU/VirtualBox approach has the advantage that it should be trivial to set up a "standard" test environment which we can use on developer desktops and in CI systems, and we can run the test on different platforms (e.g. Windows or Linux as a guest).

The Linux USB gadget driver/USB loopback device look very interesting because they involve much less overhead. On the other hand, it may be more difficult to deploy to CI environments (we'll have to test).

qmfrederik, 2021-06-09:
> but I did use the libusbdotnet benchmark example extensively over the years as a testing tool for my engagement in libusb-1.0 project as it is cross-platform

As it stands, the benchmark example doesn't work on the master (v3) branch because async transfers are not implemented. .NET has a new model for async I/O and I would like to see if we can use that model. This will take some time.

I'll have a look at the v2 branch, and see if I can the v2 version to compile correctly with .NET core. It should enable you to use the benchmark tool on all platforms which .NET Core supports. I only want to make changes related to build tooling on the v2 branch, the master branch is for refactoring work.

mcuee, 2021-06-09:
Please refer to #138.

Now you can use features/v2-dotnetcore-sdk branch under macOS if you need async examples.

mcuee, 2021-06-10:
BTW, benchmarkcon can be very flexible and test any FW with loopback FW (interrupt transfer, bulk transfer and isochronous transfer).

Test done back in 2013.

With Cypress's cystream firmware and tested under Mac OS X
Lion 10.7.5 using a Mac Mini 2011 (lowest spec version but
upgradedto 8GB RAM).

Firmware: http://www.cypress.com/?app=forum&id=167&rID=48493
( http://www.cypress.com/?docID=35838 )

Board: An ADC demo board from ADI, just take out the
EEPROM of the Cypress EZ-USB FX2LP and download
the above firmware. The one from Cypress's original
download does not seem to work with a bare board.

1. Bulk IN:
```
mymacmini:libusbdotnet xiaofanc$ mono benchmarkcon.exe notestselect
mode=async buffercount=3 buffersize=512000 read list intf=0 altf=2
LibUsbDotNet USB Benchmark v2.2.9.110
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
1. 0AC8:3420 Vimicro Corp. - PROLiNK PCC3220
2. 0424:2513
3. 05AC:8006
4. 04B4:1003 Cypress - CY-Stream
5. 046D:C52B Logitech - USB Receiver
6. 05E3:0606 USB2.0 Hub
7. 04F2:0760 Chicony - USB Keyboard
8. 0424:2513
9. 05AC:8006

Select device (1-9) :4

Benchmark device 04B4:1003 opened..
Read Test Information
        Vid / Pid       : 04B4h / 1003h
        Interface #     : 00h
        Alt Interface # : 02h
        Priority        : Normal
        Buffer Size     : 512000
        Buffer Count    : 3
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : Off

Bulk Read (Ep82h) max packet size: 512

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..

Avg. Bytes/s: 42361222.34 Transfers: 78 Bytes/s: 42361222.34
Avg. Bytes/s: 42320001.89 Transfers: 161 Bytes/s: 42281337.65
Avg. Bytes/s: 42310223.27 Transfers: 244 Bytes/s: 42291267.97
Avg. Bytes/s: 42305613.48 Transfers: 327 Bytes/s: 42292067.65
Avg. Bytes/s: 42297725.02 Transfers: 409 Bytes/s: 42266296.59
Avg. Bytes/s: 42297154.03 Transfers: 492 Bytes/s: 42294340.58
Avg. Bytes/s: 42295392.80 Transfers: 575 Bytes/s: 42284955.79
Avg. Bytes/s: 42297763.00 Transfers: 657 Bytes/s: 42314390.76
Avg. Bytes/s: 42301449.86 Transfers: 740 Bytes/s: 42330656.46
Avg. Bytes/s: 42302915.56 Transfers: 823 Bytes/s: 42315987.79
Avg. Bytes/s: 42301713.23 Transfers: 906 Bytes/s: 42289794.96
Avg. Bytes/s: 42301286.45 Transfers: 988 Bytes/s: 42296571.66
[... 131 lines cut ...]
Avg. Bytes/s: 33760922.01 Transfers: 2901 Bytes/s: 33723810.46
Avg. Bytes/s: 33759345.95 Transfers: 2966 Bytes/s: 33689154.78
qAvg. Bytes/s: 33758200.16 Transfers: 3032 Bytes/s: 33706789.24
waiting for Ep06h thread..
stopped Ep06h thread.
Write Test Information
        Vid / Pid       : 04B4h / 1003h
        Interface #     : 00h
        Alt Interface # : 02h
        Priority        : Normal
        Buffer Size     : 512000
        Buffer Count    : 3
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : Off

Bulk Write (Ep06h) max packet size: 512
        Total Bytes     : 1552896000
        Total Transfers : 3033
        Avg. Bytes/sec  : 33758522.83
        Elapsed Time    : 46.00 seconds

Press any key to exit..
```

mcuee, 2021-06-15:
Somehow this no longer works (CyStreamer FW) since I can not change to a different alternative interface. I will need to check again.

mcuee, 2021-06-27:
I have made some progress with Linux USB Gadget here. Still learning how to configure the usb gadget. The HW devices are really cheap (eg: Raspberry Pi Zero, Orange Pi Zero).
https://github.com/pyusb/pyusb/issues/235

mcuee, 2021-07-22:
@qmfrederik I will highly recommend the following US$49 Cypress FX3 development board.
https://www.cypress.com/documentation/development-kitsboards/cyusb3kit-003-ez-usb-fx3-superspeed-explorer-kit

mcuee, 2021-07-22:
Benchmarkcon results under my Mac Mini M1 (8GB RAM, 256GB SSD).

```

libusbdotnet_v2 on  v2 took 18s ❯ ./stage/BenchmarkCon/bin/Release/net5.0/BenchmarkCon list notestselect write buffercount=3 buffersize=32768 loop mode=async
LibUsbDotNet USB Benchmark v2.2.0.0
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
1. 04B4:00F1 Cypress - FX3
2. 046D:C52B Logitech - USB Receiver
3. 047F:C025 Plantronics C320-M
4. 1A40:0201 
5. 0AC8:3420 Vimicro Corp. - PROLiNK PCC3220
6. 1A40:0201 
7. 0BDA:5411 Generic - 4-Port USB 2.0 Hub
8. 05E3:0749 Generic - USB3.0 Card Reader
9. 0BDA:0411 Generic - 4-Port USB 3.0 Hub
10. 0BDA:0411 Generic - 4-Port USB 3.0 Hub
11. 0BDA:5411 Generic - 4-Port USB 2.0 Hub

Select device (1-11) :1

Benchmark device 04B4:00F1 opened..
Loop Test Information
	Vid / Pid       : 04B4h / 00F1h
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 32768
	Buffer Count    : 3
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 1024
Bulk Write (Ep01h) max packet size: 1024

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..
Avg. Bytes/s: 281226203.00 Transfers: 8086 Bytes/s: 281226203.00
Avg. Bytes/s: 288942931.94 Transfers: 17199 Bytes/s: 296153452.65
Avg. Bytes/s: 291884573.12 Transfers: 26336 Bytes/s: 297587420.67
Avg. Bytes/s: 293362093.71 Transfers: 35470 Bytes/s: 297707201.40
Avg. Bytes/s: 294178361.38 Transfers: 44600 Bytes/s: 297393131.68
Avg. Bytes/s: 294741105.78 Transfers: 53733 Bytes/s: 297520426.37
Avg. Bytes/s: 295190625.51 Transfers: 62869 Bytes/s: 297862455.43
Avg. Bytes/s: 295488757.12 Transfers: 72000 Bytes/s: 297557922.46
Avg. Bytes/s: 295693625.44 Transfers: 81130 Bytes/s: 297319243.79
Avg. Bytes/s: 295910920.67 Transfers: 90252 Bytes/s: 297857662.74
Avg. Bytes/s: 296073306.84 Transfers: 99385 Bytes/s: 297687636.52
Avg. Bytes/s: 296204760.63 Transfers: 108519 Bytes/s: 297642661.24
Avg. Bytes/s: 296320007.38 Transfers: 117654 Bytes/s: 297695970.59
Avg. Bytes/s: 296410631.73 Transfers: 126781 Bytes/s: 297583831.34
qAvg. Bytes/s: 296493732.55 Transfers: 135920 Bytes/s: 297651376.26
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
	Vid / Pid       : 04B4h / 00F1h
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 32768
	Buffer Count    : 3
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 1024
	Total Bytes     : 4453859328
	Total Transfers : 135921
	Avg. Bytes/sec  : 296490308.47
	Elapsed Time    : 15.02 seconds

Bulk Write (Ep01h) max packet size: 1024
	Total Bytes     : 4822728704
	Total Transfers : 147178
	Avg. Bytes/sec  : 321044531.44
	Elapsed Time    : 15.02 seconds

Press any key to exit..

libusbdotnet_v2 on  v2 took 18s ❯ ./stage/BenchmarkCon/bin/Release/net5.0/BenchmarkCon list notestselect write buffercount=3 buffersize=32768 loop mode=async
LibUsbDotNet USB Benchmark v2.2.0.0
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
1. 04B4:00F1 Cypress - FX3
2. 046D:C52B Logitech - USB Receiver
3. 047F:C025 Plantronics C320-M
4. 1A40:0201 
5. 0AC8:3420 Vimicro Corp. - PROLiNK PCC3220
6. 1A40:0201 
7. 0BDA:5411 Generic - 4-Port USB 2.0 Hub
8. 05E3:0749 Generic - USB3.0 Card Reader
9. 0BDA:0411 Generic - 4-Port USB 3.0 Hub
10. 0BDA:0411 Generic - 4-Port USB 3.0 Hub
11. 0BDA:5411 Generic - 4-Port USB 2.0 Hub

Select device (1-11) :1

Benchmark device 04B4:00F1 opened..
Loop Test Information
	Vid / Pid       : 04B4h / 00F1h
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 32768
	Buffer Count    : 3
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 1024
Bulk Write (Ep01h) max packet size: 1024

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..
Avg. Bytes/s: 281226203.00 Transfers: 8086 Bytes/s: 281226203.00
Avg. Bytes/s: 288942931.94 Transfers: 17199 Bytes/s: 296153452.65
Avg. Bytes/s: 291884573.12 Transfers: 26336 Bytes/s: 297587420.67
Avg. Bytes/s: 293362093.71 Transfers: 35470 Bytes/s: 297707201.40
Avg. Bytes/s: 294178361.38 Transfers: 44600 Bytes/s: 297393131.68
Avg. Bytes/s: 294741105.78 Transfers: 53733 Bytes/s: 297520426.37
Avg. Bytes/s: 295190625.51 Transfers: 62869 Bytes/s: 297862455.43
Avg. Bytes/s: 295488757.12 Transfers: 72000 Bytes/s: 297557922.46
Avg. Bytes/s: 295693625.44 Transfers: 81130 Bytes/s: 297319243.79
Avg. Bytes/s: 295910920.67 Transfers: 90252 Bytes/s: 297857662.74
Avg. Bytes/s: 296073306.84 Transfers: 99385 Bytes/s: 297687636.52
Avg. Bytes/s: 296204760.63 Transfers: 108519 Bytes/s: 297642661.24
Avg. Bytes/s: 296320007.38 Transfers: 117654 Bytes/s: 297695970.59
Avg. Bytes/s: 296410631.73 Transfers: 126781 Bytes/s: 297583831.34
qAvg. Bytes/s: 296493732.55 Transfers: 135920 Bytes/s: 297651376.26
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
	Vid / Pid       : 04B4h / 00F1h
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 32768
	Buffer Count    : 3
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 1024
	Total Bytes     : 4453859328
	Total Transfers : 135921
	Avg. Bytes/sec  : 296490308.47
	Elapsed Time    : 15.02 seconds

Bulk Write (Ep01h) max packet size: 1024
	Total Bytes     : 4822728704
	Total Transfers : 147178
	Avg. Bytes/sec  : 321044531.44
	Elapsed Time    : 15.02 seconds

Press any key to exit..

```

mcuee, 2021-09-16:
> Thanks, @mcuee . Both the Linux loopback driver and the QEMU approach look interesting, then. I'll explore them a bit further.
>
> The QEMU/VirtualBox approach has the advantage that it should be [...]

@qmfrederik 
Maybe the dummy_hcd module will make it work. That is a software emulated device controller which works on normal Linux PC.
Please take a look at this discussion: https://github.com/vpelletier/python-functionfs/issues/20

mcuee, 2022-05-08:
Reference: https://github.com/libusb/libusb/wiki/FAQ#Testing_libusb

closed 2023-02-06 by mcuee