LibUsbDotNet/LibUsbDotNet #138: How to build under ARM Linux

mcuee, 2021-06-05:
I have just installed MonoDevelop and Dotnet SDK 3.1/5.0 under a Debian 10 ARM Linux virtual machine. But it seems to me I can not build libusbdotnet.

/home/debian/.dotnet/sdk/5.0.300/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.TargetFrameworkInference.targets(5,5): Error MSB4186: Invalid static method invocation syntax: "[MSBuild]::GetTargetPlatformIdentifier('$(TargetFramework)')". Method '[MSBuild]::GetTargetPlatformIdentifier' not found. Static method invocation should be of the form: $([FullTypeName]::Method()), e.g. $([System.IO.Path]::Combine(`a`, `b`)). Check that all parameters are defined, are of the correct type, and are specified in the right order. (MSB4186) (LibUsbDotNet)

mcuee, 2021-06-05:
Interestingly the old libusbdotnet 2.2.8 binary release still works even though I can not use MonoDevelope to build from source.

```
benchmark vid=0x4D2 pid=0x162E buffercount=3 buffersize=0x2000
➜  bin sudo mono BenchmarkCon.exe vid=0x04d8 pid=0xfa2e
WARNING: The runtime version supported by this application is unavailable.
Using default runtime: v4.0.30319
LibUsbDotNet USB Benchmark v2.2.8.104
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
Benchmark device 04D8:FA2E opened..
Loop Test Information
	Vid / Pid       : 04D8h / FA2Eh
	Interface #     : 00h
	Priority        : Normal
	Buffer Size     : 4096
	Buffer Count    : 1
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
Bulk Write (Ep01h) max packet size: 32

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..

Avg. Bytes/s: 283858.23 Transfers: 65 Bytes/s: 283858.23
Avg. Bytes/s: 284239.75 Transfers: 135 Bytes/s: 284594.93
Avg. Bytes/s: 284173.36 Transfers: 205 Bytes/s: 284045.43
Avg. Bytes/s: 284366.03 Transfers: 275 Bytes/s: 284931.77
Avg. Bytes/s: 284632.48 Transfers: 345 Bytes/s: 285684.11
Avg. Bytes/s: 284581.16 Transfers: 414 Bytes/s: 284324.83
Avg. Bytes/s: 284593.73 Transfers: 484 Bytes/s: 284668.11
Avg. Bytes/s: 284298.46 Transfers: 553 Bytes/s: 282244.38
qAvg. Bytes/s: 284137.59 Transfers: 622 Bytes/s: 282854.81
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
	Vid / Pid       : 04D8h / FA2Eh
	Interface #     : 00h
	Priority        : Normal
	Buffer Size     : 4096
	Buffer Count    : 1
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
	Total Bytes     : 2551808
	Total Transfers : 623
	Avg. Bytes/sec  : 284125.14
	Elapsed Time    : 8.98 seconds

Bulk Write (Ep01h) max packet size: 32
	Total Bytes     : 2551808
	Total Transfers : 623
	Avg. Bytes/sec  : 284125.52
	Elapsed Time    : 8.98 seconds

Press any key to exit..

```

mcuee, 2021-06-07:
It seems that I have better luck with Raspberry Pi 400 running Ubuntu 64bit.

```
libusbdotnet on  master ❯ dotnet pack src/LibUsbDotNet/LibUsbDotNet.csproj -c Release
Microsoft (R) Build Engine version 16.10.0+4242f381a for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
  Restored /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/refs/LibUsbDotNet.LibUsb.refs.csproj (in 9.77 sec).
  Restored /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/LibUsbDotNet.csproj (in 1.11 min).
  LibUsbDotNet -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/bin/Release/netstandard2.0/win-x86/LibUsbDotNet.dll
  LibUsbDotNet -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/bin/Release/netstandard2.0/linux-x64/LibUsbDotNet.dll
  LibUsbDotNet -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/bin/Release/netstandard2.0/linux-arm64/LibUsbDotNet.dll
  LibUsbDotNet -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/bin/Release/netstandard2.0/win-x64/LibUsbDotNet.dll
  LibUsbDotNet -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/bin/Release/netstandard2.0/osx-x64/LibUsbDotNet.dll
  LibUsbDotNet.LibUsb.refs -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/refs/bin/Release/netstandard2.0/LibUsbDotNet.dll
  LibUsbDotNet -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/bin/Release/netstandard2.0/linux-arm/LibUsbDotNet.dll
  LibUsbDotNet -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/bin/Release/netcoreapp2.1/win-x86/LibUsbDotNet.dll
  LibUsbDotNet -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/bin/Release/netcoreapp2.1/win-x64/LibUsbDotNet.dll
  LibUsbDotNet -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/bin/Release/netcoreapp2.1/linux-arm64/LibUsbDotNet.dll
  LibUsbDotNet -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/bin/Release/netcoreapp2.1/linux-x64/LibUsbDotNet.dll
  LibUsbDotNet -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/bin/Release/netcoreapp2.1/linux-arm/LibUsbDotNet.dll
  LibUsbDotNet -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/LibUsbDotNet/bin/Release/netcoreapp2.1/osx-x64/LibUsbDotNet.dll
  LibUsbDotNet.LibUsb.refs -> /home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/refs/bin/Release/netstandard2.0/LibUsbDotNet.dll
(lots of wanrnings skipped)
```

mcuee, 2021-06-07:
However the examples can not be built.

```
libusbdotnet on  master ❯ dotnet build src/BenchmarkCon/BenchmarkCon.csproj -c Release
Microsoft (R) Build Engine version 16.10.0+4242f381a for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
  All projects are up-to-date for restore.
/home/mcuee/dotnet/sdk/5.0.300/Microsoft.Common.CurrentVersion.targets(1216,5): error MSB3644: The reference assemblies for .NETFramework,Version=v4.5 were not found. To resolve this, install the Developer Pack (SDK/Targeting Pack) for this framework version or retarget your application. You can download .NET Framework Developer Packs at https://aka.ms/msbuild/developerpacks [/home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/BenchmarkCon/BenchmarkCon.csproj]

Build FAILED.

/home/mcuee/dotnet/sdk/5.0.300/Microsoft.Common.CurrentVersion.targets(1216,5): error MSB3644: The reference assemblies for .NETFramework,Version=v4.5 were not found. To resolve this, install the Developer Pack (SDK/Targeting Pack) for this framework version or retarget your application. You can download .NET Framework Developer Packs at https://aka.ms/msbuild/developerpacks [/home/mcuee/build/libusb/libusbdotnet/libusbdotnet/src/BenchmarkCon/BenchmarkCon.csproj]
    0 Warning(s)
    1 Error(s)

```

qmfrederik, 2021-06-07:
@mcuee Things should be a bit better by now. You should be able to build the main LibUsbDotNet library on all supported .NET platforms by just running `dotnet build` using the latest version of the .NET (Core) SDK, including ARM Linux.

I haven't yet fixed all examples, though.

mcuee, 2021-06-08:
Yes it is okay to build the main LibUsbDotNet library now.

```
LibUsbDotNet on  master took 3s ❯ dotnet pack src/LibUsbDotNet/LibUsbDotNet.csproj -c Release
Microsoft (R) Build Engine version 16.10.0+4242f381a for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
  Restored /home/mcuee/build/libusb/LibUsbDotNet/src/LibUsbDotNet/LibUsbDotNet.csproj (in 476 ms).
  LibUsbDotNet -> /home/mcuee/build/libusb/LibUsbDotNet/src/LibUsbDotNet/bin/Release/netcoreapp3.1/LibUsbDotNet.dll
/home/mcuee/dotnet/sdk/5.0.300/Sdks/NuGet.Build.Tasks.Pack/buildCrossTargeting/NuGet.Build.Tasks.Pack.targets(221,5): warning NU5048: The 'PackageIconUrl'/'iconUrl' element is deprecated. Consider using the 'PackageIcon'/'icon' element instead. Learn more at https://aka.ms/deprecateIconUrl [/home/mcuee/build/libusb/LibUsbDotNet/src/LibUsbDotNet/LibUsbDotNet.csproj]
  Successfully created package '/home/mcuee/build/libusb/LibUsbDotNet/src/LibUsbDotNet/bin/Release/LibUsbDotNet.3.0.97-alpha.nupkg'.
  Successfully created package '/home/mcuee/build/libusb/LibUsbDotNet/src/LibUsbDotNet/bin/Release/LibUsbDotNet.3.0.97-alpha.snupkg'.
LibUsbDotNet on  master took 15s ❯ ls -la ./src/LibUsbDotNet/bin/Release 
total 60
drwxrwxr-x 3 mcuee mcuee  4096 Jun  8 20:17 .
drwxrwxr-x 3 mcuee mcuee  4096 Jun  8 20:17 ..
-rw-rw-r-- 1 mcuee mcuee 26958 Jun  8 20:17 LibUsbDotNet.3.0.97-alpha.nupkg
-rw-rw-r-- 1 mcuee mcuee 18794 Jun  8 20:17 LibUsbDotNet.3.0.97-alpha.snupkg
drwxrwxr-x 2 mcuee mcuee  4096 Jun  8 20:17 netcoreapp3.1

```

mcuee, 2021-06-08:
Working build under macOS
src/LibUsbDotNet/LibUsbDotNet.csproj
src/LibUsbDotNet.Tests/LibUsbDotNet.Tests.csproj
src/Examples/Read.Only/Read.Only.csproj
src/Examples/Read.Write/Read.Write.csproj 
src/Examples/Show.Info/Show.Info.csproj

qmfrederik, 2021-06-09:
@mcuee With #141 , you should be able to build `BenchmarkConsole` _from the v2 branch_ using .NET Core on Raspberry Pi.

mcuee, 2021-06-09:
Great. I will try v2 branch under Linux, macOS and Windows.

Under Windows 10 and VS2019 all seem to build fine.

mcuee, 2021-06-09:
Under macOS, it is strange that BenchmarkCon becomes a dll.
```

libusbdotnet_v2 on  v2 took 3s ❯ dotnet build stage/BenchmarkCon/BenchmarkCon.csproj -c Release
Microsoft (R) Build Engine version 16.10.0+4242f381a for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
  Restored /Users/mcuee/build/libusb/libusbdotnet_v2/stage/BenchmarkCon/BenchmarkCon.csproj (in 4.34 sec).
  Restored /Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj (in 4.34 sec).
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/Internal/Kernel32.cs(89,62): warning CS0618: 'UnmanagedType.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Please specify the type you wish to marshal as.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/Internal/Kernel32.cs(99,62): warning CS0618: 'UnmanagedType.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Please specify the type you wish to marshal as.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/Internal/Kernel32.cs(119,70): warning CS0618: 'UnmanagedType.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Please specify the type you wish to marshal as.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/Internal/SetupApi.cs(239,77): warning CS0618: 'UnmanagedType.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Please specify the type you wish to marshal as.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/Internal/SetupApi.cs(296,81): warning CS0618: 'UnmanagedType.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Please specify the type you wish to marshal as.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/WinUsb/Internal/WinUsbAPI.cs(111,78): warning CS0618: 'UnmanagedType.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Please specify the type you wish to marshal as.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsb/Internal/LibUsbRequest.cs(71,30): warning CS0618: 'Marshal.ReadByte(object, int)' is obsolete: 'ReadByte(Object, Int32) may be unavailable in future releases.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/Main/UsbDeviceFinder.cs(261,20): warning SYSLIB0011: 'BinaryFormatter.Deserialize(Stream)' is obsolete: 'BinaryFormatter serialization is obsolete and should not be used. See https://aka.ms/binaryformatter for more information.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/Main/UsbDeviceFinder.cs(272,13): warning SYSLIB0011: 'BinaryFormatter.Serialize(Stream, object)' is obsolete: 'BinaryFormatter serialization is obsolete and should not be used. See https://aka.ms/binaryformatter for more information.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/MonoLibUsb/MonoUsbEventHandler.cs(188,25): warning SYSLIB0006: 'Thread.Abort()' is obsolete: 'Thread.Abort is not supported and throws PlatformNotSupportedException.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
  LibUsbDotNet -> /Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/bin/Release/net5.0/LibUsbDotNet.LibUsbDotNet.dll
  BenchmarkCon -> /Users/mcuee/build/libusb/libusbdotnet_v2/stage/BenchmarkCon/bin/Release/net5.0/BenchmarkCon.dll

Build succeeded.

/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/Internal/Kernel32.cs(89,62): warning CS0618: 'UnmanagedType.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Please specify the type you wish to marshal as.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/Internal/Kernel32.cs(99,62): warning CS0618: 'UnmanagedType.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Please specify the type you wish to marshal as.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/Internal/Kernel32.cs(119,70): warning CS0618: 'UnmanagedType.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Please specify the type you wish to marshal as.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/Internal/SetupApi.cs(239,77): warning CS0618: 'UnmanagedType.AsAny' is obsolete: 'Marshalling arbitrary types may be unavailable in future releases. Please specify the type you wish to ma [... cut ...] a.ms/binaryformatter for more information.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/MonoLibUsb/MonoUsbEventHandler.cs(188,25): warning SYSLIB0006: 'Thread.Abort()' is obsolete: 'Thread.Abort is not supported and throws PlatformNotSupportedException.' [/Users/mcuee/build/libusb/libusbdotnet_v2/stage/LibUsbDotNet/LibUsbDotNet.csproj]
    10 Warning(s)
    0 Error(s)

Time Elapsed 00:00:14.30
```

qmfrederik, 2021-06-09:
@mcuee Yes, the .NET team decided to produce libraries and not executables on macOS because of notarization requirements: https://docs.microsoft.com/en-us/dotnet/core/install/macos-notarization-issues#apphost-is-disabled-by-default.

There are a couple of options which should work:

- Build with `dotnet build -p:UseAppHost=true`, this should create an executable
- Invoke `dotnet run` or `dotnet bin/Release/net5.0/BenchmarkCon.dll` to launch the console app.
- Or we update the project file and add `<UseAppHost>true</UseAppHost>` inside a `PropertyGroup`.

Let me know if either of those options work for you.

mcuee, 2021-06-09:
Great. I tried Option 1 and Option 2 and both are working under my Raspberry Pi 400. It should be the same for macOS.

Only thing is that the new BenchmarkCon no longer prints out the help text when running without an argument. A bit strange.

```
bin/Release/net5.0 v2 > ./BenchmarkCon 
LibUsbDotNet USB Benchmark v2.2.0.0
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet

bin/Release/net5.0 v2 > dotnet BenchmarkCon.dll vid=0x04d8 pid=0xfa2e
LibUsbDotNet USB Benchmark v2.2.0.0
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
Benchmark device 04D8:FA2E opened..
Loop Test Information
	Vid / Pid       : 04D8h / FA2Eh
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 4096
	Buffer Count    : 1
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
Bulk Write (Ep01h) max packet size: 32

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..
Avg. Bytes/s: 318764.54 Transfers: 71 Bytes/s: 318764.54
Avg. Bytes/s: 318924.33 Transfers: 149 Bytes/s: 319069.92
Avg. Bytes/s: 318866.72 Transfers: 226 Bytes/s: 318755.28
Avg. Bytes/s: 317619.77 Transfers: 303 Bytes/s: 314015.58
Avg. Bytes/s: 316840.20 Transfers: 380 Bytes/s: 313809.33
q
Avg. Bytes/s: 316400.05 Transfers: 456 Bytes/s: 314217.53
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
	Vid / Pid       : 04D8h / FA2Eh
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 4096
	Buffer Count    : 1
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
	Total Bytes     : 1871872
	Total Transfers : 457
	Avg. Bytes/sec  : 316400.55
	Elapsed Time    : 5.92 seconds

Bulk Write (Ep01h) max packet size: 32
	Total Bytes     : 1871872
	Total Transfers : 457
	Avg. Bytes/sec  : 316393.38
	Elapsed Time    : 5.92 seconds

Press any key to exit..

```

qmfrederik, 2021-06-09:
Ah, good catch! #142 should fix help text.

mcuee, 2021-06-09:
Great. Now it is all good. The following is from my Mac Mini M1.

```

libusbdotnet_v2core on  features/v2-dotnetcore-sdk took 8s ❯ dotnet ./stage/BenchmarkCon/bin/Release/net5.0/BenchmarkCon.dll
LibUsbDotNet USB Benchmark v2.2.0.0
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet

USAGE: benchmark [list] 
                 [pid=] [vid=] [intf=] [altf=] [ep=]
                 [read|write|loop] [notestselect]
                 [verify|verifydetail]
                 [retry=] [timeout=] [refresh=] [priority=]
                 [mode=] [buffersize=] [buffercount=] [packetsize=]
                 [driver=]
                 
Commands:
         list  : Display a list of connected devices before starting. 
                 Select the device to use for the test from the list.
         read  : Read from the device.
         write : Write to the device.
         loop  : [Default] Read and write to the device at the same time.

         notestselect : Skips submitting the control transfers to get/set the
                        test type.  This makes the application compatible
                        with non-benchmark firmwared. Use at your own risk!

         verify       : Verify received data for loop and read tests. Report
                        basic information on data validation errors.
         verifydetail : Same as verify except reports detail information for 
                        each byte that fails validation.
                        
Switches:
         vid        : Vendor id of device. (hex)  (Default=1234)
         pid        : Product id of device. (hex) (Default=5678)
         retry      : Number of times to retry a transfer that timeout.
                      (Default = 0)
         timeout    : Transfer timeout value. (milliseconds) (Default=5000)
                      The timeout value used for read/write operations. If a
                      transfer times out more than {retry} times, the test 
                      fails and the operation is aborted.
         mode       : Sync|Async (Default=Sync) 
                      Sync uses the libusb-win32 sync transfer functions.
                      Async uses the libusb-win32 asynchronous api.
         buffersize : Transfer test size in bytes. (Default=4096)
                      Increasing this value will generally yield higher
                      transfer rates.
         buffercount: (Async mode only) Number of outstanding transfers on
                      an endpoint (Default=1, Max=10). Increasing this value
                      will generally yield higher transfer rates.
         refresh    : The display refresh interval. (in milliseconds)
                      (Default=1000) This also effect the running status.
         priority   : AboveNormal|BelowNormal|Highest|Lowest|Normal
                      (Default=Normal) The thread priority level to use
                      for the test.
         ep         : The loopback endpoint to use. For example ep=0x01, would
                      read from 0x81 and write to 0x01. (default is to use the
                      (first read/write endpoint(s) in the interface)
         intf       : The interface id the read/write endpoints reside in.
         altf       : The alternate interface setting number to select.
         packetsize : For isochronous use only. Sets the iso packet size.
                      If not specified, the endpoints maximum packet size
                      is used.
         driver     : winusb|libusb-win32|libusb10.  Forces the use a specific
                      driver.  If not specified, all drivers are used.
         
WARNING:
          This program should only be used with USB devices which implement
          one more more "Benchmark" interface(s).  Using this application
[... 50 lines cut ...]
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 4096
	Buffer Count    : 1
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
	Total Bytes     : 2506752
	Total Transfers : 612
	Avg. Bytes/sec  : 314606.63
	Elapsed Time    : 7.97 seconds

Bulk Write (Ep01h) max packet size: 32
	Total Bytes     : 2506752
	Total Transfers : 612
	Avg. Bytes/sec  : 314601.66
	Elapsed Time    : 7.97 seconds

Press any key to exit..

```

mcuee, 2021-06-10:
BTW, it also works under Windows. 
Under Windows v2 actually works very well, including the GUI programs like Benchmark and Test_Info, which does not see to build under Linux and macOS.

mcuee, 2021-07-23:
Results with my Raspberry Pi 400 running Ubuntu Linux 64bit, testing with Cypress FX3 USB 3 board, pretty decent result at about 373MB/sec.

```
libusbdotnet_v2 on  v2 took 20s ❯ ./stage/BenchmarkCon/bin/Release/net5.0/BenchmarkCon list notestselect read mode=async buffercount=4 buffersize=65536
LibUsbDotNet USB Benchmark v2.2.0.0
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
1. 1D6B:0002 
2. 04B4:00F1 Cypress - FX3
3. 05E3:0749 
4. 0BDA:0411 
5. 0BDA:0411 
6. 1D6B:0003 
7. 04D9:0007 
8. 0AC8:3420 
9. 047F:C025 
10. 1A40:0201 
11. 046D:C52B 
12. 1A40:0201 
13. 0BDA:5411 
14. 0BDA:5411 
15. 2109:3431 
16. 1D6B:0002 

Select device (1-16) :2

Benchmark device 04B4:00F1 opened..
Read Test Information
	Vid / Pid       : 04B4h / 00F1h
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 65536
	Buffer Count    : 4
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 1024

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..
Avg. Bytes/s: 374415952.56 Transfers: 5179 Bytes/s: 374415952.56
Avg. Bytes/s: 370520588.99 Transfers: 10791 Bytes/s: 366997006.65
Avg. Bytes/s: 372141772.30 Transfers: 16518 Bytes/s: 375235328.47
Avg. Bytes/s: 371945427.83 Transfers: 22192 Bytes/s: 371375012.68
Avg. Bytes/s: 372438382.53 Transfers: 27906 Bytes/s: 374365373.45
Avg. Bytes/s: 372280690.07 Transfers: 33580 Bytes/s: 371507062.71
Avg. Bytes/s: 372390137.73 Transfers: 39273 Bytes/s: 373037023.20
Avg. Bytes/s: 372664102.40 Transfers: 44990 Bytes/s: 374557057.25
Avg. Bytes/s: 372691097.41 Transfers: 50684 Bytes/s: 372904530.71
Avg. Bytes/s: 373044564.68 Transfers: 56427 Bytes/s: 376193352.34
Avg. Bytes/s: 373105077.82 Transfers: 62130 Bytes/s: 373704870.67
Avg. Bytes/s: 373318233.85 Transfers: 67864 Bytes/s: 375643571.90
Avg. Bytes/s: 373302116.83 Transfers: 73562 Bytes/s: 373110267.64
qAvg. Bytes/s: 373382075.28 Transfers: 79277 Bytes/s: 374414346.29
stopped Ep81h thread.
Read Test Information
	Vid / Pid       : 04B4h / 00F1h
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 65536
	Buffer Count    : 4
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 1024
	Total Bytes     : 5195563008
	Total Transfers : 79278
	Avg. Bytes/sec  : 373364063.48
	Elapsed Time    : 13.92 seconds

Press any key to exit..

```

qmfrederik, 2021-07-23:
Thanks @mcuee !

MatCDev, 2023-01-25:
Sorry for digging out this conversation.
The tests you performed @mcuee , was performed with which driver for your USB device ?
When you say you tested on MacOS and Windows, are you on native ARM systems ?
If I unedrstood correctly you used a raspberry pi on which you installed an arm compatible Windows iso? And you did the same test with an arm compatible linux iso. While for macOs, you had a macOS arm available and tested directly on it.
The idea behind these questions is that I would like to be able to use a device whose driver is WinUSB on a Windows ARM virtual machine emulated via Parallels on MacOS arm. Did this scenario fit into your tests ? 
Lastly, do I need to build my LibUsbDotNet app or dll on ARM for this to work (I guess so but apparently Microsoft is able to handle some x86-64 applications by emulating CPU behavior on an ARM CPU) ?

mcuee, 2023-01-27:
@MatCDev 

Linux: tested on both 64bit x86_64 and ARM32/ARM64 (Raspberry Pi 400)
macOS: tested on ARM64 (Mac Mini M1)
Windows: tested on Windows 10/11 x86_64 only. I do not have access to Windows on ARM64. 

Somehow to me Windows on ARM64 is a dead end and I do not think it will fly.

mcuee, 2023-01-27:
@MatCDev 

If you have access to Windows on ARM64, please try it to see if libusbdotnet works or not. Thanks. By right it should work if the device is using WinUSB driver. If you develop your own firmware, make sure you use Microsoft OS Descriptors for USB Devices so that the OS will automatically pick WinUSB as the driver.

Ref: 
https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/microsoft-defined-usb-descriptors
https://github.com/pbatard/libwdi/wiki/WCID-Devices

MatCDev, 2023-01-27:
@mcuee 
Thank you for the information.
The purpose of supporting arm is to allow the user working under macOS to use our wpf application via virtualization of a windows arm (perhaps it is possible to virtualize a Win x86 on a mac arm? I will check this).
So I don't have direct access to a Windows ARM but to a Windows ARM virtual machine running on a mac M1. 
As far as firmware descriptors are concerned, everything seems perfect at this level. I manage to communicate perfectly with my tool on different instances of Windows x86-64. And the device is naturally recognized by the system as a winusb device (with WCID). LibUsbDotNet manages to recognize it and communicate with it since the last modification I made to it (PR: [https://github.com/LibUsbDotNet/LibUsbDotNet/pull/186](https://github.com/LibUsbDotNet/LibUsbDotNet/pull/186)).
I will therefore continue my research and will keep you informed of my progress.
Last question, during your tests on MacOS, did you use winusb usb descriptors in the firmware of your usb device or did you go through zadig or another driver generation system/software ?

mcuee, 2023-01-27:
@MatCDev 

I see. I tend to believe it will work.

For libusb and libusbdotnet to work under macOS, you do not need to have Microsoft OS Descriptors for USB Devices, rather make sure it as a generic USB device and no other kernel driver are loaded for the device with exclusive access. It is a difficult task to use libusb under macOS in case there is already a kernel extension loaded for the device.

You can also refer to libusb FAQ here.
https://github.com/libusb/libusb/wiki/FAQ#user-content-How_can_I_run_libusb_applications_under_Mac_OS_X_if_there_is_already_a_kernel_extension_installed_for_the_device_and_claim_exclusive_access

dwadi, 2023-02-05:
@mcuee I have Parallels setup and I can try libusb build for Windows 11 ARM, I am looking for solution for my RTL SDR, is there any build tutorial for such driver?

mcuee, 2023-02-06:
> @mcuee I have Parallels setup and I can try libusb build for Windows 11 ARM, I am looking for solution for my RTL SDR, is there any build tutorial for such driver?

@dwadi 
Please help to get the libusbdotnet v2 branch tested out on your Windows 11 ARM computer. Thanks. 

Sorry I am not familiar with RTL SDR. As of now, Zadig does not work under Windows on ARM, so you need a generic device which use WinUSB driver to test it out. The best is probably those generic USB device with the WCID descriptor.

Ref:
* https://github.com/pbatard/libwdi/issues/155 (there are people who are interested in RTL-SDR here so you may want to discuss with them).
* https://github.com/pbatard/libwdi/wiki/WCID-Devices

dwadi, 2023-02-06:
>    - pbatard/libwdi#155 <https://github.com/pbatard/libwdi/issues/155>

Hi, I only have SDR's that needs drivers to be replaced, other device that
use WinUSB drivers is PS Vita, i could help if you give me instructions
what should be done, you could even run Parallels in trial mode and create
Windows 11, it is very simple and all resources are present from Parallels
software,like Win 11 ARM or sth like Ubuntu.

mcuee, 2023-02-06:
> Hi, I only have SDR's that needs drivers to be replaced, other device that use WinUSB drivers is PS Vita, i could help if you give me instructions what should be done, you could even run Parallels [...]

Personaly I have zero interests for Windows on ARM64 so I will not test it our myself.

Since Zadig will not work under Windows on ARM64 -- so you can not use your SDR device under Windows on ARM64.

dwadi, 2023-02-06:
It is only case on Zadig? Is there any other options there? If you wanna
test for Linux, try Trial version of Parallels, I think you should have
around 2 weeks.

mcuee, 2023-02-06:
> It is only case on Zadig? Is there any other options there? 

The first thing you need to use a USB HW under Windows on ARM64 is to have the device recognized under the machine with WinUSB driver (or maybe HID driver). Since Zadig does not work yet for Windows on ARM64, you have to modify your device FW to use WCID (Microsoft OS Descriptors for USB Devices).
* https://github.com/pbatard/libwdi/wiki/WCID-Devices
* https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/microsoft-defined-usb-descriptors

> If you wanna test for Linux, try Trial version of Parallels, I think you should have around 2 weeks.

Thanks for the info. As of now I am not interested in using Parallels. I have used UTM before to running ARM Linux VMs under my Mac Mini M1. But as mentioned, I have proper ARM Linux 32/64 machines like Raspberry Pi 400 so I do not really need to use ARM Linux under my Mac Mini M1.

closed 2023-02-06 by mcuee

MatCDev, 2023-02-06:
Last version of LibUsbDotNet (which contain my last modification linked to WCID WinUSB device) is working on Windows 11 ARM running on a rpi400 (installed via Windows on Raspberry) ✔️

mcuee, 2023-02-06:
> Last version of LibUsbDotNet (which contain my last modification linked to WCID WinUSB device) is working on Windows 11 ARM running on a rpi400 (installed via Windows on Raspberry) ✔️

Great. Thanks a lot for the reports.

MatCDev, 2023-02-06:
Still have problems with Windows 11 ARM virtualized via Parallels on Mac M1. Ongoing research...

mcuee, 2023-02-06:
> Still have problems with Windows 11 ARM virtualized via Parallels on Mac M1. Ongoing research...

You may want to check with Parallels to see if they support USB or not in this case.

MatCDev, 2023-02-06:
The library works on vm Windows 11 ARM virtualized via Parallels on Mac M1. However I do not go through Zadig or other things to assign the driver (because they are not ported to ARM). I therefore use USB descriptors on the firmware side of my device which are WinUSB descriptors (so works with WinUSB device having the WCID)
Here's the info.