LibUsbDotNet/LibUsbDotNet #143: Please read the Wiki and FAQ before asking questions

mcuee, 2021-06-11:
Wiki:
https://github.com/LibUsbDotNet/LibUsbDotNet/wiki

FAQ
https://github.com/LibUsbDotNet/LibUsbDotNet/issues?q=is%3Aopen+is%3Aissue+label%3AFAQ

mcuee, 2021-06-11:
When you raise an issue, it would be good to state the version of libusbdotnet, the OS and the details or your device and code snippets. Thanks.

mcuee, 2021-06-12:
I am happy to join the project to help on the testing and supporting side. I was also on the testing side last time with Travis Robinson (libusbdotnet original author). Take note I am not the developer myself and I knew very little about dotnet, even though I know quite a bit about libusb as a non-developer admin of libusb project and libusb-win32/libusbk/pyusb project.

BryanCrotaz, 2021-06-23:
Please link the wiki and faq in the readme - it's the front page of the github site, so make it easy to find documentation?

mcuee, 2021-06-25:
> Please link the wiki and faq in the readme - it's the front page of the github site, so make it easy to find documentation?
Personally I think it is not necessary. 

But you are welcome to create pull request to improve the README or other documentations.

BryanCrotaz, 2021-06-25:
You may not, but I gave up on this project and moved to Device.Net because I couldn't work out how to use it after a day of trying.

mcuee, 2021-06-25:
It is perfectly fine if you find that Device.Net easier to use for you. I can not even build Device.Net myself.

BryanCrotaz, 2021-06-25:
The point is that it's documented and I had it up and running via Nuget in
under an hour. As a developer I don't care about compiling it - that's why
Nuget exists.

closed 2025-04-12 by mcuee

mcuee, 2025-04-12:
Focus of the project will be on the master (v3) branch.

v2 branch will not get new features, only bug fixes PR may be considered.