LibUsbDotNet/LibUsbDotNet #145: Documentation appears to be out of date

BryanCrotaz, 2021-06-23:
From documentation:

```
using System;
using System.Windows.Forms;
using LibUsbDotNet.DeviceNotify;

namespace Device.Notification
{
    internal class DeviceNotification
    {
        public static IDeviceNotifier UsbDeviceNotifier = DeviceNotifier.OpenDeviceNotifier();

        private static void Main(string[] args)
        {
            // Hook the device notifier event
            UsbDeviceNotifier.OnDeviceNotify += OnDeviceNotifyEvent;

            // Exit on and key pressed.
            Console.Clear();            
            Console.WriteLine();
            Console.WriteLine("Waiting for system level device events..");
            Console.Write("[Press any key to exit]");

            while (!Console.KeyAvailable)
                Application.DoEvents();

            UsbDeviceNotifier.Enabled = false;  // Disable the device notifier

            // Unhook the device notifier event
            UsbDeviceNotifier.OnDeviceNotify -= OnDeviceNotifyEvent;
        }

        private static void OnDeviceNotifyEvent(object sender, DeviceNotifyEventArgs e)
        {
            // A Device system-level event has occured

            Console.SetCursorPosition(0,Console.CursorTop);

            Console.WriteLine(e.ToString()); // Dump the event info to output.

            Console.WriteLine();
            Console.Write("[Press any key to exit]");
        }
    }
}
```

My code:
```
using System;
using LibUsbDotNet.Main;
using LibUsbDotNet.DeviceNotify;

namespace LibUSBTest
{
    class Program
    {
        static void Main(string[] args)
        {
            var devices = LibUsbDotNet.UsbDevice.AllDevices;
            foreach (UsbRegistry device in devices)
            {
                Console.WriteLine("--- Device "+device.FullName);
            }

            var notifier = DeviceNotifier.OpenDeviceNotifier();  // DeviceNotifier does not exist in the current context

            Console.ReadLine();
        }
    }
}
```

Using in a brand new .Net Core 2.1 Console Application, testing on Windows for deployment on Raspbian on Raspberry Pi.

BryanCrotaz, 2021-06-23:
What's the new way to get a DeviceNotifier in 2.2.29?

mcuee, 2021-06-24:
https://github.com/LibUsbDotNet/LibUsbDotNet/issues/32
I believe DeviceNotifier got lots of issues.

mcuee, 2021-06-25:
But I have just tested the latest v2 branch and the sample seems to work fine.

```
[DeviceType:DeviceInterface] [EventType:DeviceRemoveComplete] FullName:USB#VID_04D8&PID_FA2E#LUSBW1#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Vid:0x04D8
Pid:0xFA2E
SerialNumber:LUSBW1
ClassGuid:a5dcbf10-6530-11d2-901f-00c04fb951ed
[DeviceType:DeviceInterface] [EventType:DeviceArrival] FullName:USB#VID_04D8&PID_FA2E#LUSBW1#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Vid:0x04D8
Pid:0xFA2E
SerialNumber:LUSBW1
ClassGuid:a5dcbf10-6530-11d2-901f-00c04fb951ed

```

mcuee, 2021-06-25:
So I will close this for now and please refer to #32.

closed 2021-06-25 by mcuee