LibUsbDotNet/LibUsbDotNet #146: GetDescriptor Failed on pi 4

chensiel, 2021-06-30:
Device: PI4 + usb barcode reader
OS:pi os 32bit
dotnet: 5.0 sdk

var reader=OpenEndpointReader(LibUsbDotNet.Main.ReadEndpointID.Ep01);
reader.DataReceived += (p1, p2) => { };
reader.ReadBufferSize = 8;
reader.DataReceivedEnabled = true;

Works fine on windows 10 with winusb driver

Get error message GetDescriptor Failed from LibUsbDotNet.UsbDevice.UsbErrorEvent on pi os

Has anybody tested it on pi4?

mcuee, 2021-06-30:
You may have to detach the kernel hid driver first under Linux.

You can use udev rule to detach the kernel driver.
Ref: https://github.com/LibUsbDotNet/LibUsbDotNet/issues/128

chensiel, 2021-06-30:
@mcuee Thank you.

I have tried as follow

create a file named 90-kbd-LibUsbDotNet.rules at /etc/udev/rules.d/

#
# udev rules for LibUsbDotNet (libusb)
# copy this file to `/etc/udev/rules.d/` and run `sudo udevadm control --reload-rules && sudo udevadm trigger`
#

# [required] prevent usbhid from claiming barcode scanner's HID interface
ACTION=="add", SUBSYSTEM=="usb", DRIVER=="usbhid", ATTRS{idVendor}=="8888", ATTRS{idProduct}=="0f1f", RUN+="/bin/sh -c 'echo $kernel >/sys/bus/usb/drivers/usbhid/unbind'"

 and run `sudo udevadm control --reload-rules && sudo udevadm trigger`

reboot

It does not work.

lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8888:0f1f  USB2.0 Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

ls /sys/bus/usb/drivers/
brcmfmac  hub  lan78xx  smsc95xx  usb  usbfs  usbhid

Is there anything wrong?

chensiel, 2021-06-30:
I can receive data from my scanner now. 
Thank you very much.
But LibUsbDotNet.UsbDevice.UsbErrorEvent still show the same message 'GetDescriptor Failed'
I can ignore that

closed 2021-06-30 by chensiel

mcuee, 2023-04-15:
PR #203 should have fix this issue. I will close this for now.

@chensiel 

Please try latest v2 branch to see if you still have issues or not. If you still have the issues, please reopen.