LibUsbDotNet/LibUsbDotNet #147: Example of Detaching the Kernel Driver?

Krutonium, 2021-08-07:
I can't figure out how to do it, but it [Looks Possible?](https://github.com/LibUsbDotNet/LibUsbDotNet/commit/8c4e7eb5f4630bf8fa54a79f9d40287562e2f671) 

I just can't figure out how.

mcuee, 2022-05-15:
Try this fork for the v2 branch.
https://github.com/itayperl/LibUsbDotNet/commits/v2

mcuee, 2023-01-06:
@Krutonium 

Please try latest v2 branch to see if it works for you or not as PR #172 has been merged.
* #172

mcuee, 2023-04-22:
Please refer to the example here for the master branch.
https://github.com/LibUsbDotNet/LibUsbDotNet/pull/165#issuecomment-1501383699

PR #165 and PR #212 have been merged for the master branch.

mcuee, 2023-04-22:
For v2 branch, PR #172 hasbeen merged.

closed 2023-04-22 by mcuee