LibUsbDotNet/LibUsbDotNet #151: Can't connect to Samsung device in Download mode

TheAirBlow, 2021-11-28:
Code is [here](https://github.com/Samsung-Loki/Hreidmar).
It always spits out "Timeout error".

Double-checked everything. Everything is as intended.
What am I doing wrong?

mcuee, 2021-11-29:
Ref: there is an issue with the device, Heimdall and Hreidmar codes need to be modified. Not a problem with libusb or libusbdotnet.
https://github.com/Benjamin-Dobell/Heimdall/issues/497

https://github.com/libusb/libusb/issues/1024#issuecomment-963048935
> OK. I don't think libusb should do anything different here, it is just the "messenger". Heimdall must try to workaround this by not calling libusb_set_interface(), either when it seems superfluous (no other alternate interface) or when it can identify a flaky bootloader version (this might be difficult if the USB descriptors are identical).

closed 2021-11-29 by mcuee

TheAirBlow, 2021-11-29:
It was thrown out only on `_writer.Write()`. Irrelevant.

TheAirBlow, 2021-11-29:
I want to reopen this, but I can't. why

mcuee, 2021-11-30:
Continued on
- https://github.com/LibUsbDotNet/LibUsbDotNet/issues/152