LibUsbDotNet/LibUsbDotNet #158: Cannot open libusbK device

MatCDev, 2022-05-11:
Hello,
The issue is that when I try to open a device working with the libusbK driver (installed via Zadig), the opening does not work. However when I install the WinUSB driver everything works fine. My application runs under Windows 10 and the .NET Framework 4.7.2 and I use version 2.2.29 of the library. 
My goal was for my application to work regardless of the driver installed (WinUSB or libusbK). According to the [Wiki](https://github.com/LibUsbDotNet/LibUsbDotNet/wiki) of the library it would seem that it supports however well the devices functioning with the driver libusbK.
Actually my testing source code is really simple : 
``` C#
            UsbRegDeviceList winUsbDevices = UsbDevice.AllWinUsbDevices;
            Debug.WriteLine("Count :" + winUsbDevices.Count);
            foreach (UsbRegistry usbRegistry in winUsbDevices)
            {
                Debug.WriteLine("PROPERTIES");
                usbRegistry.DeviceProperties.Select(i => $"{i.Key}: {i.Value}").ToList().ForEach(Console.WriteLine);
                Debug.WriteLine("END OF PROPERTIES");
                Debug.WriteLine("IsAlive: "+usbRegistry.IsAlive);
                Debug.WriteLine("IsOpen: "+usbRegistry.Open(out mUsbDevice));
            }
```
When I'm using libusbK as driver, everything works fine (i.e. IsAlive = true) until it reaches the Open function which return false and mUsbDevice is null...
(With WinUSB: IsOpen = true and I can use mUsbDevice after that)

In the code of my application I do not use UsbRegistry but rather UsbDeviceFinder based on the VID and the PID of my device and I use the OpenUsbDevice function of the UsbDevice class. However that doesn't work either. This is why I first wanted to do some tests and I used the registry to find out in which type of device (winusb or libusb) my device was when the libusbK driver was installed on it (it seems whether it is a "winusb" device). + I checked that I only had my device contained in my UsbRegDeviceList winUsbDevices when I was testing this (my count = 1 and the properties match those of my device) . 

I do not know if the issue comes from me or from the library but I preferred to warn because I've been stuck on this issue for several days and I could not find how you manage the libusbK devices in your library.

Do you have the same issue on your side ?

mcuee, 2022-05-11:
Have you tried the examples to see if they work with your device?
Example: 
https://github.com/LibUsbDotNet/LibUsbDotNet/tree/v2/stage/Examples/Show.Info

MatCDev, 2022-05-11:
I just tested. Everything works fine with the WinUSB driver, in the console is displayed the information on the device, on the configuration, then on the interface and the endpoints of this one.
When I test with the libusbK driver nothing is displayed in the console. When I add an "else" to the first if of the foreach (the one that corresponds to Open). When I run the program again I enter my "else" which says my Open returns me false and It shows that mUsbDevice is null.
Source code : 
```
           public static UsbDevice MyUsbDevice;

            // Dump all devices and descriptor information to console output.
            UsbRegDeviceList allDevices = UsbDevice.AllDevices;
            foreach (UsbRegistry usbRegistry in allDevices)
            {
                if (usbRegistry.Open(out MyUsbDevice))
                {
                    Console.WriteLine(MyUsbDevice.Info.ToString());
                    for (int iConfig = 0; iConfig < MyUsbDevice.Configs.Count; iConfig++)
                    {
                        UsbConfigInfo configInfo = MyUsbDevice.Configs[iConfig];
                        Console.WriteLine(configInfo.ToString());

                        ReadOnlyCollection<UsbInterfaceInfo> interfaceList = configInfo.InterfaceInfoList;
                        for (int iInterface = 0; iInterface < interfaceList.Count; iInterface++)
                        {
                            UsbInterfaceInfo interfaceInfo = interfaceList[iInterface];
                            Console.WriteLine(interfaceInfo.ToString());

                            ReadOnlyCollection<UsbEndpointInfo> endpointList = interfaceInfo.EndpointInfoList;
                            for (int iEndpoint = 0; iEndpoint < endpointList.Count; iEndpoint++)
                            {
                                Console.WriteLine(endpointList[iEndpoint].ToString());
                            }
                        }
                    }
                }
                else
                {
                    Console.WriteLine("The device is not open");
                    Console.WriteLine(MyUsbDevice==null?"Device null":"Device not null");
                }
            }

            // Free usb resources.
            // This is necessary for libusb-1.0 and Linux compatibility.
            UsbDevice.Exit();
```
Console output when driver is libusbK : 
```
The device is not open
Device null
```

mcuee, 2022-05-11:
This is very strange then. Looks like your device may have problems with libusbk driver. 

Just to rule out it is libusbdotnet issue, can you test with libusbk itself? Please test with klist utility from libusbk project (which comes with the installer).
Ref: https://github.com/mcuee/libusbk/releases/tag/V3.1.0.0
Install libusbK-3.1.0.0-setup.exe to get klist which will list the info of all USB devices with libusbk.sys, libusb0.sys or WinUSB driver.

MatCDev, 2022-05-11:
When I use the libusbK library with the libusbK driver (this is what is currently working on our device on the market) everything works correctly. Moreover, if I use the libusbK library with the WinUSB driver, everything also works.
The idea behind this "dual stack" WinUSB LibusbK was ultimately to use only WinUSB as driver and completely break away from libusbK (driver as library).

mcuee, 2022-05-11:
Since kList.exe display your device informations when you use libusbK or WinUSB as driver, then your device is okay with either driver (WinUSB or libusbk.sys) and libusbk.dll is working fine with either driver.

So it is just libusbdotnet which does not work with libusbk driver, that is very strange. Sorry I am out of idea as of now.

mcuee, 2022-05-11:
Basically libusbdotnet uses either the following to talk to USB devices with supported driver under Windows.
1) WinUSB device (through WinUSB.dll)
https://github.com/LibUsbDotNet/LibUsbDotNet/tree/v2/stage/LibUsbDotNet/WinUsb/Internal

2) libusb-win32 libusb0.sys device (I believe it is directly talking to libusb0.sys using IOCTLs)
https://github.com/LibUsbDotNet/LibUsbDotNet/tree/v2/stage/LibUsbDotNet/LibUsb/Internal

3) libusbk libusbk.sys device --> I belive it is the same as libusb0.sys as libusbk.sys IOCTLs are a superset of libusbk.sys, and libusbdotnet has not been optimized for libusbk.sys yet. 

4) It talked to USB devices through libusb-1.0 library under other OS like Linux and macOS.

MatCDev, 2022-05-11:
Indeed, it is a really weird situation... So I also think that the issue comes from libusbdotnet. 
Have you already tested devices with the library (libusbdotnet) and the driver (libusbK) ?
If so is it possible to have the testing protocol or somehting like that ?
Moreover would it be possible to know where the fact that the driver is libusbK is managed at the level of the source code of the library and the changes that this entails ? I've searched but cannot find anything...
If I can help solve this issue, I will try to do it.
Thank you for your reactivity and your replies.

mcuee, 2022-05-11:
Yes for my test device, libusbdotnet works with all three drivers -- libusb0.sys, libusbk.sys and WinUSB.
Ref: https://github.com/LibUsbDotNet/LibUsbDotNet/issues/126#issuecomment-858228460

mcuee, 2022-05-11:
@qmfrederik Please help to take a look at this one to see how to debug the issue. Thanks.

mcuee, 2022-05-13:
You are right, somehow libusbk driver support is now broken. Strange.

mcuee, 2022-05-13:
In the mean time, you can use either libusb-win32 driver (libusb0,sys) or WinUSB driver,

With libusb0,sys, there is a limitation as mentioned in the Readme file.
> LibUsbDotNet 2.1.0 and above supports the genuine libusb-win32 driver package. However, access to basic device information via the windows registry is not available. See the LegacyUsbRegistry class for more information.

mcuee, 2022-05-13:
@TravisRo As this is related to the compatibility of libusbk.sys with libusb0.sys, please take a look as well if you have some time. Thanks.

mcuee, 2022-05-14:
I do not remember whether I test libusbk driver last time or not.

Not so sure if it is because of this line. This '0x8FF' IOCTL code is not available under libusbk.

https://github.com/LibUsbDotNet/LibUsbDotNet/blob/v2/stage/LibUsbDotNet/LibUsb/Internal/LibUsbIoCtl.cs#L61
`public static readonly int GET_OBJECT_NAME = CTL_CODE(FILE_DEVICE_UNKNOWN, 0x8FF, METHOD_BUFFERED, FILE_ANY_ACCESS);`

mcuee, 2022-05-14:
Not so sure if it is because of this line. This '0x8FF' IOCTL code for libusb0.sys is not available under libusbk.sys. 

https://github.com/LibUsbDotNet/LibUsbDotNet/blob/v2/stage/LibUsbDotNet/LibUsb/Internal/LibUsbIoCtl.cs#L61
`public static readonly int GET_OBJECT_NAME = CTL_CODE(FILE_DEVICE_UNKNOWN, 0x8FF, METHOD_BUFFERED, FILE_ANY_ACCESS);`

mcuee, 2022-05-14:
@MatCDev What is the OS you are using? I am using Windows 11 and it has the problem. What is the Windows you are using? Just want to gather more information.

MatCDev, 2022-05-15:
Sorry for the delay.
I am actually using Windows 10 Family version.

MatCDev, 2022-05-15:
Do you think that this simple line could be at the origin of this new imcompatibility between the libusbdotnet library and the libusbK driver ?
Is there a way to see what differences there may have been in the libusbK driver management code of the libusbdotnet library between when the tests were pass and now (when they no longer pass) ?

mcuee, 2022-05-15:
Sorry I can not help much here as I am not a developer and mainly help on the testing and technical support side of libusb, libusb-win32, libusbk and libusbdotnet project.

Travis is the developer of libusbk and the original author of libusbdotnet. Frederik is the current developer of libusbdotnet and he is familiar with libusbk as well. Hopefully they will take a look at this issue when they are available.

Another thing, take note WinUSB is the  Windows generic driver of choice now for libusbk and libusbdotnet projects. So you may want to focus on WinUSB driver. Then there are devices using libusb0.sys driver out there. There are not many devices with libusbk.sys driver.

MatCDev, 2022-05-16:
OK, no problem. Thank you for your availability and your feedback.
Yes I understand that WinUSB has become the most suitable solution. However, the tools I'm working on use libusbK as driver and library and the goal is to replace with WinUSB and a compatible library.
The best would then be to find a software static library (like libusbdotnet) which allowed to manage both libusbK devices and WinUSB devices, to continue to support the different devices at the application level.

MatCDev, 2022-05-16:
Do you have an idea of ​​the version of the library under which the tests with libusbK as driver passed ?

mcuee, 2022-05-16:
> Do you have an idea of ​​the version of the library under which the tests with libusbK as driver passed ? older than 2.2.8 ?

Sorry I can not remmeber now. It could be a combination of libusbk driver version and libusbdotnet version. It could even be that the combination never really worked.

FYI:
2.2.8 was released on 29-Oct-2010 and libusbk first official release 3.0.6 version was on 11-August-2013 (first beta version 3.0.5.5 was released on 11-Feb-2011).

MatCDev, 2022-05-16:
Ok ok, thank you !

mcuee, 2022-05-16:
Just to summarize -- libusbdotnet last official release (2.2.8.0) was released before libusbk project started. libusbdotnet was designed with libusb0.sys and WinUSB in mind, without libusbk.sys in picture.

However, supposed libusbk.sys is almost a drop in replacement of libusb0.sys, so by right it should work together. Somehow it does not. I am not so sure why.

Hopefully @TravisRo and @qmfrederik can chime in and help to explain the situation.

MatCDev, 2022-05-16:
(To keep you informed)
I think I just identified the problem and partially solved it. I will look into it during the day and come back to you as soon as possible.
I'm actually working on the v2 branch of the git project.
I did some changes in the source code and now when I start the "Test_Info" project, I can find my device and got the whole informations about it ! 😄

mcuee, 2022-05-16:
That will be great. Yes the V2 branch is the stable branch and it is good to go with V2 branch for production if you use crital features like async transfer (it is missing in the master branch).

Ref: 
* https://github.com/LibUsbDotNet/LibUsbDotNet/issues/144

MatCDev, 2022-06-02:
@mcuee @qmfrederik Hello (again), I have a local solution which provides management of libusbK devices. I don't know if you're interested in this one. If so, how do you want me to send it to you ? A fork of the project,... ? Since the library is under LGPL license, I have to publish the modifications, however how do you want that done ? 
I can also explain to you what was not working and what I put in place.

mcuee, 2022-06-02:
@MatCDev Please fork the project and then create a pull request. Thanks.

mcuee, 2023-01-06:
Close this issue as PR #162 has been merged in the v2 branch.
* #162

closed 2023-01-06 by mcuee