LibUsbDotNet/LibUsbDotNet #160: Blank device info on Linux with libusb 1.0

AnErrupTion, 2022-05-22:
I'm trying to get a simple example to work, here getting info of every USB device:
```cs
using LibUsbDotNet.LibUsb;

using var context = new UsbContext();

var devices = context.List();
foreach (var reg in devices)
{
    Console.WriteLine(reg.Info.ToString());
}
```

However, when launching the program, I just get this:
```
/home/anerruption/RiderProjects/OpenUSB/OpenUSB/bin/Debug/net6.0/OpenUSB
 
 
 
 
 
 
 
 

Process finished with exit code 0.
```

Everything is blank.
I'm using LibUsbDotNet 3.0.97-alpha with libusb 1.0 on Pop! OS 22.04. I built the program with .NET 6.

Location of my libusb binaries:
```
...flatpak apps shipping their own libusb...
/usr/lib/i386-linux-gnu/libusb-1.0.so.0.3.0
/usr/lib/i386-linux-gnu/libusb-1.0.so.0
/usr/lib/x86_64-linux-gnu/libusb-1.0.so.0
/usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0
/usr/lib/x86_64-linux-gnu/libusb-1.0.so
```

mcuee, 2022-05-23:
Did you try the example here and see if it works?
https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/src/Examples/Show.Info/ShowInfo.cs

Or the Test_Info application?
https://github.com/LibUsbDotNet/LibUsbDotNet/tree/master/src/Test_Info

AnErrupTion, 2022-05-23:
So I tried diving a bit deeper, and found that none of the USB devices were opened, and I could not open them. My USB webcam, or even my Logitech Unifying Receiver, they all can't be opened, nor are they already opened. The only thing I can get is their vendor and product IDs, and if they're opened, but that's about it (I think). any info from ``usbRegistry.Info`` is all empty, for some reason (or null).

AnErrupTion, 2022-05-23:
So, the solution from https://github.com/LibUsbDotNet/LibUsbDotNet/issues/128#issuecomment-855387445 seems to have fixed the opening issue and getting info.

But now, whenever I'm trying to set the configuration or claim an interface, I get an error saying "Resource busy". I've properly reloaded all rules, I don't understand why this happens.

AnErrupTion, 2022-05-23:
That indeed works much better, and I get a ton more info! Thanks!

closed 2022-05-27 by AnErrupTion