LibUsbDotNet/LibUsbDotNet #163: About reconnecting usb devices Prompts Safe handle has been closed.

GreenShadeZhang, 2022-06-25:
version：2.2.29
environment：windows 11 .net 6
issue ：
When I release and reconnect the device via code like this, the Safe handle has been closed.
[image]

The objects involved include reader,writer.

The sample code is as follows
https://github.com/GreenShadeZhang/ElectronBot.DotNet

GreenShadeZhang, 2022-06-25:
Add stack information

   在 System.Runtime.InteropServices.SafeHandle.DangerousAddRef(Boolean& success)
   在 System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)
   在 LibUsbDotNet.Internal.Kernel32.DeviceIoControl(SafeHandle hDevice, Int32 IoControlCode, Object InBuffer, Int32 nInBufferSize, IntPtr OutBuffer, Int32 nOutBufferSize, Int32& pBytesReturned, IntPtr pOverlapped)
   在 LibUsbDotNet.Internal.LibUsb.LibUsbAPI.ReadPipe(UsbEndpointBase endPointBase, IntPtr buffer, Int32 bufferLength, Int32& lengthTransferred, Int32 isoPacketSize, IntPtr pOverlapped)
   在 LibUsbDotNet.Main.UsbEndpointBase.ReadPipe(IntPtr pBuffer, Int32 bufferLength, Int32& lengthTransferred, Int32 isoPacketSize, IntPtr pOverlapped)
   在 LibUsbDotNet.Internal.OverlappedTransferContext.Submit()
   在 LibUsbDotNet.Main.UsbTransfer.SyncTransfer(UsbTransfer transferContext, IntPtr buffer, Int32 offset, Int32 length, Int32 timeout, Int32 isoPacketSize, Int32& transferLength)
   在 LibUsbDotNet.Main.UsbEndpointBase.Transfer(IntPtr buffer, Int32 offset, Int32 length, Int32 timeout, Int32& transferLength)
   在 LibUsbDotNet.Main.UsbEndpointBase.Transfer(Object buffer, Int32 offset, Int32 length, Int32 timeout, Int32& transferLength)
   在 LibUsbDotNet.UsbEndpointReader.Read(Byte[] buffer, Int32 timeout, Int32& transferLength)
   在 ElectronBot.DotNet.ElectronLowLevel.ReceivePacket(Int32 packetCount, Int32 packetSize) 在 C:\Users\gil\source\repos\ElectronBot.WinUI.DotNet\ElectronBot.DotNet\ElectronLowLevel.cs 中: 第 213 行
   在 ElectronBot.DotNet.ElectronLowLevel.SyncTask() 在 C:\Users\gil\source\repos\ElectronBot.WinUI.DotNet\ElectronBot.DotNet\ElectronLowLevel.cs 中: 第 267 行
   在 ElectronBot.DotNet.ElectronLowLevel.Sync() 在 C:\Users\gil\source\repos\ElectronBot.WinUI.DotNet\ElectronBot.DotNet\ElectronLowLevel.cs 中: 第 195 行
   在 ElectronBot.WinUI.MainWindow.MyButton_Click(Object sender, RoutedEventArgs e) 在 C:\Users\gil\source\repos\ElectronBot.WinUI.DotNet\ElectronBot.WinUI\MainWindow.xaml.cs 中: 第 48 行
   在 ABI.Microsoft.UI.Xaml.RoutedEventHandler.Do_Abi_Invoke(IntPtr thisPtr, IntPtr sender, IntPtr e)

mcuee, 2022-06-25:
@qmfrederik 
Please take a look if you got some time. Thanks.

GreenShadeZhang, 2022-07-01:
After I used this library, I found that this library is a good library, do you have any plans to continue to update it in the future? I want to use it under linux to connect usb devices, such as the Raspberry Pi zero 2w, is this possible?

mcuee, 2022-07-01:
> After I used this library, I found that this library is a good library, do you have any plans to continue to update it in the future? I want to use it under linux to connect usb devices, such as the Raspberry Pi zero 2w, is this possible?

Did you sort out the issues already? It should work under Linux and macOS as well as Windows. Have you tried it?

GreenShadeZhang, 2022-07-01:
> Did you sort out the issues already? It should work under Linux and macOS as well as Windows. Have you tried it?

Okay, I'll give it a try, and the problem of reconnecting the device after shutting down the device mentioned above can help confirm that it is not a problem for me to operate.

GreenShadeZhang, 2022-07-15:
Is the reason I'm talking about this problem because I'm using it the wrong way? Or is it really a bug.
 @qmfrederik

mcuee, 2023-04-23:
@GreenShadeZhang 

Can you try the latest v2 branch to see if this is still an issue or not. I think v2 branch device notification should work under Windows. But it may not work well under Linux or macOS.

mcuee, 2025-04-12:
De-focus on v2.

closed 2025-04-12 by mcuee