LibUsbDotNet/LibUsbDotNet #164: In Linux system LibUsbDotNet.Info.UsbBaseInfo.CustomDescriptors issue

vitalijusb85, 2022-07-01:
Hi, 
In the Linux system, LibUsbDotNet.Info.UsbBaseInfo.CustomDescriptors can't recognize all custom descriptors. For example, I add a print screen.  In Windows, the system works well.
[image]

mcuee, 2022-07-01:
Did you set up the right udev rules for your device?

vitalijusb85, 2022-07-01:
Hi,
yes, I set rules for the device:
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE="0666". 
The problems are the same.
But whit dfu-utils it works on Linux.

benedekkupper, 2022-07-02:
This is not specific to Linux, but more related to interfacing libusb driver instead of WinUsb. By manually replacing WinUsb driver to libusb-win32 (using Zadig tool), the above mentioned issue occurs on Windows as well.

mcuee, 2023-04-15:
PR #203 should have fix this issue. I will close this for now.

@vitalijusb85 

Please try latest v2 branch to see if you still have issues or not. If you still have the issues, please reopen.

PR #203 (2023-04-15): Add langId parameter to MonoUsbApi.GetDescriptor
MonoUsbDevice.GetDescriptor should pass the langId parameter to MonoUsbApi.GetDescriptor in order to have the correct parameters for the Control Transfer.

Otherwise, the Control Transfer fails with LIBUSB_ERROR_PIPE  (control request not supported) for all Descriptors EXCEPT Device Descriptor, Configuration Descriptor and String Descriptor index 0.

This pull request fixes null string descriptors in the UsbDevice.Info when LibUsbDotNet  is used under Linux or under Windows with libusb-1.0 backend (ForceLibUsbWinBack set true) ... This was my us [... cut]
closed 2023-04-15 by mcuee