LibUsbDotNet/LibUsbDotNet #166: Device is found, but read and write fail on Debian

dmitriyburdan, 2022-07-18:
Hello, 

I am trying to read/write to a device on Debian. I am able to successfully finding it, but UsbEndpointWriter.Write always returns MonoApiError. I am running code in Read.Write example and suspect I may not be opening the endpoint correctly.

Any help would be much appreciated.

Device info
----------------

Length:18
DescriptorType:Device
BcdUsb:0x0200
Class:PerInterface
SubClass:0x00
Protocol:0x00
MaxPacketSize0:64
VendorID:0x0801
ProductID:0x2023
BcdDevice:0x0100
ManufacturerStringIndex:1
ProductStringIndex:2
SerialStringIndex:3
ConfigurationCount:1
ManufacturerString:
ProductString:
SerialString:

Length:9
DescriptorType:Configuration
TotalLength:41
InterfaceCount:1
ConfigID:1
StringIndex:0
Attributes:0x80
MaxPower:250
ConfigString:

Length:9
DescriptorType:Interface
InterfaceID:0
AlternateID:0
EndpointCount:2
Class:Hid
SubClass:0x00
Protocol:0x00
StringIndex:0
InterfaceString:

Length:7
DescriptorType:Endpoint
EndpointID:0x81
Attributes:0x03
MaxPacketSize:64
Interval:1
Refresh:0
SynchAddress:0x00

Length:7
DescriptorType:Endpoint
EndpointID:0x02
Attributes:0x03
MaxPacketSize:64
Interval:1
Refresh:0
SynchAddress:0x00

mcuee, 2022-07-18:
You have to know the device communication protocol in order to write more useful libusb/libusbdotnet programms.

If you do not know, then you have to carry out reverse engineering.
Example: https://slomkowski.eu/tutorials/eavesdropping-usb-and-writing-driver-in-python/

dmitriyburdan, 2022-07-18:
@mcuee thanks for your reply. I know the communication protocol of this device, but having troubles starting initial communication

mcuee, 2022-08-23:
@dmitriyburdan 
If you know the communication protocol, to rule out libusbdotnet issue, you can try plain libusb (C based codes) or pyusb to see if that works.

dmitriyburdan, 2022-08-23:
@mcuee thanks for the note! we went with the node.js based hid module that has all features we needed.

mcuee, 2022-08-23:
In this case, I will close the issue. Thanks.

closed 2022-08-23 by mcuee