LibUsbDotNet/LibUsbDotNet #168: Code 31: This Device Is Not Working Properly [when debugging via ST-LINK]

tms320, 2022-08-22:
Hi!
I'm developing custom USB device with two identical interfaces using STM32F107 microcontroller. Each interface has two bulk end-points (IN and OUT).
When I connect my device to PC there are two devices for each interface appear in the Windows device manager.
I generated WinUSB drivers for each interface (sub-device) using Zadig. Drivers installed successfully.
Test_Bulk.exe application works properly, I can send data to both my interfaces to OUT-end-points.
But if I run my STM32 software in debug mode (from STM32CubeIDE + ST-LINKV2) I am getting the problem. If I try to send data via Test_Bulk.exe application I get error:
Code 31: This Device Is Not Working Properly.
But the Windows device manager shows that everything is Ok, my device is working properly.
Disabling/Enabling the device in device manager (debug session in IDE is alive during this process) doesn't help.
Deleting the device from device manager and rescan configuration of devices (debug session in IDE is alive during this process) doesn't help.
When I modify my STM32 software and leave only one USB interface - the problem disappearing.
What am I doing wrong?
Where could be the cause of the problem?

mcuee, 2022-08-23:
I think it is difficult to get USB device working in debug mode due to timing issues. If your device works fine in normal operation, no need to worry,

tms320, 2022-08-24:
I found the solution:
After run software in debug mode (in STM32IDE) it is necessary to reset the usb device. I don't know how to do it via Windows Device Manager. I am using "UsbTreeView.exe" tool for it. In this utility, my device is displayed as a single composite device. And there is "Restart device" command in context menu.

mcuee, 2022-08-24:
Thanks for reporting back and glad that you sorted out the issue.

closed 2022-08-24 by mcuee